system76/firmware-open #200: darp7 firmware upgrade broke shutdown functionality and something is deleting efibootmgr entries.  Fixed with firmware downgrade.

HippieMitch, 2021-04-17:
I recently upgraded to the latest firmware.  After doing so, the darp7 didn't shutdown on it's own when it says "shutting down in 5 seconds", the power indicator remained on.  I manually shut it down after about ten minutes.  It turned on okay and said that the firmware installation was successful.  It booted fine.  When I tried to turn the device off from the operating system, the green power light remained on after the display turned off until I manually shut the device down again (the same behavior from the firmware installation).  On reboot it said that no bootable disks were found.  I ran "efibootmgr list" from a live cd and a chroot environment and all of my boot entries were gone.  I reinstalled the boot loader and it worked for a time.  The efi entries were deleted again when I rebooted the device a few times, reboot does work still.  I just downgraded to the previous firmware and everything seems fine for a few days now.  I used these instructions [https://support.system76.com/articles/open-firmware-smmstore/](url) if anyone else is having issues.  I didn't take any pictures, but I know it isn't kernel related as this behavior happened from the firmware installer too.  I just want this documented to help the devs.

michael-catanzaro, 2021-04-18:
Adding that the 04/07 darp7 firmware upgrade broke shutdown functionality on my darp7 as well.

OS shutdowns but PC stays powered on (keyboard backlight, power indicator etc.)

Attaching screenshot of kernel version, OS version, and firmware version.
[image]

jacobgkau, 2021-04-20:
The shutdown issue is already reported here: https://github.com/pop-os/linux/issues/41 (It may not be entirely a kernel issue, but the issue is currently in the kernel repository because a kernel update exposed the issue.)

I'm not sure about the "no bootable media" issue. The smmstore-clearing firmware will clear boot entries (it was published only for a specific issue that required clearing the smmstore to recover.) Even without any boot entries in efibootmgr, the firmware should still attempt to boot an available OS if one is present. Does unplugging the AC adapter for a few seconds to reset the EC allow the system to boot again?

HippieMitch, 2021-04-20:
Thanks for finding that.  Yeah, you are right about unplugging the AC adapter.  I'm glad they're tracking it already.  Is the implication of it being something in the kernel that we can update the firmware to work with the newer kernel changes?  Anyway, thanks for the info!  Should I close this?

jacobgkau, 2021-04-20:
If unplugging the AC adapter to reset the EC takes care of the "no bootable media" issue, then yes, I think this is a duplicate and we should close it.

We've got a kernel update approved to go out for Pop!_OS users (and anyone with our Ubuntu PPA) that will revert the commit exposing this issue, but https://github.com/pop-os/linux/issues/41 will remain open until we have a distribution-agnostic fix, whether that's a firmware update or an upstream kernel update, so please follow that issue for updates. Thank you for the report!

HippieMitch, 2021-05-05:
@michael-catanzaro What did you end up doing to fix this?  I am having no issues with the downgraded firmware.  Using the 5.11 kernel as others have mentioned doesn't seem to work for me.  I am also using Arch :).  
[image]

jacobgkau, 2021-05-05:
https://github.com/pop-os/linux/issues/41 was closed when the Pop!_OS/Ubuntu kernel patch went out. I'm re-opening this issue to continue to track the problem (it can be recreated with a Fedora 34 live disk.) It sounds like this is likely a VBT issue and would need to be fixed in the firmware.

(Anyone adventurous enough to compile a kernel is welcome to try reverting this commit in their kernel of choice, as a temporary workaround: https://github.com/pop-os/linux/commit/e219ef912a025a5e80e6bc9622620100763285fc )

crawfxrd, 2021-12-08:
Is this still reproducible?

jacobgkau, 2021-12-08:
> Is this still reproducible? I don't see it on the new galp5.

I'm not seeing it on lemp10 with Fedora 35. Assuming this was addressed in the upstream kernel.

closed 2021-12-08 by jacobgkau