system76/firmware-open #201: Plundervolt locked MSRs needs a toggle

SpookySkeletons, 2021-04-18:
Coreboot configuration should provide a build-time toggle to disable the plundervolt mitigations so the user can easily undervolt their CPU.
I guess as a start for those affected: what microcode revision did Intel begin to lock the MSRs in? One could easily rollback the ucode to fix the issue as a stopgap.

MilesBHuff, 2021-04-20:
Please!  I have terrible thermals on my oryp7, and the inability to undervolt is just incredibly unfortunate.

MilesBHuff, 2021-04-20:
If System76 wants to get really ambitious, adding a way to configure voltages in the BIOS itself would allow the mitigation to stay enabled in userland.

SpookySkeletons, 2021-07-07:
MSR 0x194 bit 20 seems to be the culprit.

The microcode update seems responsible for toggling this bit in the boot sequence which can't be undone until a reboot so as long as the BIOS can prevent a wrmsr at bit 20 undervolt will be made available.

SpookySkeletons, 2021-07-07:
Trouble. It appears that Intel baked the latest microcode update into these 10th gen H CPUs silicon with Plundervolt mitigations.

So unless Intel wants to release signed microcode firmware that doesn't lock 0x194 we might be straight out of luck even if System76 gets in on the gig.

SpookySkeletons, 2021-08-22:
Good news: It looks like vendors such as XMG and Schenker get access to a special microcode revision courtesy of Intel which may be ripped from their native BIOS and integrated into coreboot! Will test as soon as I can and let you know if the system can then benefit from limited undervolting range.

SpookySkeletons, 2021-08-31:
It works.

Quick guide without having to rebuild your firmware as I couldn't figure it out on Gentoo and would have to spin up a Pop_OS instance to rebuild I just gave up with the rust issues:

1. Extract the exact ucode body using UEFItool in wine you will need from an XMG or a Schenker machine. This will vary based on your CPU. There will be several, choose which CPU you have and extract that one.
2. Use flashrom to read your bios, latest git.
3. cbfstool system76.rom remove -n cpu_microcode_blob.bin
4. cbfstool system76.rom print to confirm you removed the blob
5. cbfstool system76.rom add -f Microcode_Intel_10th_UNDERVOLT.bin.ucb -n cpu_microcode_blob.bin -t microcode
6. print again to confirm the blob was added, mine was the same ucode revision as the removal but less bytes so it left some null space after it
7. sudo ./flashrom -p internal -w system76.rom
8. reboot and enjoy the coolness

System76 can add XMG or Schenker ucode to their repos/ fork off intel to fix the issue quite easily if they please it seems.

Titaniumtown, 2021-12-03:
Any plans on this being integrated into firmware-open?

Localacct21, 2022-01-13:
> If System76 wants to get really ambitious, adding a way to configure voltages in the BIOS itself would allow the mitigation to stay enabled.

i second this!

rhjdvsgsgks, 2022-05-25:
> It works.
> 
> Quick guide without having to rebuild your firmware as I couldn't figure it out on Gentoo and would have to spin up a Pop_OS instance to rebuild I just gave up with the rust issues:
> 
>     1. Extract the exact ucode body using UEFItool in wine you will need from an XMG or a Schenker machine. This will vary based on your CPU. There will be several, choose which CPU you have and extract that one.
> 
>     2. Use flashrom to read your bios, latest git.
> 
>     3. cbfstool system76.rom remove -n cpu_microcode_blob.bin
> 
>     4. cbfstool system76.rom print to confirm you removed the blob
> 
>     5. cbfstool system76.rom add -f Microcode_Intel_10th_UNDERVOLT.bin.ucb -n cpu_microcode_blob.bin -t microcode
> 
>     6. print again to confirm the blob was added, mine was the same ucode revision as the removal but less bytes so it left some null space after it
> 
>     7. sudo ./flashrom -p internal -w system76.rom
> 
>     8. reboot and enjoy the coolness
> 
> 
> System76 can add XMG or Schenker ucode to their repos/ fork off intel to fix the issue quite easily if they please it seems. Please also note, this is applicable to ALL 10th gen system76 notebooks, can be configured to set sane undervolts by default using coreboot to give each notebook a slight performance boost update if anyone has interest.

unfortunately it not working for me. i tried microcode from SCHENKER_SLIM15, XMG_DJ15, N960Sx_SGOW_03THZX, x2100_17012022_linux, and CLCN21WW. but none of them able to unlock msr 0x150. can you share the bios which you used to extract the microcode?

rodrigo-ceccato, 2025-01-15:
hope this get solved... undervolting would be a way to actually get the advertised 14hrs battery life on the lemur pro...