system76/firmware-open #210: oryp5

kevr, 2021-05-20:
Hello guys. What exactly needs to be done to get the open firmware working on the oryp5? I would love to use it and have experience writing software.

I'm very interested in working on a patch for this. Is this something you guys would be open to?

Thanks for your time!

crawfxrd, 2021-05-20:
- #132: NVIDIA GC6 2.1 support
  - Ref: system76/coreboot#38
- #154: TAS5825M initialization error
  - Ref: https://github.com/system76/smart-amp
- EC: Fan curve improvements
  - Ref: system76/ec#164, system76/ec#167, system76/ec#190

kevr, 2021-05-24:
Thank you very much! I'll begin working on it within days.

kevr, 2021-05-30:
Bit of a stopper; unfortunately, I need another laptop before I can attempt messing with this stuff on my oryp5. I need at least one laptop to continue working on things. But, I will be saving up and purchasing a new affordable laptop, and when I do, I will immediately jump on this.

DraconicNEO, 2022-06-29:
Any progress on getting open firmware working on the oryp5? I know many oryp5 users (including myself) would be very happy to finally have open firmware.

crawfxrd, 2022-07-05:
Need to test the if the GPU works with the coreboot driver rewrite.

Need to test if system76/smart-amp#12 fixes the problems with the speakers.

crawfxrd, 2022-07-08:
GPU is not detected after the rewrite.

#154 is still a problem.

crawfxrd, 2022-07-09:
Should be in a good state to test with system76/coreboot#123.

crawfxrd, 2023-07-28:
Dev work was complete and was working when I returned the oryp5 (minus the parts I broke :slightly_smiling_face:).

Release PR is open, if we ever get around to it..

closed 2023-07-28 by crawfxrd