system76/firmware-open #214: fatal: reference is not a tree: 16778e4a41477498eaf5791cae57ae404515973e

svrooman, 2021-06-19:
Hi,

I'm trying to update my keyboard map in the firmware. The PgUp and PgDn keys are killing me. I had updated it once before but the new firmware revision overwrote my changes.

I'm currently running into this issue:

```
sudo dmidecode -t bios | grep Version

Version: 2021-04-07_236914e
```

```
git ls-tree 236914e ec

160000 commit 16778e4a41477498eaf5791cae57ae404515973e	ec
```

```
git checkout 16778e4a41477498eaf5791cae57ae404515973e

fatal: reference is not a tree: 16778e4a41477498eaf5791cae57ae404515973e
```

I then tried:

```
git log --find-object=16778e4a41477498eaf5791cae57ae404515973e

commit 4a88f78546705790eabbc0f59ddf1c0556bc9a16
Author: Jeremy Soller <jeremy@system76.com>
Date:   Fri Apr 2 13:43:52 2021 -0600

    Update EC and changelog
```

```
git checkout 4a88f78546705790eabbc0f59ddf1c0556bc9a16
```

But the `ec` directory is empty.

```
ls -lha ec                                                                                                        10:23:47 AM

total 8.0K
drwxrwxr-x  2 XXX XXX 4.0K Jun 19 09:46 .
drwxrwxr-x 15 XXX XXX 4.0K Jun 19 10:21 ..
```

Any guidance would be appreciated, Thank you!

crawfxrd, 2021-06-19:
`16778e4` is the EC project commit, not the firmware-open commit.

You should either:

- Checkout `236914e` in firmware-open and update all submodules
- Checkout `16778e4` in just the EC project

(Both result in the same thing WRT the EC commit used.)

Then you can build and flash EC firmware that corresponds to the BIOS firmware release.

svrooman, 2021-06-20:
@crawfxrd Ohhhh, I see what I did there. Doi. Thanks for the clarification / help. Firmware built, and keys mapped! Thanks very much for the help.

closed 2021-06-20 by crawfxrd