system76/firmware-open #215: oryp6 - fan buzzy sounds at lower speed on latest firmware

nmatkheev, 2021-06-20:
Hey guys,

I've been experiencing buzzy sounds after upgrading to latest firmware (`2021-03-16_7c3e89e`)  
Before that FW version fan was okay.  
Any suggestions will be appreciated

leviport, 2021-06-21:
Interesting, that update was made to fix fan buzzing at low fan speeds. Do you have the 17" or 15" oryp6? 

There were recently some more [fan speed improvements](https://github.com/system76/ec/pull/190) made that just haven't been in a firmware/EC release yet. My 15" oryp6 doesn't seem to buzz after applying the fan smoothing changes, so you may have quieter fans after those changes.

MilesBHuff, 2021-07-20:
I actually have fan buzzing as well, on an oryp7 17".  It only happens when fans spin up from 0, or spin down from 25;  so I may have a different issue.

This is just a guess:  but maybe OP had no buzzing at 20%, but has buzzing at 25% (see:  https://github.com/system76/ec/issues/163#issuecomment-800586557).  It might be that the fans, being mainland Chinese, have variable workmanship from chàbùduō-ism.

There may be little recourse than for OP to downgrade or flash a custom version of the EC.
Or maybe the lowest allowed speed could be increased a third time?

markwylde, 2021-09-16:
Yeah I've been experiencing a buzzing since the last firmware upgrade, just when the fans are starting up for maybe 1 second, then it sounds fine. Very distracting.

crawfxrd, 2022-07-27:
Was this fixed by system76/ec#190?

leviport, 2022-07-27:
It did not. In fact, smoothing from 0% to 25% made the buzzing occur for much longer until it reached 25% or 0%, so we [set the lowest fan point at 25%](https://github.com/system76/ec/pull/167) and made sure it didn't smooth between 0% and 25%. There is still a little bit of noise when the fan starts and stops, but it seems fairly infrequent now.