system76/firmware-open #216: Boot Option Restoration/Restored

ahoneybun, 2021-06-24:
Issue:

A blue screen that is labeled either Boot Option Restoration/Restored is shown when the drive (Samsung in this case). If you click 'Continue boot' is picked you are taken to the bootloader (GRUB) in this testing. Here is a photo of the screen:

[image]

System: galp5
GPU: NVIDIA 1650
CPU: i7-1165G7
Drive Samsung 970 EVO Plus 500GB

Recreate:

1. Install Pop 20.04 LTS
2. Install Ubuntu 20.04 LTS
3. Reboot into Open Firmware and select the drive in One Time Boot

jackpot51, 2021-06-25:
This might be specific to the galp5. I have this model and can test it out.

thomas-zimmerman, 2021-06-28:
I know support has seen this happen on a couple lemp9 and lemp10 systems as well.

tunix, 2022-07-13:
I see this screen on my lemp10 every few boots since I've installed Fedora Silverblue on it. I think that, out of 1 of every 3 boots, I see it. Waiting few seconds and the computer reboots by itself and I'm back to GRUB which will take me to the desktop. Let me know if I can give you further data about this error.

thomas-zimmerman, 2022-07-13:
@tunix We had another system that was doing this and it took a hardware firmware flash to fix. Have you opened a support case for us to supply a USB image of the firmware to see if that clears this up?

tunix, 2022-07-13:
@thomas-zimmerman No, I haven't. I have System76's Firmware Manager installed and received updates before. But I haven't received any updates after a restoration to a previous build was made by System76. Currently I have `2021-07-20_93c2809`

Should I still do this? Why is it not available through this Firmware Manager?

thomas-zimmerman, 2022-07-14:
Adding the ability to move backwards on firmware release is a decent amount of work and there can be updates around ME firmware versions, where you have firmware breaking changes that can not be downgraded.

There is a command line tool that can reinstall the latest firmware release:
```
sudo apt install system76-firmware
sudo system76-firmware-cli schedule
```

tunix, 2022-07-16:
Hi @thomas-zimmerman, I already have system76-firmware package installed on Fedora Silverblue. I issued the schedule command and on the next boot, it did flashed the firmware successfully. Hovewer, once I booted the computer back, I saw the Boot Option Restoration screen again. 😞

aaron-trout, 2022-12-10:
Where does this screen come from, is it the Samsung SSD producing it? My UEFI boot entry for my Linux install keeps disappearing every now and then and I had no idea why until I noticed this screen flash up for a second on boot today! I'm not using a System76 machine so sorry if this is dirtying up this thread / going off topic (struggling to find anything about this boot restoration screen online). Of the listed hardware the Samsung SSD is the only thing I have in common:

> System: galp5
> GPU: NVIDIA 1650
> CPU: i7-1165G7
> Drive Samsung 970 EVO Plus 500GB

I am using a Gigabyte motherboard with AMD Ryzen and Samsung SSDs (an M2 PCIe and also a regular SATA drive).

benintech, 2022-12-23:
Same question here. I too don't have a System76, so this blue screen is definitely not specific to System76 computers. @aaron-trout , it's not from your samsung SSD either, because my SSD is a Western Digital one. I am using an Asus ROG G752VS laptop, with Intel i7 processor.

Does the blue screen belong to GRUB maybe ?

You can also check [this thread](https://unix.stackexchange.com/questions/705040/install-linux-via-usb-boot-shows-boot-option-restore/729458#729458), where I explain in which circumstances exactly this blue screen started to appear. There are very few infos on this topic on the web, I really would like to understand where that blue screen came from, and what's its purpose.

crawfxrd, 2022-12-23:
It's from grub. I believe this happens when you attempt to boot from the auto-generated entry for the drive.

benintech, 2022-12-23:
> It's from grub.

Thanks, I had a hunch that was the case.

> I believe this happens when you attempt to boot from the auto-generated entry for the drive.

In my case the blue screen started to appear after I applied [Debian's workaround for buggy UEFI systems](https://wiki.debian.org/UEFI#Force_grub-efi_installation_to_the_removable_media_path) that don't install the bootloader in the right directory. That workaround also installs the bootloader in the ["removable media path"](https://wiki.debian.org/UEFI#Booting_from_removable_media). Note : in my installation I chose to install the EFI partition on a USB stick. Without that workaround it didn't boot at all, with that workaround it boots but opens the blue screen if I previously started the system without the EFI USB stick plugged in (details [here](https://unix.stackexchange.com/questions/705040/install-linux-via-usb-boot-shows-boot-option-restore/729458#729458)).

erik-msnr, 2024-05-10:
I am probably late, but I encountered this while doing a migration from CentOS to Rocky / Alma Linux. 
I was able to solve this through deleting the `/boot/efi/EFI/BOOT` folder before rebooting into the new OS.

yangl1996, 2025-01-27:
Also late to the party but I found some reference of the blue screen here.

https://wiki.batocera.org/secureboot