system76/firmware-open #218: Dual boot options keep appearing and disappearing in dual boot on one drive setup

Raikiri, 2021-06-30:
On my Oryx Pro I wanted to have Windows 10 and to keep the stock PopOS on the same physical drive. I know it's not advised and I know I was sort of going the hard mode when I chose to, but I still gave it a try.

So before installing Windows 10, open firmware boot menu had these options:
PopOS
PopOS recovery
Samsung EVO 1TB (name of my ssd device)

Right after installing Windows 10 I checked the boot menu again and saw this:
PopOS
PopOS recovery
Windows Boot Loader
Samsung EVO 1TB  (name of my ssd device)

Then I ran a Windows Update and only this option was left:
Samsung EVO 1TB

Choosing this option boots me into systemd loader where I can select either PopOS or Windows Boot Loader, but all other entries were gone from the bios boot options.

So I ran windows boot recovery thingie from its installation drive and Windows Boot Loader re-appeared in the firmware boot menu after that, but some time later, it disappeared again.

So right now I'm at the stage where I still have just the physical drive listed as my only boot option in the bios, which boots me into systemd loader where I am able to log in either Windows or PopOS, but I still expect the firmware boot menu to work as well, but its options are missing.

Am I doing something wrong? Is it working as intended? Is there a way to restore (and keep) boot options available from the firmware?

mbk5631, 2021-07-06:
This is very similar to a problem I'm running into. I have a lemp9. I have two drives, One has pop-os and the other has windows. 

I was running firmware rev 2020-09-17-f10af76 from September to May.  Then I updated to 2021-03-11_50eedc2 because I wanted the battery thresholds. Now it loses the windows boot loader.  I can repair it through a windows USB key and it will work again for a couple reboots. Sometimes the record will persist if I boot pop-os, and sometimes it won't.  Doesn't seem to matter if I am booting windows to windows, it will still lose the record sometimes.

The windows drive has bitlocker turned on, but I was able to reproduce the problem after turning off bitlocker (decrypting the drive). Obviously, running without it presents security issues on a portable device.

When it's working efibootmgr reports...

BootCurrent: 0001
Timeout: 2 seconds
BootOrder: 0004,0001,0000,0002
Boot0000* UiApp	MemoryMapped(11,0x840000,0xffffff)/FvFile(xxxxxxxx-xxxx-xxxx-xxxxxxxxxxxx)
Boot0001* Pop!_OS 21.04	HD(1,GPT,xxxxxxxx-xxxx-xxxx-xxxxxxxxxxxx,0x800,0x100000)/File(\EFI\SYSTEMD\SYSTEMD-BOOTX64.EFI)
Boot0002* Samsung SSD 970 EVO Plus 1TB 	PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,xx-xx-xx-xx-xx-xx-xx-xx)N.....YM....R,Y.
Boot0004* Windows Boot Manager	HD(1,GPT,xxxxxxxx-xxxx-xxxx-xxxxxxxxxxxx,0x800,0x100000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS....... (there are more characters here)

When it's not working I get 
BootCurrent: 0001
Timeout: 2 seconds
BootOrder: 0000,0001
Boot0000* UiApp	MemoryMapped(11,0x840000,0xffffff)/FvFile(xxxxxxxx-xxxx-xxxx-xxxxxxxxxxxx)
Boot0001* Samsung SSD 970 EVO Plus 1TB 	PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,xx-xx-xx-xx-xx-xx-xx-xx)N.....YM....R,Y.

jacobgkau, 2021-08-05:
@mbk5631 Does the systemd-boot menu show Windows in its menu like it did for @Raikiri? If it's showing up there, I would recommend just using that.

If not, you may need to copy some Windows bootloader files into the Pop!_OS EFI partition so you can create a systemd-boot entry manually. There's a community post here that includes a copy command that might work: https://pop-planet.info/forums/threads/copy-the-microsoft-bootloader-into-pops-efi-beginners-guide.357/

mbk5631, 2021-08-08:
I can select Windows using the systemd boot but it's not really a viable solution because the next step is to re-enter the BitLocker recovery key each time. It doesn't read the key from the TPM chip.

My work-around is each time I boot into Linux, which is most of the time, I run 

sudo efibootmgr -c -L "Windows Boot Manager" -l "\EFI\Microsoft\Boot\bootmgfw.efi" -d /dev/nvme0n1 -p 1

Then the next reboot will show the Windows boot manager and boot correctly into Windows.

mbk5631, 2021-08-11:
BTW just to clarify this is a regression. It worked in firmware from last September.

pspacek, 2021-08-17:
I'm observing similar symptoms on darp7 on firmware 2021-04-07_236914e, which is I believe the last released.

In my case the boot menu "reverts" to state with just single single entry named after the SSD drive. My workaround is also re-add entries using efibootmgr, which stays there for couple reboot cycles and then disappears again.

I was not able to pinpoint and verify what is causing this "reversion", but I have a vague suspicion it has something to do with PopOS updater.

crawfxrd, 2021-08-17:
When it happens, check if SMMSTORE was cleared by CMOS variable.

```
make -C coreboot/util/cbmem
sudo ./coreboot/util/cbmem -c > cbmem.txt
grep "CMOS reset" cbmem.txt
```

You can also try `scripts/smmstore.sh`. This will read the flash, extract the SMMSTORE region into `smmstore.rom`, and attempt to parse & print the variables.

mbk5631, 2021-08-18:
ok I built coreboot this morning. Here's the output of coreboot

output of 

make -C coreboot/util/cbmem
sudo ./coreboot/util/cbmem/cbmem -c > cbmem.txt
grep "CMOS reset" cbmem.txt

was

SMMSTORE: CMOS reset, clearing store

Full text for cbmem is below. I have the output for smmstore - since I can't read most of it I'll mail it to you rather than posting it.

coreboot-2021-03-11_50eedc2 Thu Mar 11 22:19:48 UTC 2021 bootblock starting (log level: 7)...
CPU: Intel(R) Core(TM) i5-10210U CPU @ 1.60GHz
CPU: ID 806ec, Whiskeylake V0, ucode: 000000d5
CPU: AES supported, TXT NOT supported, VT supported
MCH: device id 9b61 (rev 0c) is CometLake-U (4+2)
PCH: device id 0284 (rev 00) is Cometlake-U Premium
IGD: device id 9b41 (rev 02) is CometLake ULT GT2
FMAP: Found "FLASH" version 1.1 at 0x450000.
FMAP: base = 0xff000000 size = 0x1000000 #areas = 5
FMAP: area COREBOOT found @ 450200 (12254720 bytes)
CBFS: mcache @0xfef21400 built for 17 files, used 0x3b8 of 0x2000 bytes
CBFS: Found 'fallback/romstage' @0x80 size 0xcb8c in mcache @0xfef21438
BS: bootblock times (exec / console): total (unknown) / 0 ms

coreboot-2021-03-11_50eedc2 Thu Mar 11 22:19:48 UTC 2021 romstage starting (log level: 7)...
pm1_sts: 0100 pm1_en: 0000 pm1_cnt: 00001800
gpe0_sts[0]: 00000000 gpe0_en[0]: 00000000
gpe0_sts[1]: 00000000 gpe0_en[1]: 00000000
gpe0_sts[2]: 00000000 gpe0_en[2]: 00000000
gpe0_sts[3]: 00000800 gpe0_en[3]: 00000000
TCO_STS:   0000 0000
GEN_PMCON: e0015038 00000200
GBLRST_CAUSE: 00000000 00000000
prev_sleep_state 5
CBFS: Found 'fspm.bin' @0x59dc0 size 0x8e000 in mcache @0xfef2164c
POST: 0x34
FMAP: area RW_MRC_CACHE found @ 400000 (65536 bytes)
SPD INDEX = 0
CBFS: Found 'spd.bin' @0x55dc0 size 0x200 in mcache @0xfef2159c
SPD: module type is DDR4
SPD: module part number is K4AAG165WA-BCTD     
SPD: banks 8, ranks 1, rows 17, columns 10, density 16384 Mb
SPD: device width 16 bits, bus width 64 bits
SPD: module size is 8192 MB (per channel)
memory slot: 0 configuration done.
POST: 0x36
POST: 0x92
POST: 0x98
CBMEM:
IMD: root @ 0x99eff000 254 entries.
IMD: root @ 0x99efec00 62 entries.
External stage cache:
IMD: root @ 0x9abff000 254 entries.
IMD: root @ 0x9abfec00 62 entries.
FMAP: area RW_MRC_CACHE found @ 400000 (65536 bytes)
MRC: Checking cached data update for 'RW_MRC_CACHE'.
SF: Detected 00 0000 with sector size 0x1000, total 0x1000000
MRC: 'RW_MRC_CACHE' does not need update.
2 DIMMs found
SMM Memory Map
SMRAM       : 0x9a000000 0x1000000
 Subregion 0: 0x9a000000 0xa00000
 Subregion 1: 0x9aa00000 0x200000
 Subregion 2: 0x9ac00000 0x400000
top_of_ram = 0x9a000000
MTRR Range: Start=99000000 End=9a000000 (Size 1000000)
[... 2920 lines cut ...]
CPU physical address size: 39 bits
CPU physical address size: 39 bits
CPU physical address size: 39 bits
CPU physical address size: 39 bits
CPU physical address size: 39 bits
CPU physical address size: 39 bits
CPU physical address size: 39 bits
Loading segment from ROM address 0xff56c714
  Entry Point 0x00800860
BS: BS_PAYLOAD_LOAD run times (exec / console): 318 / 0 ms
POST: 0x95
POST: 0x95
POST: 0x88
POST: 0x89
SMMSTORE: CMOS reset, clearing store
FMAP: area SMMSTORE found @ 410000 (262144 bytes)
Finalizing chipset.
Finalizing SMM.
APMC done.
POST: 0xfe
BS: BS_PAYLOAD_LOAD exit times (exec / console): 158 / 0 ms
POST: 0x7b
mp_park_aps done after 0 msecs.
Jumping to boot code at 0x00800860(0x99a81000)
POST: 0xf8

mbk5631, 2021-08-18:
ok, I'm really not the slightest bit familiar with this code but I can only find (via grep) one place this message is generated, and in that same place is the only place I see the variable preserve_smmstore being set.

clear_store_on_reset appears to be the only place preserve_smmstore is set.

If preserve_smmstore does not return success and if the return message was not CMOS_CHECKSUM_INVALID then the preserve lval is set in the first block. That preserve value is never written.  The only case in which preserve_smmstore is set is if the return value when fetched was CMOS_CHECKSUM_INVALID, at which point smmstore_clear_region is called and if that's successful, preserve_smmstore is set.

So on our machines is get_option(preserve_smmstore) returning something besides CB_SUCCESS and besides CB_CMOS_CHECKSUM_INVALID?

The only get_option I found in the tree is in payloads/libpayload/drivers/options.c. It only returns 0 and 1 and doesn't appear to use any of the enumerations. Is it the right one???

I could be totally wrong here... 

Anyway back to my day job

crawfxrd, 2021-08-18:
CMOS values can be viewed with `nvramtool`.

```
make -C coreboot/util/nvramtool
sudo ./coreboot/util/nvramtool -a
````

mbk5631, 2021-08-18:
ok thanks!

so right now it returns

pop-os:~/firmware-open$ sudo ./coreboot/util/nvramtool/nvramtool -a
boot_option = Fallback
reboot_counter = 0x0
preserve_smmstore = 0x1

So the value does exist.

There's a cmos_get_option in option.c.  Is that the one that's supposed to be called?  Ignore my earlier comments if that's true.

looks like there's an inline in option.h wrapping it with get_option. Missed that earlier.

The only cmos option the boot complains about not finding is poweron_after_fail, which isn't this one. 

BTW you check for checksum after finding the variable and only then do you return CMOS_CHECKSUM_INVALID

So I have to assume it's finding the variable then failing the checksum test. The only way the reset gets executed. (there’s no printk for successful search)

If CMOS_CHECKSUM_INVALID is returned to clear_store_on_reser then smstore_clear_region is called. If that succeeds then preservE_smmstore is overwritten.  If that call does not succeed then the previous value of preserve_smmstore is not changed and remains present.

The failure of smmstore_clear_region is not logged, so based on the code it was successful or CMOS_CHECKSUM_INVALID in fact was not returned.

cmos_checksum_valid does not log but the only two conditions I see for calling smmstore_clear_region are if the variable is not present or if CMOS_CHECKSUM_INVALID is returned.

It’s now present, and since it was not logged as a missing variable on reboot then the value returned should have been CMOS_CHECKSUM_INVALID.

Do you agree?

mbk5631, 2021-08-18:
If I ask nvramtool for the checksum (nvramtool -c) it gives me 0x1

$ sudo ./nvramtool -c
0x1

If I dump everything I get several checksums that are not 0x1...  ? 

$ sudo ./nvramtool -Y
entries
0 384 r 0 reserved_memory
384 1 e 2 boot_option
388 4 h 0 reboot_counter
408 1 h 1 preserve_smmstore
984 16 h 0 check_sum

enumerations
1 0 Disable
1 1 Enable
2 0 Fallback
2 1 Normal
3 0 Emergency
3 1 Alert
3 2 Critical
3 3 Error
3 4 Warning
3 5 Notice
3 6 Info
3 7 Debug
3 8 Spew

checksums
checksum 392 983 984

crawfxrd, 2021-08-18:
`-Y` prints the CMOS layout. This is the `cmos.layout` file in the coreboot tree.

crawfxrd, 2021-08-19:
Can force the issue by syncing time on Windows.

crawfxrd, 2021-08-19:
Windows is writing the century byte (0x32) and invalidating the checksum. coreboot loads the default of 0 for the option and SMMSTORE is cleared.

pspacek, 2021-08-20:
FTR I'm seeing this issue even on machine which has only PopOS on it, and the UEFI menu contains "extra" boot options for iPXE and UEFI shell. I.e. no Windows involved, and the only OS which gets booted up is PopOS.

crawfxrd, 2021-08-20:
Can you reliably reproduce it by booting _only_ Pop?

You can try dumping the CMOS (`nvramtool -x`) before shutdown/reboot to see if something is writing to it.

pspacek, 2021-08-23:
> Can you reliably reproduce it by booting _only_ Pop?
Reliably - no, I did not find a way to reproduce it at will.

On the other hand, I'm almost 100% sure UEFI Boot Manager items sometimes disappear even if the only OS started on that machine is Pop. I've nuked Pop installation and replaced it with Arch yesterday, but I have already witnessed Boot Manager items disappearing as well, so I will monitor it and try the `nvram -x` trick to see what is going on.

`nvramtool -x` is currently crashing with:
```
nvramtool: util/nvramtool/accessors/cmos-mem.c:17: mem_hal_read: Assertion `mem_hal_data != (unsigned char*)-1' failed.
```
Do I need to do some setup? (I'm executing it under root and `nvramtool -Y` shows a sensible layout.)

jacobgkau, 2021-08-23:
@pspacek This issue was identified and fixed in https://github.com/system76/coreboot/pull/72, which is why the issue is marked as closed. It won't be part of the next firmware update, but once an update dated later than August 20th is released, then the fix should be included. In the meantime, you could try building and flashing updated firmware locally on your system to confirm if that PR fixed the issue. First, install Rust nightly from [rustup.rs](https://rustup.rs/), then run these commands:

```
git clone https://github.com/system76/firmware-open
cd firmware-open
./scripts/update.sh
./scripts/deps.sh
cd coreboot
git checkout master
cd ..
./scripts/build.sh <your-model-here>
./scripts/flash.sh <your-model-here>
```

The flashing script will power off the machine. After flashing, you can revert to normal firmware by using the normal firmware manager to "update."

If you're still seeing the issue with the latest version of `coreboot`, then it could actually be a different issue.

zancas, 2022-04-14:
Hi @jacobgkau is building and flashing my own version of the open-firmware still the best solution, or is there a new official firmware I can install?

My system (a galp5, support ticket:  63754) is still on:

`Version: 2021-07-20_93c2809` according to the boot config screen.  Does that sound right to you?

zancas, 2022-04-15:
> @pspacek This issue was identified and fixed in [system76/coreboot#72](https://github.com/system76/coreboot/pull/72), which is why the issue is marked as closed. It won't be part of the next firmware update, but once an update dated later than August 20th is released, then the fix should be included. In the meantime, you could try building and flashing updated firmware locally on your system to confirm if that PR fixed the issue. First, install Rust nightly from [rustup.rs](https://rustup.rs/), then run these commands:
> 
> ```
> git clone https://github.com/system76/firmware-open
> cd firmware-open
> ./scripts/update.sh
> ./scripts/deps.sh
> cd coreboot
> git checkout master
> cd ..
> ./scripts/build.sh <your-model-here>
> ./scripts/flash.sh <your-model-here>
> ```
> 
> The flashing script will power off the machine. After flashing, you can revert to normal firmware by using the normal firmware manager to "update."
> 
> If you're still seeing the issue with the latest version of `coreboot`, then it could actually be a different issue.

Hi!   When your instructions say:   `<your-model-here>`  does that mean the four-letter-then-single-digit as seen on the firmware splash screen...    e.g. `galp5`?

curiousercreative, 2022-04-26:
Oh man, we're still waiting on a new galp5 firmware release! @zancas yes, I believe you're looking for `galp5`. There's also a way to specify this in a config.mk file, not sure where that's documented but that's a better solution for those of us who are only ever building a flashing for our own device.

mbk5631, 2022-10-02:
Yeah I got to that point already.  Been 15 months now I've been putting up with this. I'm not about to build my own firmware - concerned about bricking or winding up with worse - and I have work to do. At this point I'm trying to decide whether I'm ready to buy a new laptop about 3 years prematurely,  It was ok but not ideal as long as I could boot into Windows fairly easily depending on whether I needed to do a meeting or not but, well, geez! Probably will go back to Mac.

curiousercreative, 2022-10-03:
@mbk5631 out of curiosity, what do you boot into Windows for? Every work meeting requires Windows? That sounds like hell...

mbk5631, 2022-10-03:
Anything involving Zoom or other meeting software or bluetooth. Meeting software eventually bogs down.  Somehow the wifi is sometimes flakey on some less common AP manufacturers (like what we have in this building) where it winds up renegotiating regularly. Neither pipewire nor pulseaudio work well for anything but playback. Anything requiring me to produce a word, or excel, or powerpoint that someone not using LibreOffice is going to see - somehow I almost always find the features that aren't 99.9% compatible. This has gotten somewhat better with a Microsoft Office online subscription, but that's costing me more than just buying the base product for a Mac. Adobe product support (I don't need much). I have tried using VMs and wine, but the level of tinkering is high and even though I have a separate drive for Windows, it just doesn't work well. Linux is my preferred environment - it's the one in which I'm most comfortable, but I'm also quite comfortable on Macs (can always drop into bash).

mbk5631, 2022-10-03:
A note - I don't mind booting into Windows for some things, and the issue with the AP flakiness appears to have been resolved recently. But this defect where I have to keep rewriting the boot record and then escape past the encryption key that doesn't get found on first reboot is a nuisance on something they should have patched a year ago. The boot record rewrite is scripted, so I just have to boot linux/log in to get Windows to work again - but that's bad enough as it slows me down - but Windows also doesn't always find the encryption key so when that happens I have to escape out of the "enter your bitlocker key" page and then it will find the key on the second pass.

curiousercreative, 2022-10-03:
@mbk5631 are you on the pop-os chat server? There's a System76 channel, but I'd be curious if others have a pleasant dual boot experience. I'd search and discuss on the chat server to see if solutions or workarounds exist.

Generally, it sounds like you're having all kinds of pains. I started migrating from macOS about two years ago and I wasted a lot of effort trying to keep a macOS VM running and using it a couple times a week. Ultimately, my experience improved immensely when I dropped it altogether. Most Windows or macOS users will not migrate to Linux when they discover that a piece of software doesn't work well on their OS, they'll just ignore it and notify whoever cares that it doesn't work on their OS. Whether or not you're feeling that bold is for you to decide. More helpful perhaps, I wouldn't recommend running Linux Desktop with your software requirements. I understand the desire to use Linux, against all odds even, but I bet your life will improve dramatically once you amicably part ways with either Linux or the software that's not running well or flat out incompatible.

Ultimately, we'd all like to see these bugs squashed but I understand S76 is quite limited on firmware resources in comparison to outstanding issues and I don't imagine this is as high priority as several other longstanding issues that are likely impacting more S76 users.

curiousercreative, 2022-10-03:
@mbk5631 and not to discount your experience by any means, just more data points, I haven't experienced any WiFi flakiness over the past year and a half. We had a bluetooth kernel bug preventing suspend a couple months ago, but I've been back to solid bluetooth lately. I'm not in meetings all day and my work uses Google Meet (I have it installed as a Chromium PWA), but I've never had any trouble with Zoom when I do participate in them. The only problem I can relate to is while I was running Wayland (I ran for about a year until recently) screen sharing wasn't working in Zoom last time I tried. Running X11 for the past couple months on my galp5 and screen sharing works as expected as does general video call performance. In contrast to others in a conference room, I see consistent load and fan speeds throughout long video calls (in part because I don't have any effects on my own image). All of this is to say, there may be remedies. The MS Office and Adobe software on the other hand, that's more what my last comment is about with cutting your losses and recognizing that Linux Desktop might not be the best fit.

mbk5631, 2022-10-03:
I went into this knowing I needed to dual boot.  I'm mostly ok with doing that when I need to - I spent 20 years as a linux-based  developer, and I've done that before. Dual boot worked fine for the first few months I had my lemp9. Then it was broken. The fix sits in git, but doing the build of just that one fix and applying it to my laptop, on which I depend, is not something I'm about to do. At this point I do need MS and Adobe far more than I did when I bought the laptop, so the result is I'm spending more time in Windows than I used to. But frankly, I should be able to do that as needed and not have to rewrite the boot menu each time. Never had trouble with that in a laptop before.

jthornhill, 2022-10-03:
I don't understand the inclination to blame users here for wanting to do things computers should be able to successfully do.

I ostensibly chose System76 over just going to Clevo (or any other vendor really) because they provide additional development and support with Linux users in mind.

The reality of the lemp10 user experience: dual boot (or in my case, wiping everything and using a different Linux distribution) was initially supported just fine. Then, a new firmware ships that breaks this very-normal-thing that works if I buy almost any off the shelf PC. A "fix" was published in the repo, which closed this issue, but it wasn't made available until months later. The finally released version was pulled for other reasons.

I mean, I would never do business with a company that left things in this state again. I lobbied my employer to get a laptop for me with good Linux support, yet this is the result. If this experience wasn't so poor, I might have even advocated for System76 over e.g. Framework in the future, but the disregard for the firmware quality makes System76 a non-starter in the future.

leviport, 2022-10-03:
Thank you all for the feedback. I'll reopen this issue, but I must ask that we remain focused on tracking the bug on this bug tracking platform, not venting frustrations.

curiousercreative, 2022-10-03:
@jthornhill my intention was not user blame. I think we're all on the same page here and validate your experience and others as something that needs fixing. Here's hoping we see a fix sooner than later, I'd look into it myself if it were impacting me.

crawfxrd, 2023-07-28:
Original issue fixed with the move to SMMSTOREv2 and removal of clearing NVRAM on CMOS reset.

Track #340 to see if a release has been made for your model.

Use #437 if you have the issue on new firmware.

closed 2023-07-28 by crawfxrd