system76/firmware-open #221: Gazelle (gaze15) boot slowed by "failed to write source OUI"

thomas-zimmerman, 2021-07-23:
Customer (support case 62613) is running a 5.12 based kernel and getting a slower than normal boot caused by:
```
"<timestamp in seconds and millionths> failed to write OUI"
```
Messages that scroll on the screen for a couple seconds. Looks like the source is this patch that landed in 5.12:
https://patchwork.kernel.org/project/intel-gfx/patch/20200916171855.129511-2-lyude@redhat.com/
(customer found and reported the patch!)

eliminmax, 2022-05-09:
I know this is old, but I can confirm that it's affecting my Gazelle (gaze15) as well, and has been since I got it in April '21. I am currently on Pop!_OS 21.10 with kernel 5.16.19-76051619-generic. Not a big deal, just a minor annoyance for me, but I figured I'd add this to the issue thread.

crawfxrd, 2022-05-16:
Firmware: 2022-05-16_3a5b422
Kernel: 5.17.5-76051705-generic

```
[    2.876451] fbcon: i915drmfb (fb0) is primary device
[    4.365377] i915 0000:00:02.0: [drm] *ERROR* Failed to write source OUI
[    4.553118] Console: switching to colour frame buffer device 240x67
```

Causes a 1.5s delay in kernel boot time.

Vuwij, 2022-09-04:
[image]
Im getting a similar issue, but it only occurs when I close the lid and re-open it, it hangs here

thefirethirteen, 2022-11-09:
> [image]
> Im getting a similar issue, but it only occurs when I close the lid and re-open it, it hangs here

Exact same message here, also when waking from suspend (or closing and re-opening the lid).
Has anyone found any workaround beside a force shutdown or Alt+SysRq + `REISUB` (for a safer shutdown I believe)?

topoipat, 2023-08-03:
I have the same issue, no idea how to fix

jacobgkau, 2023-12-01:
This issue is about the log message showing up at boot and it slowing down the boot.

When you resume from suspend, the logs you're seeing are just the boot logs still in the buffer. They're not likely to be relevant to the reason the system failed to resume.

Anyone getting failures to resume should reach out to [support](https://support.system76.com/) for troubleshooting, since that can be caused by hardware issues.

firdavsDev, 2024-11-02:
may be just disable suspend.