system76/firmware-open #222: galp5 fan turns on after suspend when USB-C dock is connected (only with kenrel 5.11, 5.8 is ok)

hansmerkl, 2021-07-24:
The fan on my galp5 always turns on when my USB-C dock is connected and the laptop goes into suspend mode. It stays on as long as the laptop stays in suspend mode.

The two docks I have tried this with are:

Anker
https://smile.amazon.com/gp/product/B07ZVKTP53/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1
Hootoo
https://smile.amazon.com/gp/product/B07SHR5JKR/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Is this a firmware issue? What else may be controlling the fan?
Is there a log that may show why the fan turns on?

bflanagin, 2021-07-26:
Thanks for the report @hansmerkl. 

There are numerous logs that might shed light on the issue. The first two places to look would be dmesg('sudo dmesg') and journalctl ('sudo journalctl -r'). I recommend you check each application's available options to help narrow down your search.  Also, If you could supply us with the output of those logs we can help diagnose the issue further. Use the commands below to generate the files we need and attach them to the issue. 

```
sudo dmesg > dmesg.txt
sudo journalctl -r > journal.txt

```
Its also worth noting that sense your are a System76 customer you may want to put in a support ticket over at https://support.system76.com which is often a faster way to get problems resolved than github. Feel free to reference this issue in your support ticket so that the engineers stay in the loop with this issue.

jacobgkau, 2021-07-26:
Suspend/resume not working is currently a known issue on TGL-U systems (lemp10, darp7, galp5). It's currently being tracked here: https://github.com/system76/firmware-open/issues/199 

I'm renaming that issue to be more easily discoverable, and closing this one as a duplicate. @hansmerkl Thank you again for the report. Any additional info you want to provide would fit great in the other issue.

closed 2021-07-26 by jacobgkau

hansmerkl, 2021-07-26:
I am glad to hear I am not the only one with this problem. 

Will add the log output to https://github.com/system76/firmware-open/issues/199