system76/firmware-open #225: Test fwupd Capsule On Disk support

jackpot51, 2021-08-02:
After https://github.com/fwupd/fwupd/releases/tag/1.6.2, fwupd now supports Capsule On Disk. We should test doing firmware updates using that.

crawfxrd, 2022-01-07:
Now that we have the new coreboot and edk2 releases, what would be the next steps for this?

crawfxrd, 2023-10-27:
Adding `EsrtFmpDxe` and adding a skeleton device FMP to edk2 causes the ESRT to be published. This shows up in sysfs under `/sys/firmware/efi/esrt` and will also be listed by fwupd as "UEFI ESRT device".

crawfxrd, 2023-11-02:
Capsules can be generated using edk2's Python script:

```
GenerateCapsule -e --guid <FMP_GUID> --fw-version 0x1 --lsv 0 -o firmware.cap firmware.bin
```

UEFI requires the firmware version be represented as a u32.

crawfxrd, 2024-07-20:
Dasharo is working on Capsule Update support for coreboot-based firmware:

- https://github.com/orgs/Dasharo/projects/7