system76/firmware-open #226: gaze16: touchpad not working in Windows

jacobgkau, 2021-08-11:
The touchpad is working in Linux, but not in Windows. I've installed the two drivers listed at https://github.com/system76/windows-drivers#gazelle-gaze16.

I noticed this device in the Device Manager that looks like it could be related:
[image]

AnthonyMFL, 2021-08-15:
Same problem

Seibz, 2021-08-19:
In testing the Gazelle model gaze16, the touchpad also does not work in Fedora 34 with all patches applied, on kernel 5.13.10-200 and with the System76 drivers installed from https://copr.fedorainfracloud.org/coprs/szydell/system76/
[system76-logs.zip](https://github.com/system76/firmware-open/files/7017641/system76-logs.zip)

mlandry1, 2021-08-28:
same problem here in Windows.

closed 2021-09-14 by jackpot51

JCDetwiler, 2021-09-19:
Having the same problem in Arch Linux on gaze16-3050.

jackpot51, 2021-09-19:
We will release firmware fixing this soon.

dj911ice, 2021-12-27:
The firmware update that was pushed fixed the issue! Christmas really came this year!