system76/firmware-open #227: UEFI menu timeout is not respected

pspacek, 2021-08-17:
Versions affected: 2021-04-07_236914e and also 2021-07-20_93c2809

Default EFI timeout is 2 seconds:
```
# efibootmgr 
BootCurrent: 0001
Timeout: 2 seconds
```

Changing timeout using efibootmgr command seems to succeed:
```
# efibootmgr --timeout 1
BootCurrent: 0001
Timeout: 1 seconds
```

... but the value seems to be ignored on boot, and is reset to 2 seconds:
```
# efibootmgr 
BootCurrent: 0001
Timeout: 2 seconds
```

crawfxrd, 2021-11-10:
Looks like `Timeout` is useless in edk2 because it overwrites it with `PcdPlatformBootTimeOut` on every boot.

jackpot51, 2021-11-10:
Our current edk2 platform does not use the Timeout variable, as @crawfxrd mentioned. We might think about fixing this.

pspacek, 2022-01-10:
It just occurred to me that easiest way out might be _removing_ the variable.

crawfxrd, 2024-12-14:
Found out you can connect PCD entries to UEFI variables.

- https://tianocore-docs.github.io/edk2-DscSpecification/release-1.28/2_dsc_overview/29_pcd_sections.html#2932-pcdsdynamichii

PR #594 (2025-01-07): edk2: Enable boot timeout control via "Timeout"
Resolves: #227
closed 2025-01-07 by jackpot51