system76/firmware-open #230: Fails to boot with USB keyboard/mouse plugged in.

merpnderp, 2021-08-23:
My gaze16 fails to get past the System76 screen when my CyberpowerPC usb keyboard or mouse are plugged into the USB ports. I never see the text to hit esc or enter, just the System76 in big letters and then it hangs. If I unplug them, it boots correctly.

leviport, 2021-08-23:
I'm not seeing this behavior on our lab machines. Can you give us the model numbers of your mouse and keyboard that are causing this to happen? Could you also tell us which GPU your gaze16 has, and whether it's a 17" or a 15" display?

merpnderp, 2021-08-24:
The laptop has the 3060 and it is a 15".

After testing again, it is only the keyboard that keeps the boot loader from finishing. The only marking on the keyboard is "CYBERPOWERPC Gaming Keyboard".