system76/firmware-open #233: gaze16 - Unidentified devices under Windows

mlandry1, 2021-08-28:
I've installed the two drivers listed at https://github.com/system76/windows-drivers#gazelle-gaze16.

But two devices still appears as unknown in Windows 10 Pro V20H2 Build 19042.1165.

[image]
[image]

jacobgkau, 2021-08-30:
Some unidentified devices in Device Manager are expected when using Windows on Open Firmware machines. There is only an issue if functionality is not working. (For gaze16, we do currently have https://github.com/system76/firmware-open/issues/226 open to track the touchpad not working; even once that is fixed, I would still expect the unidentified ACPI items to show up in Device Manager.)

crawfxrd, 2021-11-10:
- INTC1026: Tiger Lake IPC Controller device (`coreboot/src/soc/intel/tigerlake/pmc.c`)
- 17761776: System76 EC ACPI device (`coreboot/src/ec/system76/ec/acpi/s76.asl`)

crawfxrd, 2023-07-28:
Issue is fixed. Track #340 to see if it has been released.

closed 2023-07-28 by crawfxrd