system76/firmware-open #235: Thunderbolt does not work on kernel 5.13

crawfxrd, 2021-08-31:
Tested on darp7.

```
$ uname  -r
5.13.0-7614-generic
```

```
modprobe -r thunderbolt
echo file dd.c +p > /sys/kernel/debug/dynamic_debug/control
modprobe thunderbolt
dmesg
```

```
bus: 'pci': driver_probe_device: matched device 0000:00:0d.2 with driver thunderbolt
bus: 'pci': really_probe: probing driver thunderbolt with device 0000:00:0d.2
thunderbolt: probe of 0000:00:0d.2 failed with error -110
```

This does not happen on kernel 5.11.

This behavior was introduced by ACPI 6.4 changes to `_OSC`.
Ref: pop-os/linux#61

crawfxrd, 2021-09-02:
Timeout occurs when attempting to read the config space in `tb_cfg_get_upstream_port()`.

crawfxrd, 2021-09-08:
We could fix up and apply [CB:42768](https://review.coreboot.org/c/coreboot/+/42768) to force FW CM.

nkalkhof, 2021-09-13:
will CB:42768 enable TBT4 on the lemp10 for 5.13+ kernels?

closed 2021-12-21 by crawfxrd