system76/firmware-open #236: oryp6 keyboard sometimes not activating on post after multiple hard resets while plugged in

JoonasC, 2021-09-07:
Sometimes after hard resetting multiple times (long pressing on power button) during boot, the keyboard will not be activated on post until the laptop is unplugged.

crawfxrd, 2021-09-08:
How frequently are you able to reproduce this? Does the keyboard not work at all, even when booted to the OS? Are there errors/warnings in dmesg related to the keyboard?

JoonasC, 2021-09-08:
I'm unable to test whether it works when the OS is fully booted, because of disk encryption, but since it doesn't work during the whole boot process i think it's safe to say that it wouldn't work

JoonasC, 2021-09-08:
This isn't a big problem though, as it only happens when the laptop is hard reset multiple times during boot (before and after POST), and it starts working again when the laptop is unplugged. It's just an interesting corner case that I thought I'd report.

JoonasC, 2021-09-08:
I also have an external keyboard and display plugged in, dunno if that could have something to do with it.

crawfxrd, 2021-09-08:
Are you on firmware version `2021-07-20_93c2809`? (Can check with `dmidecode -t bios` or Firmware panel in Settings.)

JoonasC, 2021-09-08:
Nope

JoonasC, 2021-09-08:
Running 2021-03-16_7c3e89e

JoonasC, 2021-09-08:
Firmware manageri doesn't show new updates

crawfxrd, 2021-09-08:
A bug was recently fixed in system76-firmware that was preventing updates. You can delete `/var/cache/system76-firmware-daemon/tail` and check then Firmware panel in Settings to get the update.

JoonasC, 2021-09-08:
I'm running ubuntu, and followed the https://support.system76.com/articles/install-ubuntu guide

JoonasC, 2021-09-08:
Maybe I did something wrong?

JoonasC, 2021-09-08:
OK, nvm. It's showing updates now

JoonasC, 2021-09-20:
The latest firmware upgrade seems to have fixed the issue, thanks.

closed 2021-09-20 by JoonasC