system76/firmware-open #237: gaze16: Power-off under load when booted with AC adapter unplugged

jacobgkau, 2021-09-07:
On gaze16-3050 and gaze16-3060 (both 15" models), if I boot with the AC adapter unplugged, the system powers off a few seconds after starting a stress command (seems like overdraw):
```
stress -c 4 -i 4 -m 4 -d 4
```

There are two workarounds:
- Boot with the AC adapter plugged in, then unplug before stressing.
- Boot on battery and select the `Battery Life` power profile before stressing.

A customer has seen this while compiling large `go` projects and when running `yarn install` on projects with many dependencies. The customer observed that the CPU frequency is limited to 2300MHz when booted unplugged and set to battery mode, and 1400MHz when booted plugged in and subsequently unplugged, while it goes higher when booted unplugged and set to balanced mode.

jackpot51, 2021-09-07:
Sounds to me like booting on battery makes the PECI command that sets PL4 run too early.

jackpot51, 2021-09-07:
@crawfxrd do you have a model? I wonder if you can see failures to set PL4 if you boot on battery with debugging.

crawfxrd, 2021-09-07:
We don't specify a PL4 value, so coreboot doesn't program the MSR for it.

jackpot51, 2021-09-07:
@crawfxrd this is in EC code, it uses PECI to dynamically alter PL4.

crawfxrd, 2021-09-07:
Also, I don't see this happen on my gaze16-3050. I've had stress running while on battery for 2+ minutes now.

jackpot51, 2021-09-07:
I think this code might be failing in some cases: https://github.com/system76/ec/blob/f55e49300130dbd945fe66c3970718825957b1ff/src/board/system76/common/power.c#L361

It would output a line starting with `power_peci_limit failed` or `power_peci_limit unknown response` if this is the case

crawfxrd, 2021-09-07:
Power on battery:
```
power_peci_limit 28 = 0
power_peci_limit unknown response: 0x00
```

Power on AC:
```
power_peci_limit 180 = 0
power_peci_limit unknown response: 0x00
```

jackpot51, 2021-09-07:
Interesting. Are there any more messages? It ought to retry 16 times. Does that just repeat 16 times?

jackpot51, 2021-09-07:
I think https://github.com/system76/ec/pull/240 might let it retry setting the setting for longer, until it succeeds. There could be side-effects though

crawfxrd, 2021-09-07:
Yes, it retries and fails all 16 tries.

After booting to OS, plug/unplug does return success for writing the config.

leviport, 2021-10-05:
https://github.com/system76/ec/pull/240 was merged in the latest gaze16 firmware release. I think we can close this now.

closed 2021-10-05 by leviport