system76/firmware-open #246: galp5 - 2021-09-20 FW build - boot entries disappearing.

ZeddieXX, 2021-09-21:
I have a GALP5 (iGPU) and on both the 2021-07-20 and 2021-09-20 firmware, the boot entries seem to disappear when I perform a cold boot.  I have Pop!_OS 21.04 and Windows 10/11 installed and I perform dual-boot.  I also use an Akitio Node TB3 eGPU with an RTX 3070 Ti and USB mouse.  It may be related.

crawfxrd, 2021-09-21:
Does it happen reliably?

Verify the SMMSTORE is being cleared:

```
make -C coreboot/util/cbmem
sudo ./coreboot/util/cbmem/cbmem -c | grep "CMOS reset"
```

If you can, after every boot (need it before and after boot entries disappear), dump the CMOS to see what values changed:

```
make -C coreboot/util/nvramtool
sudo ./coreboot/util/nvramtool -a
sudo ./coreboot/util/nvramtool -x
```

stnley, 2021-10-03:
Just wanted to +1. I experience the same phenomenon on both FW versions. The only OS which has stayed reliably is Ubuntu 20.04. I don't have the expertise to debug but I'll try the steps provided by @crawfxrd

ZeddieXX, 2021-10-16:
Sorry, I just don't have that much time to troubleshoot this issue, but today I had some time to look into this.  Very first line gave me an error:
zeddie@galp5:~/temp/firmware-open$ make -C coreboot/util/cbmem
make: *** coreboot/util/cbmem: No such file or directory.  Stop.

Sure enough, there is nothing inside the coreboot directory.  I did a new git clone https://github.com/system76/firmware-open/

ZeddieXX, 2021-10-18:
I'm actually trying this:
https://support.system76.com/articles/open-firmware-smmstore

I removed the NVME drive and flashed this firmware via a USB drive.  I then updated to the current available firmware (2021-07-20).  I'm going to reinstall the OSes again and see if this fixes the issue.

closed 2021-12-21 by crawfxrd