system76/firmware-open #250: Fan always starting and stopping

CourchesneA, 2021-09-29:
I believe this has started happening only after the firmware update last week "2021-07-20_93c2809".

According to `sensors`, the CPU and GPU temp of my oryp7 is oscillating between 54 and 55 °C. Everytime the 55 is reached (once every ~5seconds or so), the fans start and emit a really annoying noise. The noise itself isn't that bad, but the constant restart makes it hard to concentrate on anything.

Is there anyway to start the fans on an average of last temperatures, or just make it such that the fans are always on ?

Philip-Scott, 2021-10-01:
Also seeing this myself on an oryp6. Temperatures are oscillating around 50° (even making it too hot to have on a lap), and then the fans kick off dropping the temperature a few degrees and ramping up again every few seconds under a light load. 

## Fans off
[image]

## Fans on
[image]

im7FFY, 2021-11-02:
Got the same problem on my lemp9 since a while now, did not monitor temps while it occurred though.

Since the last firmware update (2021-07-20_93c2809) the fan curve seems agressive as hell.
My laptop now from time to time starts to have those crazy moments of spinning on at ~= 7000RMP then stops, then spin again etc...

This is really annoying as it kinda sounds like a biker on a motorcycle trying to impress the chicks around.
Sometimes those acceleration are even less than a second which sounds really weird.

CourchesneA, 2022-01-03:
I did some testing with different options and fan point value, but I was not able to find a good setup, there will always be oscillation between fan on and fan off. I could make it more bearable by adding `FAN_POINT(35,25)` to keep the fan always on at minimum speed, which is way less annoying.

Would love to see a better solution if there are other ideas

jamuir, 2022-08-16:
I am also seeing this (or rather *hearing* this) on my Oryx Pro 6.

It is very annoying to hear the hum and rattle of the CPU and GPU fans constantly switching on and off.

I use `watch sensors` to observe the fans switch on and off -- here is what I see:

```
# fans off
system76_acpi-acpi-0
Adapter: ACPI interface
CPU fan:        0 RPM
GPU fan:        0 RPM
CPU temp:     +52.0°C
GPU temp:     +54.0°C
```
```
# fans on
system76_acpi-acpi-0
Adapter: ACPI interface
CPU fan:     1501 RPM
GPU fan:     1862 RPM
CPU temp:     +52.0°C
GPU temp:     +54.0°C
```
I like the idea of keeping the fans running at some minimum speed to stop the on-off noise.

Is there some work-around I can apply now to stop the noise?

Destreyf, 2022-09-20:
I am having a very similar issue, and like @jamuir I can see my fans going from 0 RPM to 2600-2800 RPM.

although my temps jump at ~58-60c, which is IMO too hot for a machine that's sitting idle.

bartlebee13, 2022-10-14:
I am also experiencing this issue on an Oryx Pro 6 Pop!_OS 22.04. Nvidia Graphics Mode w/ 2 external monitors and only vscode, chrome, discord, and slack running.

I noticed when I switch the power profile to `Balanced` from `High Performance` the fans aren't as loud but this problem has been around for a while it seems.

jorgegonzalez, 2022-12-21:
Bumping for the same issue

markwylde, 2023-01-24:
> I like the idea of keeping the fans running at some minimum speed to stop the on-off noise.
Is there a way to do this? I know we can keep the fans running at max by pressing `Ctrl` + `Fn` + `1`. But would there be a way to. have tiers, so you press  `Ctrl` + `Fn` + `1` once, at it starts the fans at 100, then 500, 1000, 2000, 3000, 5000?

Is there a `/sys/class/...` file we can use to set the minimum fan speed?

CourchesneA, 2023-01-24:
It's a bit anoying having to rebuild when a new firmware is announced, but it's not too long. For me here are the steps:
* Clone the repo
* Modify the [file](https://github.com/system76/ec/blob/2a8befc195cd821232d2934593d29337e4eddb3e/src/board/system76/oryp7/board.mk) to add a new fan curve point
* Build the firmware
    *  `cd ec`
    * `./scripts/deps.sh`
    * `source $HOME/.cargo/env`
    * `make BOARD=system76/<model>`
* flash
    * `make BOARD=<vendor>/<model> flash_internal`

ddetton, 2023-08-10:
I just tried this with my new oryp11 and the flash step failed.  Any ideas how to fix this?

make BOARD=system76/oryp11 flash_internal

```
dean@oryx-pro:~/repos/system76/firmware-open/ec$ make BOARD=system76/oryp11 flash_internal
warning: unused manifest key: registries
warning: use of deprecated unit variant `hidapi::HidError::OpenHidDeviceError`
   --> src/main.rs:431:43
    |
431 |                     Err(hidapi::HidError::OpenHidDeviceError.into())
    |                                           ^^^^^^^^^^^^^^^^^^
    |
    = note: `#[warn(deprecated)]` on by default

warning: `system76_ectool` (bin "system76_ectool") generated 1 warning
    Finished release [optimized] target(s) in 0.00s
file board: Ok("system76/oryp11")
file version: Ok("2023-08-02_13dd6a1-dirty")
ec board: Ok("system76/oryp11")
ec version: Ok("2023-06-22_e5c3632")
Waiting 5 seconds for all keys to be released
Sync
Result: Err(Protocol(1))
Sync
System will shut off in 5 seconds
Sync
failed to flash 'build/system76/oryp11/2023-08-02_13dd6a1-dirty/ec.rom': Protocol(1)
make: *** [src/board/system76/common/common.mk:117: flash_internal] Error 1
```

jacobgkau, 2023-08-10:
I just assisted @ddetton in Mattermost. It's now necessary to run these commands from the `ec` directory before EC can be flashed:

```
./scripts/ectool.sh security unlock
sudo systemctl reboot # Or reboot your preferred way.
```

See https://github.com/system76/firmware-open/pull/159. After that, the `flash_internal` command will work again.

proffalken, 2023-10-26:
FWIW, I'm still seeing the protocol error on my oryp11 *after* running the command from @jacobgkau above to unlock the security mode.

After running `security unlock` and rebooting, I get prompted to type in a passcode for flashing the firmware, but then it just reboots into Pop!_OS and when I re-run the `make flash_internal` command I get the following output:

```
warning: unused manifest key: registries
warning: use of deprecated unit variant `hidapi::HidError::OpenHidDeviceError`
   --> src/main.rs:431:43
    |
431 |                     Err(hidapi::HidError::OpenHidDeviceError.into())
    |                                           ^^^^^^^^^^^^^^^^^^
    |
    = note: `#[warn(deprecated)]` on by default

warning: `system76_ectool` (bin "system76_ectool") generated 1 warning
    Finished release [optimized] target(s) in 0.00s
file board: Ok("system76/oryp11")
file version: Ok("2023-09-07_181d4c5-dirty")
ec board: Ok("system76/oryp11")
ec version: Ok("2023-09-08_42bf7a6")
Waiting 5 seconds for all keys to be released
Sync
Result: Err(Protocol(1))
Sync
System will shut off in 5 seconds
Sync
failed to flash 'build/system76/oryp11/2023-09-07_181d4c5-dirty/ec.rom': Protocol(1)
make: *** [src/board/system76/common/common.mk:117: flash_internal] Error 1
```

My fans are still causing a significant amount of noise, and I'm now getting complaints from my co-workers!

Logic-Beach, 2024-02-24:
Is there any update to this issue? I am trying to adjust the fan curve, but am running into the same issue as @proffalken 
```
root@pop-os:/home/jarm/repos/firmware-open/ec# make BOARD=system76/serw13 flash_internal
warning: unused manifest key: registries
warning: use of deprecated unit variant `hidapi::HidError::OpenHidDeviceError`
   --> src/main.rs:431:43
    |
431 |                     Err(hidapi::HidError::OpenHidDeviceError.into())
    |                                           ^^^^^^^^^^^^^^^^^^
    |
    = note: `#[warn(deprecated)]` on by default

warning: `system76_ectool` (bin "system76_ectool") generated 1 warning
    Finished release [optimized] target(s) in 0.00s
file board: Ok("system76/serw13")
file version: Ok("2023-09-07_181d4c5-dirty")
ec board: Ok("system76/serw13")
ec version: Ok("2023-09-19_16ef69c")
Waiting 5 seconds for all keys to be released
Sync
Result: Err(Protocol(1))
Sync
System will shut off in 5 seconds
Sync
failed to flash 'build/system76/serw13/2023-09-07_181d4c5-dirty/ec.rom': Protocol(1)
make: *** [src/board/system76/common/common.mk:117: flash_internal] Error 1
```

tcollinsworth, 2024-03-02:
This needs a PID controller implementation that finds the right fan speed for the current load, or at least adjusts it in tiny increments that will be far less distracting. There is no possible way to configure it for stability with simple thresholds. Given this uses the same chip as the Arduino MEGA 2560, there is a ton of open-source resources on how to do this. And a quick search shows there is a standard [arduino PID library](https://www.arduino.cc/reference/en/libraries/pid/).

chrisbirster, 2024-03-21:
> Is there any update to this issue? I am trying to adjust the fan curve, but am running into the same issue as @proffalken
>
> ```
> root@pop-os:/home/jarm/repos/firmware-open/ec# make [...]

I'm running into the same issue as the others above on a oryp11

proffalken, 2024-03-21:
FWIW, I've found that selecting "Optimise Battery" (or words to that effect - I don't have that laptop to hand at the moment!) from the power menu seems to calm the fans nicely.

brainthinks, 2024-03-31:
The docs are linked in the README, but I didn't see anything in the form of a "Getting Started" section, so here is what I had to do to tame the obnoxious fan on my `bonw15`. I am doing this on my `bonw15` machine running Linux Mint, NOT PopOS or Ubuntu.

* clone this repo
* install the dependencies - `./scripts/install-deps.sh`
    * note that this will take a while, and for me it failed initially, because I didn't have the necessary `git config` global values set in the `/coreboot` subproject (I don't set my global config values because I use different profiles for different projects)
      ```bash
      cd coreboot
      git config user.name "Your Name"
      git config user.email "your@email.com"
      cd ..
      ```
    * after setting my name and email in the subproject, I ran the script a second time, and it worked
* exit my terminal and open a new one - this is necessary to load some PATH values or something from the `install-deps.sh` script
* OPTIONAL (note that I didn't actually have to do this, but I was prepeared to, based on the other comments in this thread) - check the ec lock status:
    * ```bash
      cd ec
      ./scripts/ectool.sh security
      ```
    * mine reported "Lock", so if I had actually needed to unlock it, I could have done it like this:
    * ```bash
      ./scripts/ectool.sh security unlock
      sudo reboot # Or reboot your preferred way.
      ```
* edit the necessary `board.mk` file. For me, that was the file at `ec/src/board/system76/bonw15/board.mk`
    * ```
# Custom fan curve
CFLAGS+=-DBOARD_HEATUP=0
CFLAGS+=-DBOARD_COOLDOWN=10
CFLAGS+=-DBOARD_FAN_POINTS="\
	FAN_POINT(30, 10), \
	FAN_POINT(35, 12), \
	FAN_POINT(40, 15), \
	FAN_POINT(45, 18), \
	FAN_POINT(50, 20), \
	FAN_POINT(55, 25), \
	FAN_POINT(60, 30), \
	FAN_POINT(65, 35), \
	FAN_POINT(70, 40), \
	FAN_POINT(75, 50), \
	FAN_POINT(80, 60), \
	FAN_POINT(85, 70), \
	FAN_POINT(90, 80), \
	FAN_POINT(95, 90), \
	FAN_POINT(99, 100) \
"
```
* ensure you are in the `firmware-open` directory (project root)
* run the build script
    * @see - https://github.com/system76/firmware-open/blob/master/docs/building.md
    * note that my build failed because I didn't have `pkg-config` installed: `sudo apt install pkg-config`
    * then I got an error about libcrypto:
    ```bash
    sudo apt-get install libssl-dev
    sudo updatedb
    ```
    * e.g. `./scripts/build.sh bonw15`
* plug in the AC power adapter
* run the flash script
    * @see https://github.com/system76/firmware-open/blob/master/docs/flashing.md
    * ./scripts/flash.sh bonw15
* it will prompt you to reboot, e.g. `sudo reboot`
* follow all of the upcoming prompts, which is basically to hit enter at various times and to type in a security code
* reminder that when you restart for the first time with the new firmware, your screen will stay black for about 30 seconds - this is normal

Those steps worked for me on my `bonw15` running Linux Mint. I hope this helps someone.

I'll come back to edit this post with updated fan values if I find these unsatisfactory.

brainthinks, 2024-04-02:
After a few days with my unedited configuration above, the fan noise is now better, but still obnoxious. I had originally dismissed some of the comments here saying the configuring the power settings helps, because I don't want my performance being throttled. However, I have been using the laptop without being plugged in for about the last 30 min, and the fan has been completely silent this whole time, though my sensor reading indicates 2k rpm. I'm currently not doing development, but even at idle when plugged in, the fan will turn on.

My guess is that if you're using something like a bonobo, you bought it specifically for the CPU (and GPU) horsepower. However, if you don't have crazy CPU expectations or requirements, maybe this is a viable strategy. I work remote most of the time, so the obnoxious fan only bothers me. When I'm near my coworkers, however, this seems to be worth testing.

To try to get a casual idea of what I find tolerable, I installed the "Sensors Monitor" applet - I'm sure there's something comparable in your distro. It took a bit of time to get configured, but it's been invaluable. For instance, I now know that these are the current sensor readings, as i sit here in blissful silence:

[image]

notably absent from the applet right now is the GPU temp - apparently when on battery, the profile disables the discrete GPU in favor of the integrated one (either that or it's because I don't have any external monitors plugged in). 

A fan reading of >2k rpm doesn't seem that much lower than idle and plugged in, and I can't even hear it. This makes me wonder if the GPU fan specifically is the culprit.

crawfxrd, 2024-05-13:
Temperature hysteresis for fan-off would likely be sufficient for this. Have it cool like 2°C lower than first point before turning off. (Older ITE EC chips apparently had this functionality when "smart fan mode" was an option.)

jbaribeault, 2024-05-31:
> After a few days with my unedited configuration above, the fan noise is now better, but still obnoxious. I had originally dismissed some of the comments here saying the configuring the power settings helps, because I don't want my performance being throttled. However, I have been using the laptop without being plugged in for about the last 30 min, and the fan has been completely silent this whole time, though my sensor reading indicates 2k rpm. I'm currently not doing development, but even at idle when plugged in, the fan will turn on.
> 
> My guess is that if you're using something like a bonobo, you bought it specifically for the CPU (and GPU) horsepower. However, if you don't have crazy CPU expectations or requirements, maybe this is a viable strategy. I work remote most of the time, so the obnoxious fan only bothers me. When I'm near my coworkers, however, this seems to be worth testing.
> 
> To try to get a casual idea of what I find tolerable, I installed the "Sensors Monitor" applet - I'm sure there's something comparable in your distro. It took a bit of time to get configured, but it's been invaluable. For instance, I now know that these are the current sensor readings, as i sit here in blissful silence:
> 
> ![image](https://private-user-images.githubusercontent.com/2215066/318874688-bceff36f-e1e1-41ec-82df-8199110daf25.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcxMDIwNDcsIm5iZiI6MTcxNzEwMTc0NywicGF0aCI6Ii8yMjE1MDY2LzMxODg3NDY4OC1iY2VmZjM2Zi1lMWUxLTQxZWMtODJkZi04MTk5MTEwZGFmMjUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUzMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MzBUMjA0MjI3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MmViNGJiMDI5N2VlZjc2OWQ5YzhjODMyOGVlY2FiNTI5N2U4ZjViOTFjM2FhM2MyZWVmZmUyMTA4YTAyM2YxYiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.CJnuI0BDDurUk645dxiYtdzykwMcVZyADUb7Oyx6I5E)
> 
> notably absent from the applet right now is the GPU temp - apparently when on battery, the profile disables the discrete GPU in favor of the integrated one (either that or it's because I don't have any external monitors plugged in).
> 
> A fan reading of >2k rpm doesn't seem that much lower than idle and plugged in, and I can't even hear it. This makes me wonder if the GPU fan specifically is the culprit.
> 
> Edit 1 - so it turns out I didn't read the rest of the board file. Under the CPU fan curve is indeed the GPU fan curve, which I left stock. I'll update my firmware later this week and report back. Here are the default values for `bonw15`:
> 
> ```
> # Enable DGPU support
> CFLAGS+=-DHAVE_DGPU=1
> CFLAGS+=-DBOARD_DGPU_HEATUP=5
> CFLAGS+=-DBOARD_DGPU_COOLDOWN=20
> CFLAGS+=-DBOARD_DGPU_FAN_POINTS="\
> 	FAN_POINT(60, 40), \
> 	FAN_POINT(65, 60), \
> 	FAN_POINT(70, 75), \
> 	FAN_POINT(75, 90), \
> 	FAN_POINT(80, 100) \
> "
> ```

Hey how did you manage to get the RPMs for the fan? I can't find a sensor that shows that data at all on my BONW15

tcollinsworth, 2024-06-01:
This will show all the sensors and CPU and GPU fan speeds.

# temp sensors
#
https://www.cyberciti.biz/faq/how-to-check-cpu-temperature-on-ubuntu-linux/
gnome-terminal --tab --title="Temp Sensors" -- bash -c "watch -n 0.5
sensors"

alejandrofierro, 2024-06-26:
> After a few days with my unedited configuration above, the fan noise is now better, but still obnoxious. I had originally dismissed some of the comments here saying the configuring the power [...]
Hi @brainthinks were you able to change the GPU fan curve? Is it better now? Based on which commit did you compile the firmware``?  I just received my bonw15 a week ago and this fan noise is making me want to throw it out the window, my firmware is at `2023-10-20_2e4e34b`

brainthinks, 2024-08-05:
ok, well, I really tried to give this thing a chance, but this fan is driving me insane. I'll be trading in my bonobo soon. I think this is only partly the fault of System 76 - surely they could have used a quieter fan.... But I suspect that the rest of the blame goes to Intel for these 13th and 14th gen chips. 

The final verdict is that I cannot get the fan to gently ramp to higher speeds. The chip takes in a ton of juice to get the advertised performance, and as a result, the temp will jump 40 degrees C in less than a second. I don't think there is a configuration that can account for this.

@alejandrofierro here is the commit I am using for `ec`:

```
commit 38d4666a9007b6d36e344bbc36109876e519028e (HEAD, origin/master, origin/HEAD, master)
Author: Tim Crawford <tcrawford@system76.com>
Date:   Thu Mar 21 10:47:29 2024 -0600

    oryp12: Use OZ26786 charger
    
    Ref: Schematics sheet 58 ("AC In Charger")
    Signed-off-by: Tim Crawford <tcrawford@system76.com>
```

and here is the commit i'm using for `firmware-open`:

```
commit 2c6913bc0abb7906cfd134122a077134a82fba74 (HEAD -> master, origin/master, origin/HEAD)
Author: Jeremy Soller <jackpot51@gmail.com>
Date:   Mon Mar 25 14:30:56 2024 -0600

    lemp13: Update coreboot to fix power LED flicker
```

I did also update my GPU fan settings, but as I stated, it isn't enough to tame the fan. Here they are if you're interested:

```
# Enable DGPU support
CFLAGS+=-DHAVE_DGPU=1
CFLAGS+=-DBOARD_DGPU_HEATUP=5
CFLAGS+=-DBOARD_DGPU_COOLDOWN=20
CFLAGS+=-DBOARD_DGPU_FAN_POINTS="\
	FAN_POINT(30, 10), \
	FAN_POINT(35, 12), \
	FAN_POINT(40, 15), \
	FAN_POINT(45, 18), \
	FAN_POINT(50, 20), \
	FAN_POINT(55, 25), \
	FAN_POINT(60, 30), \
	FAN_POINT(65, 35), \
	FAN_POINT(70, 40), \
	FAN_POINT(75, 50), \
	FAN_POINT(80, 60), \
	FAN_POINT(85, 70), \
	FAN_POINT(90, 80), \
	FAN_POINT(95, 90), \
	FAN_POINT(99, 100) \
"
```

I wish I had a working solution for this, since this laptop has so much power... The only way I could get the fans to stop being obnoxious was to disconnect from the AC power. I'm sure there's some power management settings you could adjust to simulate this while plugged in, but to me that defeats the purpose, since you have to starve the cpu of power to get it to not shoot up 40 deg.

brainthinks, 2024-08-05:
in case there is any confusion about what my actual updated firmware is, here are the full contents of `firmware-open/ec/src/board/system76/bonw15/board.mk`

```
# SPDX-License-Identifier: GPL-3.0-only

board-y += board.c
board-y += gpio.c

EC=ite
CONFIG_EC_ITE_IT5570E=y
CONFIG_EC_FLASH_SIZE_128K = y

# Enable eSPI
CONFIG_BUS_ESPI=y

# Enable firmware security
CONFIG_SECURITY=y

# Include keyboard
KEYBOARD=15in_102

# Set keyboard LED mechanism
CONFIG_HAVE_KBLED = y
KBLED=bonw14

# Set discrete GPU I2C bus
CFLAGS+=-DI2C_DGPU=I2C_1

# Set battery I2C bus
CFLAGS+=-DI2C_SMBUS=I2C_4

# Set touchpad PS2 bus
CFLAGS+=-DPS2_TOUCHPAD=PS2_3

# Set USB-PD I2C bus
CFLAGS+=-DI2C_USBPD=I2C_1

# Set smart charger parameters
# XXX: PRS1 and PRS2 are in parallel for adapter Rsense?
CHARGER=oz26786
CFLAGS+=\
	-DCHARGER_ADAPTER_RSENSE=5 \
	-DCHARGER_BATTERY_RSENSE=5 \
	-DCHARGER_CHARGE_CURRENT=3072 \
	-DCHARGER_CHARGE_VOLTAGE=17400 \
	-DCHARGER_INPUT_CURRENT=16920

# Set USB-PD parameters
USBPD=tps65987

# Set CPU power limits in watts
CFLAGS+=\
	-DPOWER_LIMIT_AC=330 \
	-DPOWER_LIMIT_DC=55

# Custom fan curve
CFLAGS+=-DBOARD_HEATUP=0
CFLAGS+=-DBOARD_COOLDOWN=10
CFLAGS+=-DBOARD_FAN_POINTS="\
	FAN_POINT(30, 10), \
	FAN_POINT(35, 12), \
	FAN_POINT(40, 15), \
	FAN_POINT(45, 18), \
	FAN_POINT(50, 20), \
	FAN_POINT(55, 25), \
	FAN_POINT(60, 30), \
	FAN_POINT(65, 35), \
	FAN_POINT(70, 40), \
	FAN_POINT(75, 50), \
	FAN_POINT(80, 60), \
	FAN_POINT(85, 70), \
	FAN_POINT(90, 80), \
	FAN_POINT(95, 90), \
	FAN_POINT(99, 100) \
"

# Enable DGPU support
CFLAGS+=-DHAVE_DGPU=1
CFLAGS+=-DBOARD_DGPU_HEATUP=5
CFLAGS+=-DBOARD_DGPU_COOLDOWN=20
CFLAGS+=-DBOARD_DGPU_FAN_POINTS="\
	FAN_POINT(30, 10), \
	FAN_POINT(35, 12), \
	FAN_POINT(40, 15), \
	FAN_POINT(45, 18), \
	FAN_POINT(50, 20), \
	FAN_POINT(55, 25), \
	FAN_POINT(60, 30), \
	FAN_POINT(65, 35), \
	FAN_POINT(70, 40), \
	FAN_POINT(75, 50), \
	FAN_POINT(80, 60), \
	FAN_POINT(85, 70), \
	FAN_POINT(90, 80), \
	FAN_POINT(95, 90), \
	FAN_POINT(99, 100) \
"

# Add system76 common code
include src/board/system76/common/common.mk
```

Warning though, my file is from a months-old commit at this point, so I suggest you only use it for reference and only use the CPU and GPU fan configuration. If you copy and paste these contents directly, you may be missing some configuration that is more up to date.

ddetton, 2024-08-05:
I found any fan speed above 75% to be unbearable so this is what I did on my oryp11:

```
FLAGS+=-DBOARD_FAN_POINTS="\
	FAN_POINT(45, 25), \
	FAN_POINT(50, 25), \
	FAN_POINT(55, 25), \
	FAN_POINT(60, 25), \
	FAN_POINT(65, 25), \
	FAN_POINT(70, 30), \
	FAN_POINT(75, 35), \
	FAN_POINT(80, 40), \
	FAN_POINT(85, 45), \
	FAN_POINT(90, 50), \
	FAN_POINT(95, 75)  \
	```

crawfxrd, 2024-08-06:
Working on it. Current progress in https://github.com/system76/ec/pull/495.

crawfxrd, 2024-12-30:
system76/ec#510 will address the issue with the new fan control mechanism (system76/ec#496).

lanerussell, 2025-04-25:
Checking https://github.com/system76/firmware-open/blob/master/CHANGELOG.md, it appears the latest firmware release is 2025-02-04. However, when I attempt to update my lemp13, it seems to stay on 2024-07-08. I'd like to get the upgrade including the fan speed changes mentioned by @crawfxrd in December; what process should I follow?

bartlebee13, 2025-04-29:
@lanerussell I used [this video](https://www.youtube.com/watch?v=FEYUBlftpi4) to modify the fan curve myself because I also wasn't getting firmware updates on my oryx pro (oryp6). 

You can check [this link](https://github.com/system76/firmware-open/blob/master/docs/features.md) to see the firmware feature support docs. The manual for your lemp13 is located [here](https://tech-docs.system76.com/models/lemp13/README.html).

This way you can modify the fan curve yourself while you wait for the firmware update.

lanerussell, 2025-05-02:
Thanks for the information @bartlebee13, but that is a little deeper than I'm comfortable going. In the end, I did decided to return my lemp13 due to concerns over thermal management. I believe most of the blame for that lies with Intel's chip design, but the light thermal mass and fan curves of the lemp13 definitely draw greater attention to the problem in my case.

I appreciate the work System76 is doing in the space and hope that it continues to mature in the coming years. All the best!

MehmetTopsakal, 2025-05-02:
Good decision nerussell!

I wish I had a chance to return my lemp13. 

A laptop that I have to run at 700 MHz due to annoying fan that spins at 10k rpm every 30 seconds otherwise... What a big disappointment!....

chirpcel, 2025-06-21:
Same on darp11-b :/
Fan constantly starting and stopping on AC power. FW Version: 04c4e20

MehmetTopsakal, 2025-06-21:
Return your laptop if you have a chance!

farfromrefug, 2025-09-15:
@crawfxrd I would like to use your PR for my gazelle 16. But my firmware is "stuck" on 2023 version. Does not propose a newer version. Is it safe to build the firmware from after your PR and use it on my gazelle 16?