system76/firmware-open #255: Oryp6 can't boot after flashing the ec

FengXi7420, 2021-10-04:


FengXi7420, 2021-10-04:
Here's some debug info:
predator@predator-Predator-PH717-72:~/ec$ make BOARD=system76/oryp6 console_external
sudo test -c "/dev/serial/by-id/usb-Arduino__www.arduino.cc__0042_85035303038351308102-if00"
sleep 1 && echo C | sudo tee "/dev/serial/by-id/usb-Arduino__www.arduino.cc__0042_85035303038351308102-if00" &
sudo tio -b 1000000 -m INLCRNL -t "/dev/serial/by-id/usb-Arduino__www.arduino.cc__0042_85035303038351308102-if00"
[tio 15:21:30] tio v1.32
[tio 15:21:30] Press ctrl-t q to quit
[tio 15:21:30] Connected
C
[15:21:31] Entering console mode
[15:21:31]
[15:21:37] System76 EC board 'system76/oryp6', version '2021-09-20_d3ee037-dirty'
[15:21:37] 00: power_on_ds5
[15:21:37] POWER_STATE_DS5
[15:21:37] Power adapter plugged in
[15:21:37] Charger enabled
[15:21:37] Battery:
[15:21:37] Temperature: ERROR 0025
[15:21:37] Voltage: ERROR 0025
[15:21:37] Current: ERROR 0025
[15:21:37] Charge: ERROR 0025
[15:21:37] Status: ERROR 0025
[15:21:37] Charger (bq24780s):
[15:21:37] ChargeOption0: 8108
[15:21:37] ChargeOption1: C210
[15:21:37] ChargeOption2: 0384
[15:21:37] ChargeOption3: 1A40
[15:21:37] ChargeCurrent: 0600
[15:21:37] ChargeVoltage: 32F0
[15:21:37] DishargeCurrent: 1800
[15:21:37] InputCurrent: 1000
[15:21:37] ProchotOption0: 4A54
[15:21:37] ProchotOption1: 8120
[15:21:37] ProchotStatus: 0020
[15:21:37] Lid open
[15:21:52] 81: Power switch press
[15:21:52] 81: power_on_s5
[15:21:52] VA_EC_EN = true
[15:21:52] DD_ON = true
[15:21:52] EC_RSMRST_N = true
[15:21:52] EC_EN = true
[15:21:52] POWER_STATE_S5
[15:21:57] 81: power_off_s5
[15:21:57] PM_PWROK = false
[15:21:57] EC_EN = false
[15:21:57] EC_RSMRST_N = false
[15:21:57] DD_ON = false
[15:21:57] VA_EC_EN = false
[15:21:57] POWER_STATE_DS5
[15:21:57] 84: Power switch release
[15:22:32] 30: Power switch press
[15:22:32] 30: power_on_s5
[15:22:32] VA_EC_EN = true
[15:22:32] DD_ON = true
[15:22:32] EC_RSMRST_N = true
[15:22:32] EC_EN = true
[15:22:32] POWER_STATE_S5
[15:22:37] 30: power_off_s5
[15:22:37] PM_PWROK = false
[15:22:37] EC_EN = false
[15:22:37] EC_RSMRST_N = false
[15:22:37] DD_ON = false
[15:22:37] VA_EC_EN = false
[15:22:37] POWER_STATE_DS5
[15:22:37] 33: Power switch release
[15:22:49] 4B: Power switch press
[15:22:49] 4B: power_on_s5
[15:22:49] VA_EC_EN = true
[15:22:49] DD_ON = true
[15:22:50] EC_RSMRST_N = true
[15:22:50] EC_EN = true
[15:22:50] POWER_STATE_S5
[15:22:55] 4B: power_off_s5
[15:22:55] PM_PWROK = false
[15:22:55] EC_EN = false
[15:22:55] EC_RSMRST_N = false
[15:22:55] DD_ON = false
[15:22:55] VA_EC_EN = false
[15:22:55] POWER_STATE_DS5
[15:22:55] 4E: Power switch release
[15:22:59] 42: Power switch press
[15:22:59] 42: power_on_s5
[15:22:59] VA_EC_EN = true
[15:22:59] DD_ON = true
[15:22:59] EC_RSMRST_N = true
[15:23:00] EC_EN = true
[15:23:00] POWER_STATE_S5
[15:23:05] 42: power_off_s5
[15:23:05] PM_PWROK = false
[15:23:05] EC_EN = false
[15:23:05] EC_RSMRST_N = false
[15:23:05] DD_ON = false
[15:23:05] VA_EC_EN = false
[15:23:05] POWER_STATE_DS5
[15:23:09] A8: Power switch release
[15:23:11] A2: Power switch press
[15:23:11] A2: power_on_s5
[15:23:11] VA_EC_EN = true
[15:23:11] DD_ON = true
[15:23:11] EC_RSMRST_N = true
[15:23:12] EC_EN = true
[15:23:12] POWER_STATE_S5
[15:23:17] A2: power_off_s5
[15:23:17] PM_PWROK = false
[15:23:17] EC_EN = false
[15:23:17] EC_RSMRST_N = false
[15:23:17] DD_ON = false
[15:23:17] VA_EC_EN = false
[15:23:17] POWER_STATE_DS5
[15:23:17] A5: Power switch release

crawfxrd, 2021-10-04:
> Can't flash the bios using ch341a at this situation

CH351A will always work to flash the BIOS if the board is working.

Verify you can flash and boot using the current release, `2021-07-20_93c2809`.

FengXi7420, 2021-10-04:
> > Can't flash the bios using ch341a at this situation
> 
> CH351A will always work to flash the BIOS if the board is working.
> 
> Verify you can flash and boot using the current release, `2021-07-20_93c2809`.

i flashed the bios again but it doesn't work. By the way, how to get the current release?

crawfxrd, 2021-10-04:
> i flashed the bios again but it doesn't work.

Did you also flash the EC?

> By the way, how to get the current release?

Current release is `2021-07-20_93c2809`, so check out `93c2809` in firmware-open and update the submodules.

jackpot51, 2021-10-04:
@FengXi7420 what firmware did you have prior to flashing, and what hardware (like GPU/RAM/SSD) does your System76 oryp6 have?

Is it a genuine System76 laptop purchased from system76.com?

FengXi7420, 2021-10-04:
> @FengXi7420 what firmware did you have prior to flashing, and what hardware (like GPU/RAM/SSD) does your System76 oryp6 have?
> 
> Is it a genuine System76 laptop purchased from system76.com?

If i falsh back to clevo's ec the computer can boot how big the system's ec should be?

jackpot51, 2021-10-04:
@FengXi7420 is it a genuine System76 laptop? Other vendors of the same Clevo model can have incompatible configurations. The EC ROM built from our open source EC project should be a little more than 32KiB in size. It is padded to fit the 128KiB internal EC ROM.

FengXi7420, 2021-10-04:
> @FengXi7420 is it a genuine System76 laptop? Other vendors of the same Clevo model can have incompatible configurations. The EC ROM built from our open source EC project should be a little more than 32KiB in size. It is padded to fit the 128KiB internal EC ROM.

It's not a genuine system76 laptop, so i can't use the system76 EC?

jackpot51, 2021-10-04:
That is correct

jackpot51, 2021-10-04:
You will need to port the board you have yourself. Schematics from whoever sold you the board will be required

closed 2021-10-04 by jackpot51

FengXi7420, 2021-10-04:
> You will need to port the board you have yourself. Schematics from whoever sold you the board will be required

That means i can't just simply flash the system76 oryp6's ec, i need to do more？

FengXi7420, 2021-10-04:
i have the Schematics how can i port my board?

crawfxrd, 2021-10-04:
Start with:

- [firmware-open: Adding a new board](https://github.com/system76/firmware-open/blob/de3b0a65fa633f3c7cb05138086390463336d38f/docs/adding-a-new-board.md)
- [ec: Adding a new board](https://github.com/system76/ec/blob/9fa83ad3595ffd2a5d0355a305de5635c0b71c83/doc/adding-a-new-board.md)

FengXi7420, 2021-10-04:
Thank you i will give it a try

FengXi7420, 2021-10-05:
> Start with:
> 
> - [firmware-open: Adding a new board](https://github.com/system76/firmware-open/blob/de3b0a65fa633f3c7cb05138086390463336d38f/docs/adding-a-new-board.md)
> - [ec: Adding a new board](https://github.com/system76/ec/blob/9fa83ad3595ffd2a5d0355a305de5635c0b71c83/doc/adding-a-new-board.md)

The only difference of oryp6 and pc70dd is the bios chip the others are the same
oryp6:MX25L12872F
pc70dd:MX25L12873F

crawfxrd, 2021-10-05:
Board GPIO config is the same?

You said it boots with proprietary EC firmware. Is that coreboot BIOS firmware + proprietary EC firmware?

It sounds like EC GPIO config is wrong for S0 power on. Did you generate the GPIO config from the proprietary EC to compare?

jackpot51, 2021-10-05:
Regardless, since this is not a System76 product, we can't really provide support for it. You are on your own for porting the EC firmware.

FengXi7420, 2021-10-05:
Yeah,the board gpio is the same it can boot with coreboot+proprietary ec
I haven't try to generate the GPIO config from the proprietary ec yet
One more thing when using coreboot+proprietary ec the machine can't boot
with battery only

FengXi7420, 2021-10-05:
i'm really confused
Everything is same but why the ec won't work