system76/firmware-open #256: Oryp6 - Random bluetooth fails after installing firmware 2021-07-20_93c2809

JoonasC, 2021-10-04:
Since installing the firmware 2021-07-20_93c2809 on my Oryp6 bluetooth will sometimes randomly fail to activate on some boots, this can be fixed by rebooting the machine.

Dmesg log:
```
[   38.242172] Bluetooth: Core ver 2.22
[   38.242188] Bluetooth: HCI device and connection manager initialized
[   38.242192] Bluetooth: HCI socket layer initialized
[   38.242194] Bluetooth: L2CAP socket layer initialized
[   38.242197] Bluetooth: SCO socket layer initialized
[   38.383536] Bluetooth: hci0: Bootloader revision 0.4 build 0 week 30 2018
[   38.384516] Bluetooth: hci0: Device revision is 2
[   38.384518] Bluetooth: hci0: Secure boot is enabled
[   38.384519] Bluetooth: hci0: OTP lock is enabled
[   38.384520] Bluetooth: hci0: API lock is enabled
[   38.384521] Bluetooth: hci0: Debug lock is disabled
[   38.384522] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
[   38.386405] Bluetooth: hci0: Found device firmware: intel/ibt-19-0-4.sfi
[   38.516163] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   38.516166] Bluetooth: BNEP filters: protocol multicast
[   38.516170] Bluetooth: BNEP socket layer initialized
[   40.049327] Bluetooth: hci0: Waiting for firmware download to complete
[   40.049467] Bluetooth: hci0: Firmware loaded in 1624083 usecs
[   40.049489] Bluetooth: hci0: Waiting for device to boot
[   40.064532] Bluetooth: hci0: Device booted in 14700 usecs
[   40.064688] Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-19-0-4.ddc
[   40.066578] Bluetooth: hci0: Applying Intel DDC parameters completed
[   40.069549] Bluetooth: hci0: Firmware revision 0.0 build 191 week 21 2021
[   49.687226] Bluetooth: RFCOMM TTY layer initialized
[   49.687231] Bluetooth: RFCOMM socket layer initialized
[   49.687234] Bluetooth: RFCOMM ver 1.11
[   63.867557] Bluetooth: hci0: unexpected event for opcode 0x0500
[   65.909962] Bluetooth: hci0: urb 00000000633ce82a failed to resubmit (113)
[   76.093805] Bluetooth: hci0: Failed to read MSFT supported features (-110)
```

crawfxrd, 2021-10-04:
What kernel are you on?

JoonasC, 2021-10-04:
I'm on the `5.13.0-7614-generic` kernel

JoonasC, 2021-10-04:
If it's a kernel bug, then I guess this can be closed?

JoonasC, 2021-11-08:
This has been fixed with the latest kernel update.

closed 2021-11-08 by JoonasC