system76/firmware-open #261: lemp9 - unable to add EFI entries

garybowers, 2021-11-02:
lemp9 f/w 2021-07-20_93c2809

Trying to install any OS I am getting the error:  

`[WARN distinst:crates/chroot/src/command.rs:98] Created "/boot/efi/EFI/systemd".
[WARN distinst:crates/chroot/src/command.rs:98] Created "/boot/efi/EFI/BOOT".
[WARN distinst:crates/chroot/src/command.rs:98] Created "/boot/efi/EFI/Linux".
[WARN distinst:crates/chroot/src/command.rs:98] Copied "/usr/lib/systemd/boot/efi/systemd-bootx64.efi" to "/boot/efi/EFI/systemd/systemd-bootx64.efi".
[INFO distinst:crates/chroot/src/command.rs:108] running "chroot" "/tmp/distinst.neDn6NrlG3OU" "efibootmgr" "--create" "--disk" "/dev/nvme0n1" "--part" "1" "--write-signature" "--label" "Pop!_OS 21.04" "--loader" "\\EFI\\systemd\\systemd-bootx64.efi"
[ERROR distinst:src/installer/state.rs:37] configuring bootloader error: command failed with exit status: exit code: 5
[ERROR distinst:src/installer/mod.rs:300] errored while installing system: command failed with exit status: exit code: 5
[INFO distinst:ffi/src/installer.rs:188] Install error: command failed with exit status: exit code: 5`

Eventually getting into recovery mode, the firmware utility does not detect a machine known to it.

This seems similar to [https://github.com/system76/firmware-open/issues/145](https://github.com/system76/firmware-open/issues/145)

This machine is unusable in it's current form, will add to the already open ticket given the machine hangs even after getting a new motherboard.

jacobgkau, 2021-11-02:
> This seems similar to #145

I agree it seems similar to that issue. Does clearing the SMMSTORE using the instructions in [this support article](https://support.system76.com/articles/open-firmware-smmstore) help?

> Eventually getting into recovery mode, the firmware utility does not detect a machine known to it.

This is expected, the recovery environment is not set up to perform firmware updates (although they can be performed manually if the boot partition is manually mounted.)

garybowers, 2021-11-02:
Sorry I meant to close this, yeah I found that support article and that indeed did fix the issue.

garybowers, 2021-11-02:
Not bothering.

closed 2021-11-02 by garybowers