system76/firmware-open #263: License?

wwahammy, 2021-11-03:
First off, I love my System76 laptop and your company's commitment to open source. I was hoping you could clarify and add a license to this repo as well as all the associated repos. It's not really clear what rights we have as users. I assume you intended to put this under an open source license (minus the icky binaries which are, sadly, unavoidable). Would you be able to clarify what the license is?

crawfxrd, 2021-11-03:
All submodules should already have licenses.

jackpot51, 2021-11-03:
Please see the licenses of each component - coreboot, EDK2, and each of the apps have different open source licenses. Some components like the FSP have a freely redistributable closed source license.

PR #275 (2022-07-08): Add info and text for firmware binaries licenses
Resolves: #263

What's the license info for other binaries used?

- `fd.rom`
- `vbt.rom`
- `usb4-retimer.rom` (do we actually use this anywhere?)
- `shell.efi`
- `uecflash.efi`
- `meset.efi`
- `IntelGopDriver.efi`
closed 2022-07-08 by crawfxrd