system76/firmware-open #264: Can't build on 21.10

jacobgkau, 2021-12-13:
When attempting to build firmware from scratch on Pop!_OS 21.10, coreboot/crossgcc fails at the following line:

```
Building GCC v8.3.0 for target ... failed. Check 'build-i386-elf-GCC/build.log'.
```

Here is the `build.log` file: [build.log](https://github.com/system76/firmware-open/files/7705783/build.log)

I brought this up about a month ago and I recall @crawfxrd mentioning that using update-alternatives to set gcc-9 and gcc++-9 as the defaults _might_ get this working. This issue is to track any work required to get the `build.sh` script working again on 21.10. Ideally, any setup required would be handled in `deps.sh`.

crawfxrd, 2021-12-30:
GCC 11 is now used for the coreboot toolchain.

What would be nice to have is to [provide toolchain binaries](https://doc.coreboot.org/contributing/project_ideas.html#provide-toolchain-binaries).

jacobgkau, 2022-01-05:
I believe this was fixed by https://github.com/system76/firmware-open/pull/260. I've been building and flashing on lemp10 with 21.10.

closed 2022-01-05 by jacobgkau