system76/firmware-open #268: Support for /sys/class/dmi/id/product_uuid

iansltx, 2021-12-29:
Trying to use osquery (as part of Vanta Agent) and they need this file to exist and be populated. With open firmware, the file doesn't exist at this point, and the machine in question doesn't seem to have a proprietary firmware equivalent to work with (lemp10).

Happy to hack on this, but not sure where to start.

crawfxrd, 2021-12-29:
Why? What does Vanta use the UUID for?

Set by `smbios_system_set_uuid()` in coreboot.

iansltx, 2021-12-29:
Unique device ID for asset management, more or less. Standard compliance stuff.

iansltx, 2021-12-30:
Same problem, different vendor (also using coreboot), slightly different application: https://forums.puri.sm/t/coreboot-populating-sys-class-dmi-id-product-uuid/7506

iansltx, 2022-02-03:
Apologies for bumping this without much info, but looks like we can't use a kernel module to get around this, and I'm getting flak on allowing new hires to spec System76 machines until this gets fixed, as anything they pick is likely Open Firmware only at this point and other vendors' firmware includes the entry :/

frankk74, 2022-02-22:
This might or might not be a related issue:

https://github.com/acidanthera/bugtracker/issues/711

crawfxrd, 2022-06-08:
3mdeb's solution for Dasharo: https://review.coreboot.org/c/coreboot/+/64639

antonshmakov, 2022-08-12:
Any way the solution above is applicable to original problem? Having the exact same issue -- trying to activate Vanta Agent on Lemur Pro (lemp11)

colinbird, 2022-08-19:
Also having this issue on brand new Gazelle

allan-simon, 2023-01-31:
same issue here with the gazelle and vanta

allan-simon, 2023-01-31:
seems it's about implementing the method `smbios_system_set_uuid`

fetherolfjd, 2023-05-01:
Just ran into this as well attempting trying to install the Vanta agent on Oryx Pro.

crawfxrd, 2023-06-27:
https://github.com/system76/coreboot/pull/182 will enable support for reading it from a CBFS file. firmware-update will then need to be updated to either copy it from the current firmware, or generate it if it doesn't exist, and then inject it into the new firmware image before flashing it.

tupshin, 2024-10-03:
This is blocking me from using a system76 laptop with the Vanta agent, needed for hipaa/sox compliance. Makes it impossible to use them for corporate/startup activities

iansltx, 2024-10-03:
FWIW it's entirely possible to have an alternative remediation to "our systems all check into Vanta" and still be compliant. We got SOC2 Type II, via Vanta, with multiple System76 boxes, this way. You just have to document your controls. Same with HIPPA (disclosures apply that IANAL but Vanta should tell you the same thing).

GuySartorelli, 2025-08-25:
I also have a problem with a missing `/sys/class/dmi/id/product_uuid` file (which I need to install "Vanta Device Monitor" for compliance reasons) on Pop!OS.
I don't think I'll be able to get approval to modify my firmware.