system76/firmware-open #270: 1dc7780 Causing long boot times with Thunderbolt Dock attached

echo-84, 2022-01-05:
ory8 with wd19tb dock attached in thunderbolt mode causes a 2-3 minute boot just to see the bootloader. It's also dificult to access the firmware menu from this state. When dock falls back to USB-C mode (which happens separately) it takes the normal amount of time to boot.

Running: 2021-12-21_1dc7780-dirty

Using commit 1dc7780 as it's needed to fix issue #235

crawfxrd, 2022-01-06:
Can you provide the cbmem log after booting with it plugged in?

```
make -C coreboot/util/cbmem
sudo ./coreboot/util/cbmem/cbmem -c > ~/cbmem.txt
```

echo-84, 2022-01-10:
Some things I've noticed but haven't extensively tested yet are:

- Reboot from thunderbolt mode will work in thunderbolt again. 
- Shutdown and power on will fail to USB-C
- Shutdown unplug wait 30 seconds, plug back in, power on will use thunderbolt.

Here are both cbmem's with thunderbolt in both modes. 
[cbmem-thunderbolt-mode.txt](https://github.com/system76/firmware-open/files/7837297/cbmem-thunderbolt-mode.txt)
[cbmem-usbc-mode.txt](https://github.com/system76/firmware-open/files/7837296/cbmem-usbc-mode.txt)

crawfxrd, 2022-01-10:
Most notable thing is the resource allocator failing to find prefmem ranges large enough:

```
$ rg 'ERROR:' cbmem-thunderbolt-mode.txt 
951:  ERROR: Resource didn't fit!!!   PCI: 00:07.0 24 *  size: 0x40000000 limit: ffffffffffffffff prefmem
1006:  ERROR: Resource didn't fit!!!   PCI: 03:00.0 24 *  size: 0x20000000 limit: ffffffffffffffff prefmem
1007:  ERROR: Resource didn't fit!!!   NONE 14 *  size: 0x20000000 limit: ffffffffffffffff prefmem
1023:  ERROR: Resource didn't fit!!!   PCI: 24:04.0 24 *  size: 0x20000000 limit: ffffffffffffffff prefmem
1044:  ERROR: Resource didn't fit!!!   NONE 14 *  size: 0x20000000 limit: ffffffffffffffff prefmem
```

And it takes 2.4s longer to run the payload:

```diff
-BS: BS_PAYLOAD_LOAD run times (exec / console): 841 / 0 ms
+BS: BS_PAYLOAD_LOAD run times (exec / console): 3292 / 0 ms
```

so edk2 is probably contributing to the issue as well.

echo-84, 2022-01-19:
I've been watching the upstream commits but nothing relevant has popped out to me yet, granted I don't really know what to look for. I thought [coreboot commit 60946](https://review.coreboot.org/c/coreboot/+/60946) might help but it didn't seem to help with TB.

For now I have rolled back to the original release b9be89 and backlisted the thunderbolt module since USB-C mode works fine for my needs atm.

If there is something else to try or additional info that can be pulled I'm willing to give it a shot.

bag-builds, 2022-10-01:
Same problem, default value of 0x20000000 for this two parameters CONFIG_PCIEXP_HOTPLUG_MEM, CONFIG_PCIEXP_HOTPLUG_PREFETCH_MEM, cause long boot times with Thunderbolt Dock attached and it's impossible to use it for booting from the thunderbolt drive connected to the dock, tested with 0x800000 and work, I didn't test different value.
/models/galp5/coreboot.config
commit 74e75f
instead with the upstreamed version it doesn't freeze but I'm not able to boot from the thunderbolt drive connected to the thunderbolt dock

crawfxrd, 2022-10-01:
Something to check is using the Intel recommended values:

```diff
diff --git a/models/oryp8/coreboot.config b/models/oryp8/coreboot.config
index 4d0073f9d92d..2b8ea4f68ac1 100644
--- a/models/oryp8/coreboot.config
+++ b/models/oryp8/coreboot.config
@@ -9,8 +9,6 @@ CONFIG_IFD_BIN_PATH="$(FIRMWARE_OPEN_MODEL_DIR)/fd.rom"
 CONFIG_ME_BIN_PATH="$(FIRMWARE_OPEN_MODEL_DIR)/me.rom"
 CONFIG_PAYLOAD_ELF=y
 CONFIG_PAYLOAD_FILE="$(FIRMWARE_OPEN_UEFIPAYLOAD)"
-CONFIG_PCIEXP_HOTPLUG_MEM=0x2000000
-CONFIG_PCIEXP_HOTPLUG_PREFETCH_MEM=0x20000000
 CONFIG_POST_IO=n
 CONFIG_RUN_FSP_GOP=y
 CONFIG_SMMSTORE=y0
```

```diff
diff --git a/src/mainboard/system76/oryp8/Kconfig b/src/mainboard/system76/oryp8/Kconfig
index 2a60dd71a670..0a506a271857 100644
--- a/src/mainboard/system76/oryp8/Kconfig
+++ b/src/mainboard/system76/oryp8/Kconfig
@@ -18,12 +18,12 @@ config BOARD_SPECIFIC_OPTIONS
        select MEMORY_MAPPED_TPM
        select MAINBOARD_HAS_TPM2
        select NO_UART_ON_SUPERIO
-       select PCIEXP_HOTPLUG
        select PCIEXP_HOTPLUG_PREFETCH_MEM_BELOW_4G
        select SOC_INTEL_TIGERLAKE
        select SOC_INTEL_TIGERLAKE_PCH_H
        select SOC_INTEL_TIGERLAKE_S3
        select SOC_INTEL_COMMON_BLOCK_HDA_VERB
+       select SOC_INTEL_ENABLE_USB4_PCIE_RESOURCES
        select SPD_READ_BY_WORD
        select SYSTEM_TYPE_LAPTOP
        select TPM_MEASURED_BOOT
```

Also need to check if we can drop `PCIEXP_HOTPLUG_PREFETCH_MEM_BELOW_4G`. But I don't remember what conditions triggered the original issue requiring it. Running out of MTRRs? (And I think there was more work being done upstream for the memory allocator that would be relevant in that case.)

bag-builds, 2022-10-02:
checked Intel recommended values, same situation, freeze at boot with thunderbolt dock attached. What I can do is just some test, because I don't have any knowledge about this

lethedata, 2024-11-13:
@crawfxrd I no longer have the hardware to work on this issue. Not sure if I should close it or leave it open for others who might still be having this issue.

With the needa-repro tag I'm leaning more towards closing it