system76/firmware-open #272: Arch Linux (LEMP 10): charge profile ACPI tables are not exposed

ethindp, 2022-01-06:
I haven't updated to the latest source code firmware, so am running the currently stable release. Am using Arch Linux with Linux 5.15.12 and the following system76 drivers:

* system76-driver 20.04.46
* system76-dkms 1.0.13-1
* system76-acpi-dkms 1.0.2+4+gb49caa5-1
* system76-io-dkms 1.0.1-2
* system76-oled 0.1.4-2

However, when attempting to use the `charge-profile` command of `system76-power` it claims I'm not running firmware that supports charge profiles.s Additionally, the `system76-power` daemon panics on boot with the message:

```
Starting System76 Power Daemon...
[INFO] Starting daemon
[INFO] Connecting to dbus system bus
[INFO] Rescanning PCI bus
[INFO] 0000:00:02.0: Intel graphics
[INFO] 0000:00:02.0: Function for 0000:00:02.0
[INFO] Disabling NMI Watchdog (for kernel debugging only)
[INFO] Setting automatic graphics power
[WARN] Failed to set automatic graphics power: does not have switchable graphics
[INFO] Initializing with the balanced profile
[INFO] Registering dbus name com.system76.PowerDaemon
[INFO] Adding dbus path /com/system76/PowerDaemon with interface com.system76.PowerDaemon
Started System76 Power Daemon.
[ERROR] fan daemon: platform hwmon not found
[INFO] Handling dbus requests
thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 2, kind: NotFound, message: "No such file or directory" }', src/hid_backlight.rs:81:55
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[INFO] caught signal: SIGTERM
Stopping System76 Power Daemon...
[INFO] daemon exited from loop
system76-power.service: Deactivated successfully.
Stopped System76 Power Daemon.
```

According to S76 support this is because Linux seems to be unable to locate the ACPI tables that the firmware exposes. I have installed pretty much everything related to system76 in the AUR and I'm still not able to use this feature. I don't know what else to do and switching to POP!_OS just for this is not an option.

crawfxrd, 2022-01-06:
There is no concept of profiles in the firmware.

What firmware version are you on?

```sh
cat /sys/class/dmi/id/bios_version
```

If `system76-acpi` is loaded and `/sys/class/power_supply/BAT0/charge_control_{end,start}_threshold` exists, then it's not a firmware issue.

```sh
lsmod | grep system76
cat /sys/class/power_supply/BAT0/charge_control_{start,end}_threshold
```

It's probably system76-power itself, given the other issues on Arch.

ethindp, 2022-01-06:
Okay, so apparently its decided to work fine now. Yay!

closed 2022-01-06 by ethindp