system76/firmware-open #278: Can't boot lemp10 after installing latest released firmware

brbar, 2022-01-11:
- Model: lemp10
- BIOS version: recent release (January 2022)
- Have two SSDs installed

Installed latest released firmware update and was unable to boot the system. Live PopOS USB was also non working.
During boot to PopOS seen the error message "frozen state for nvme device" and it was stuck on those messages.

In order to solve the issue I installed my original factory SSD and removed my both SSDs I currently using. It allows you to boot to the PopOS on SSD. Reverted back to 2021-07-20_93c2809 and my previous configuration with two SSDs start working as before.

### Steps to reproduce
Install second SSD and update to latest firmware on lemp10

crawfxrd, 2022-01-11:
> Have two memory sticks

This isn't possible on lemp. There's only 1 DIMM slot.

> Have two SSDs installed

What SSDs are you using?

jacobgkau, 2022-01-11:
Cannot recreate with two SSDs (tested 2x NVMe and 1x NVMe/1x SATA) and one additional DIMM in addition to the soldered RAM. Need more info about the particular hardware that's affected.

brbar, 2022-01-11:
> This isn't possible on lemp. There's only 1 DIMM slot.
I mean that I have not only factory soldered memory but I installed my own memory stick too. If this is related to another recent issue you solved in c73e48228

> What SSDs are you using?
both ADATA XPG SX8200 Pro 2Tb

crawfxrd, 2022-01-11:
Does it boot with the original drive and either of the ADATA drives installed?

brbar, 2022-01-11:
> Does it boot with the original drive and either of the ADATA drives installed?

I tried to boot with the factory drive and one ADATA and it still unable to boot.
I did not tried to boot with just one ADATA drive.

brbar, 2022-01-13:
Do you need any additional information in order to fix that issue?

crawfxrd, 2022-01-13:
@jacobgkau do we have any non-Samsung drives to test with?

> I tried to boot with the factory drive and one ADATA and it still unable to boot.

Was that booting from the factory drive or the ADATA drive?
Does it still happen if you switch the slots the factory drive and the ADATA drive are in?
If you're able to boot the factory drive with the ADATA drive installed, are you able to mount the ADATA drive and read/write to it?

jacobgkau, 2022-01-14:
> @jacobgkau do we have any non-Samsung drives to test with?

I have tested the following non-Samsung PCIe NVMe drives, all are able to boot normally:
- Toshiba/OCZ RD400 (256GB)
- PNY CS3030 (1TB & 4TB)
- Sabrent Rocket NVMe 4.0 (1TB)
- Kingston KC2500 (1TB)
- Western Digital Black SN850 (500GB)
- Western Digital Blue SN550 (500GB)
- Western Digital Green SN350 (240GB)

We do not currently have any ADATA drives in the lab.

crawfxrd, 2022-01-18:
As a shot in the dark, I disabled RTD3 on the CPU PCIe port (used for the SSD):

- [lemp10-76ec.zip](https://github.com/system76/firmware-open/files/7891010/lemp10-76ec.zip)

brbar, 2022-01-20:
I played a little bit with SSDs and found some new information.

Let's assume the following:
slot0 - is a vertical SSD disk slot located above CMOS battery and on the right side from the battery
slot1 - is a horizontal SSD disk slot located on the right side from memory slot
WD - is a factory SSD which were shipped with laptop with PopOS and with a factory partitioning
ADATA Bootable - bootable ADATA disk with EFI partition, encrypted
ADATA - second ADATA disk, non bootable, encrypted

-------------------
1.
slot0: WD
slot1: <empty>

result: works good
-------------------
2.
slot0: <empty>
slot1: WD

result: works good
-------------------
3.
slot0: ADATA Bootable
slot1: ADATA

result: not working
-------------------
4.
slot0: ADATA
slot1: ADATA Bootable

result: not working
-------------------
5.
slot0: WD
slot1: ADATA Bootable

result: not working, unable to boot nor from SSD nor from live usb, see the errors on photos attached
-------------------
6.
slot0: ADATA Bootable (named nvme1)
slot1: WD (named nvme0)

result: works good, I can boot to PopOS from WD and read / write from my ADATA Bootable
-------------------
7.
slot0: <empty>
slot1: ADATA Bootable

result: not working, I can see the password prompt, once enter the password it start booting but I see error messages and it is stuck
-------------------

the case #6 is the only case where I can access my ADATA drive
in the other cases with ADATA it seems like it is able to access it, because in case #7 system can decrypt the disk and start booting, but the controller errors do not allow to finish
PopOS live usb also stuck during boot if one of my ADATA disks attached
there is no issues with 2021-07-20_93c2809 at all
[image]
[image]
[image]

jacobgkau, 2022-01-27:
https://github.com/pop-os/pop/issues/2006 mentioned the boot issue on master firmware with an ADATA SSD is also happening on galp5. It sounds like that user's also seeing suspend/resume failures on currently-released firmware. I've updated the issue title to be easier to find.

curiousercreative, 2022-01-27:
As Jacob linked to, my ADATA SX8200 (non pro) will not suspend on current galp5 firmware (2021.07.20) and reproduces the same boot failure as OP (regardless of whether it's the boot device) on master as of today.

brbar, 2022-01-28:
@curiousercreative I also have a similar issue with suspend as mentioned here https://github.com/system76/firmware-open/issues/282#issuecomment-1017108836

@jacobgkau seems like you need to order one of ADATA SSD for your testing lab ;)

@jacobgkau could you please clarify is it possible to use more than 2Tb per slot on lemp10?

jacobgkau, 2022-01-28:
> @jacobgkau seems like you need to order one of ADATA SSD for your testing lab ;)

Our priority is supporting the hardware that we sell and ship, but we may get an ADATA SSD eventually. We're currently seeing some issues with other SSD models that we do ship, so I'd wait to see if the fixes for those also help here rather than jumping the gun.

> @jacobgkau could you please clarify is it possible to use more than 2Tb per slot on lemp10?

We do not currently ship SSDs in the lemp10 greater than 2TB, and it would not be officially supported. I personally _really_ wanted to offer them, but the Sabrent 4TB/8TB models with Phison E12 controllers were never 100% stable: https://github.com/pop-os/linux/pull/35 

The PNY 4TB/8TB models that I'm aware of also use Phison E12 controllers, and are affected by the same issues. (We did work directly with both Sabrent and PNY while investigating this.) With all of this in mind, I would not recommend using an SSD greater than 2TB, although you are welcome to try. It may work, but if it doesn't, it may not be something that's ever fixable (from what I understand, the issues partially stemmed from the motherboard's physical power delivery capabilities.)

curiousercreative, 2022-01-28:
@jacobgkau I'll happily send you my ADATA if that's helpful. Maybe you can send me a loaner or just outright swap.

curiousercreative, 2022-01-28:
Here's output from `fwupdmgr get-devices`
```
ADATA SX8200NP:
      Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
      Summary:            NVM Express Solid State Drive
      Current version:    SVN139B
      Vendor:             Silicon Motion, Inc. (NVME:0x126F)
      Serial Number:      2I3120051517
      GUIDs:              95d5f080-c19c-593f-bb78-ea8b8c309337 ← NVME\VEN_126F&DEV_2262&SUBSYS_126F2262&REV_03
                          3bdea050-e1dc-51de-85f9-1ff504270e1b ← NVME\VEN_126F&DEV_2262&SUBSYS_126F2262
                          e596cb13-ce08-5b05-833e-d564ad32ae49 ← NVME\VEN_126F&DEV_2262&REV_03
                          242d4bb0-16d1-56da-8ba9-f5e8bd695382 ← NVME\VEN_126F&DEV_2262
                          46d19cd3-4b42-58cf-8c02-a33f07f94c66 ← ADATA SX8200NP
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

curiousercreative, 2022-01-31:
@brbar if you need suspend back, you can try what I'm doing: https://github.com/pop-os/pop/issues/2006#issuecomment-1024960150

jacobgkau, 2022-01-31:
@curiousercreative This user never indicated that their suspend wasn't working on their lemp10, only that boot was broken with the latest firmware update (which has been rolled back this morning.) Assuming no information has been left out, I would _strongly_ recommend installing the firmware rollback update as opposed to pinning an outdated kernel version.

jacobgkau, 2022-01-31:
> @brbar if you need suspend back, you can try what I'm doing: pop-os/pop#2006 (comment)

If the firmware rollback from this morning fixes the issues for @brbar on lemp10, then I would not recommend pinning an outdated kernel version.

curiousercreative, 2022-01-31:
@jacobgkau ah, I was confused by the issue title. I'd suggest reverting the title back to it's original "Can't boot lemp10 after installing latest released firmware" or something omitting suspend as that doesn't appear to be part of the same issue. I am curious to know whether @brbar's lemp10 enters suspend properly (flashing LED) with their ADATA NVMe installed and on what kernel.

jacobgkau, 2022-01-31:
@curiousercreative This user reported an issue where their system failed to boot only when ADATA SSDs were installed. The issue title is correct, regardless of whether you are seeing the same behavior or not on your unit of a different model.

jacobgkau, 2022-01-31:
@curiousercreative I've pinged you on Mattermost in case you'd like to discuss our triaging strategies in more detail.

brbar, 2022-01-31:
@curiousercreative thanks for advice, I tried to install 5.15.15 kernel and it is not freezing now - I can close laptop lid, open back and proceed working. Did not tried to install 5.13 yet

@jacobgkau as I said I already been reverted to 2021-07-20 and it is working as before. Yes it is not working as it should because it is clearing my EFI entries but I use workaround for that /boot/efi/EFI/BOOT/BOOTX64.EFI plus I want Intel ME to be disabled because this is one of the main reasons people purchase System76 hardware with open firmware

@jacobgkau I can confirm that my laptop is also unable to suspend (flashing led) with ADATA (I did not tried to do that with the factory shipped SSD)

crawfxrd, 2022-05-05:
May also be resolved by disabling Advanced Error Reporting for the CPU PCIe RP.

crawfxrd, 2022-05-16:
Fixed with updated coreboot in fd9d73bdb9c8.

curiousercreative, 2022-10-03:
can this be closed/

crawfxrd, 2022-10-03:
I want to keep these open until firmware is released (#340).

crawfxrd, 2023-01-17:
2022-11-21_b337ac6 has been released for all TGL-U models.

closed 2023-01-17 by crawfxrd