system76/firmware-open #279: Oryp8 TPM Disabled

Localacct21, 2022-01-13:
- Model: Oryp8
- BIOS Version:	2021-09-23_b9b0e89
- EC version: 2021-09-23_b9b0e89
- OS: Windows 10

### Steps to reproduce

TPM is turned off even though the BIOS reports its turned on.

[image]

boot windows

### Expected behavior

TPM should be active

How do i turn it on?

Localacct21, 2022-01-14:
after searching i cannot find anything stating how to turn it on. the bios reports its on. windows reports its off
[image]
.

crawfxrd, 2022-01-14:
The TPM device is enabled on all models.

If you boot Linux, does it get detected?

```
sudo dmesg | grep tpm
ls /dev/tpm*
ls /sys/class/tpm/tpm*/
```

Localacct21, 2022-01-14:
[image]

This was the result

Localacct21, 2022-01-14:
> The TPM device is enabled on all models.
> 
> If you boot Linux, does it get detected?
> 
> ```
> sudo dmesg | grep -i tpm
> ls /dev/tpm*
> ls /sys/class/tpm/tpm*/
> ```

[image]

This was the result.

Localacct21, 2022-01-16:
So linux is saying the same thing. The TPM chip is not functioning

crawfxrd, 2022-01-16:
I'll have to flash back to the current release to check, since it is working on `master` on my unit.

If you are able to build cbmem, you can dump the coreboot log to check if it fails to initialize the device.

```
make -C coreboot/util/cbmem
sudo ./coreboot/util/cbmem/cbmem -c > cbmem.txt
```

I don't remember what the logs would be, but grepping for "TPM" should be sufficient.

Localacct21, 2022-01-17:
I cannot compile I'm getting errors

crawfxrd, 2022-01-18:
I flashed back to `2021-09-23_b9b0e89` and can confirm that TPM detection is broken.

coreboot logs:

```
tis_probe: No TPM device found
tlcl_lib_init: tis_init returned error
TPM: Can't initialize.
```

Localacct21, 2022-01-18:
So is this why it's not being reported to windows or Linux? Because the TPM detection is broken In this version of coreboot? When will this be fixed? The laptop is advertised as having TPM.

crawfxrd, 2022-01-18:
> So is this why it's not being reported to windows or Linux?

Yes. coreboot can't find and initialize the device, so the OS can't see it either.

> When will this be fixed?

This issue is fixed, but there are a couple other issues that need to be addressed as well before making a release.

Localacct21, 2022-01-19:
Could you please direct me on how to compile the master With Intel ME. I tried compiling before and it worked just fine but I disabled ME in the process. I need Intel ME for some of the software I use to work.

Localacct21, 2022-01-19:
> If you are using `2021-09-23_b9b0e89`, IME is hardcoded to be disabled.
> 
> If you are using `master`, see [`docs/intel-me.md#configuring`](https://github.com/system76/firmware-open/blob/239e2cbaed0d9ec0fa8a9f75b5a787f5a744147a/docs/intel-me.md#configuring).

Thank you Tim. Yes I want to recompile Master to have Intel ME.

crawfxrd, 2022-09-12:
Fixed with `2022-09-07_74e75ff`.

closed 2022-09-12 by crawfxrd