system76/firmware-open #282: lemp10 trouble waking from sleep after firmware update'

davidskeck, 2022-01-16:
- Model:  lemp10
- BIOS version: 2022-01-06_c73e482
- EC version: 2022-01-06_c73e482
- OS: Pop!_OS 21.10

Since updating the firmware, I am having issues waking from sleep. When I press any key, the display backlight will illuminate, but nothing ever shows up on the screen. I have to hold the power button until it turns off and then turn it back on.

### Steps to reproduce

1. Update to the latest firmware on the lemp10
2. Plug in the charger and let the lemp10 charge and go to sleep
3. Unplug and then open the laptop
4. The screen will illuminate, but it won't show the login screen, just a black screen.

### Expected behavior

The display should show the login screen.

### Actual behavior

There is a black screen displayed.

### Additional info

This appears to happen when it is charging, but I haven't seen it after short duration sleep, like for a few minutes. It could either be caused by the charger or a longer duration sleep.

jacobgkau, 2022-01-16:
Thank you for the report.

> but not after I press Super+Esc to put the computer to sleep.

Super-Esc does not put the computer to sleep, it just locks the screen and powers off the display. Fn-F12 puts the computer to sleep, same as closing the lid without an external display connected.

- Can you provide details about the SSDs, RAM, and any peripherals attached to the system?
- Are the SSDs and RAM the same ones that shipped with your system, or are they aftermarket?
- When you close the lid to suspend, does the power LED start blinking? What color is it blinking?
- When you open the lid and the backlight comes on, does the power LED stop blinking? Does pressing Ctrl-Alt-F5 do anything, or does the screen remain blank?
- If you are able to generate a log archive using the System76 Driver app immediately after booting back up after a failure, and can provide that log archive, it could potentially be useful for determining the issue. If you would rather provide the log archive privately, you can do so in a [support ticket](https://support.system76.com/) (you may have to open the ticket first before you get the option to attach any files.)

davidskeck, 2022-01-16:
Good to know about Super+Esc.

* Yes, I have 40 GB of RAM total, 32 GB Crucial that has caused me no issues at all so far.
* The one that shipped with this unit is still installed, but the boot drive is now a Samsung 970 Plus 1TB. The Western Digital SN750 that came with it is the backup drive in the secondary slot now.
* I'll have to get back to you on the power LED. Currently using the system and don't want to execute a reboot cycle to get this info just yet.
* The LED is not blinking when I open the lid. It is solid green.
  * Pressing Ctrl+Alt+F5 shows a terminal, which has a lot of output that flashes on and off. I have captured two images which show the contents of the two different screens that flash.
* I will generate this log and open a support ticket to attach it to.

Thanks
[image]
[image]
!

ktbarrett, 2022-01-18:
I can confirm I also am having the exact same issue with my Lemur Pro starting after the most recent firmware update. [So](https://www.reddit.com/r/System76/comments/s5vspb/trouble_after_the_latest_firmware_update_i_have_a/) [are]() [others]().

> Can you provide details about the SSDs, RAM, and any peripherals attached to the system?
My [lshw dump](https://github.com/system76/firmware-open/files/7885386/hwdump.txt).

> Are the SSDs and RAM the same ones that shipped with your system, or are they aftermarket?
All factory.

> When you close the lid to suspend, does the power LED start blinking? What color is it blinking?
Yes. It blinks slowly green.

> When you open the lid and the backlight comes on, does the power LED stop blinking?
It stops blinking and becomes solid green.

> Does pressing Ctrl-Alt-F5 do anything, or does the screen remain blank?
Testing...

jacobgkau, 2022-01-18:
Thank you for the information. I'm currently testing on an i5-1135G7 unit (the same as both of yours) and have not been able to make the issue happen yet with a variety of RAM and SSD combinations. I've asked our support team to forward us the logs from the ticket.

If anyone experiencing the issue could try pressing Spacebar during boot, then `e` to edit the kernel boot parameters, remove the last four items (`quiet loglevel=0 systemd.show_status=false splash`), add `no_console_suspend` and press Enter to boot, then recreate the issue, does any additional information show up on the screen where it was blank before?

If you are able to access the machine over SSH after the issue has occurred and check `sudo dmesg` for any output, that may also be helpful. ([Removing the RAM or additional SSDs](https://tech-docs.system76.com/models/lemp10/repairs.html) to see if the issue does or doesn't correspond to any particular configurations is something else that could help narrow down or eliminate possibilities for the cause.)

Ceridan, 2022-01-18:
I have experienced the same issue with my Lemur Pro 10 immediately after the update of the firmware to 2022-01-06_c73e482.

Let me provide my hw dump and answers to the questions above, probably it will help to investigate the problem.

> Can you provide details about the SSDs, RAM, and any peripherals attached to the system?

[lemp10_hw_dump.txt](https://github.com/system76/firmware-open/files/7891387/lemp10_hw_dump.txt)

> Are the SSDs and RAM the same ones that shipped with your system, or are they aftermarket?

All factory.

> When you close the lid to suspend, does the power LED start blinking? What color is it blinking?

The power LED starts blinking green light.

> When you open the lid and the backlight comes on, does the power LED stop blinking?

It stops blinking and becomes solid green.

> Does pressing Ctrl-Alt-F5 do anything, or does the screen remain blank?

The screen remains blank.

jacobgkau, 2022-01-18:
@crawfxrd This isn't quite the same symptom, but if I place a Samsung 980 Pro 500GB in Slot 2 and nothing in Slot 1, then I am unable to enter suspend; the LED never starts blinking, and I start getting I/O errors.

When the 980 Pro is in Slot 1 instead of Slot 2, this doesn't happen, even if another NVMe drive is populated in Slot 2.

I am getting cbmem outputs and will try the firmware from https://github.com/system76/firmware-open/issues/278#issuecomment-1015651722 to see if that makes a difference.

jacobgkau, 2022-01-18:
cbmem output with 980 Pro in slot 1 (working): [cbmem-slot1.txt](https://github.com/system76/firmware-open/files/7891593/cbmem-slot1.txt)
cbmem output with 980 Pro in slot 1 (not working): [cbmem-slot2.txt](https://github.com/system76/firmware-open/files/7891594/cbmem-slot2.txt)

The firmware from https://github.com/system76/firmware-open/issues/278#issuecomment-1015651722 does not seem to make a difference (although I had to flash it by modifying an existing firmware USB instead of just using the files as provided, since the files as provided just booted to an EFI shell-- the version after flashing still showed `2022-01-06_c73e482`.)

crawfxrd, 2022-01-18:
I used the wrong ZIP script. I updated it; version should be `2022-01-18_efe70c4`.

jacobgkau, 2022-01-18:
Got it, I see `2022-01-18_efe70c4` after flashing now but it did not fix the issue I'm able to recreate. Still seeing the power LED remain solid green and I/O errors after I try to suspend.

crawfxrd, 2022-01-18:
It's weird that it's working on the CPU PCIe RP, but not the PCH RP. I would have expected the opposite. But I don't notice anything strange in the cbmem log.

JChampt, 2022-01-18:
I am having the same issue too with a Lemur Pro I got yesterday.  The OS prompted me for a firmware update and now the system can't wake from sleep.  

Here is what I see as an error.

[image]

I have a stock Lemur Pro choosing the 16gb option and the 250gb pcie3 hard drive currently using 2022-01-06_c73e482 firmware. 

It was waking from sleep just fine with the 2021-07-20_93c2809 firmware.  I would like to try rolling back to that, but I haven't figured out how to do that yet.

girishsj, 2022-01-18:
I see the following when the lid is opened ...
[image]

The messages keep repeating with the fan speed reaching maximum. If I try to interrupt (Ctrl-C), I see ...

[image]

ktbarrett, 2022-01-18:
@jacobgkau Added the output after loading from sleep to https://github.com/system76/firmware-open/issues/282#issuecomment-1014984674. Seems to be the same as @JChampt's.

crawfxrd, 2022-01-18:
If someone affected by errors related to 00:06.0 could try this firmware, `2022-01-18_efe70c4`:

- [lemp10-76ec.zip](https://github.com/system76/firmware-open/files/7891737/lemp10-76ec.zip)

This disables RTD3 on the CPU PCIe RP, like it is in the previous release.

We likely cannot enable RTD3 (so no power savings) until coreboot 4.16, which will have the rest of the logic implemented.

Ceridan, 2022-01-18:
@crawfxrd will you provide this fix in the official firmware update in the near future?

davidskeck, 2022-01-18:
> If someone affected by errors related to 00:06.0 could try this firmware, `2022-01-18_efe70c4`:
>
>     * [lemp10-76ec.zip](https://github.com/system76/firmware-open/files/7891737/lemp10-76ec.zip) [...]

How do you install this?

jacobgkau, 2022-01-18:
> @crawfxrd will you provide this fix in the official firmware update in the near future?

@Ceridan If the fix works, of course it will be available through the official update channel in the near future.

> How do you install this?

@davidskeck Extract the contents of the ZIP file onto a FAT32-formatted USB flash drive and boot from it.

girishsj, 2022-01-18:
OK. It worked. Now, it prompts to update the firmware again. I somehow need to differentiate between the problematic update and the update that fixes the problem. :)

crawfxrd, 2022-01-18:
@girishsj do you have any warnings or errors in dmesg related to device 00:06.0?

If you suspend the system, does the power LED blink?

davidskeck, 2022-01-18:
> > @crawfxrd will you provide this fix in the official firmware update in the near future?
> 
> @Ceridan If the fix works, of course it will be available through the official update channel in the near future.
> 
> > How do you install this?
> 
> @davidskeck Extract the contents of the ZIP file onto a FAT32-formatted USB flash drive and boot from it. You should see a firmware updater screen similar to when installing updates over-the-air.

OK. I installed the new firmware and I no longer have this issue. The power LED does not blink when I close the lid, however. I am also being prompted to install the latest firmware again, of course.

girishsj, 2022-01-18:
@crawfxrd here is what I see in dmesg:
[  818.402573] PM: suspend entry (s2idle)
[  818.417991] Filesystems sync: 0.015 seconds
[  818.537743] Freezing user space processes ... (elapsed 0.003 seconds) done.
[  818.540877] OOM killer disabled.
[  818.540879] Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
[  818.542277] printk: Suspending console(s) (use no_console_suspend to debug)
[  818.872204] ACPI: EC: interrupt blocked
[  823.293236] ACPI: EC: interrupt unblocked
[  823.532219] ACPI Error: No handler or method for GPE 6B, disabling event (20210730/evgpe-839)
[  823.661746] nvme nvme1: 8/0/0 default/read/poll queues
[  823.853285] OOM killer enabled.
[  823.853288] Restarting tasks ... 
[  823.853776] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[  823.858457] done.
[  823.858470] thermal thermal_zone0: failed to read out thermal zone (-61)
[  823.927976] video LNXVIDEO:00: Restoring backlight state
[  824.151654] PM: suspend exit

girishsj, 2022-01-19:
After applying the temporary fix, I am observing that my Lemur Pro is running hotter than earlier. I am not which is more bad -- Keeping lid open -- or -- Closing the lid but running the CPU hot. I am going to roll back the fix (by upgrading to the official firmware update). Is there a way to rollback the update completely? In other words, can I get the previous version of firmware to apply?

crawfxrd, 2022-01-19:
Here is `2021-07-20_93c2809`:

- [lemp10-76ec.zip](https://github.com/system76/firmware-open/files/7893964/lemp10-76ec.zip)

davidskeck, 2022-01-19:
> After applying the temporary fix, I am observing that my Lemur Pro is running hotter than earlier. I am not sure which is more bad -- Keeping lid open -- or -- Closing the lid but running the CPU hot. I am going to roll back the fix (by upgrading to the official firmware update). Is there a way to rollback the update completely? In other words, can I get the previous version of firmware to apply?

I noticed a lot more battery drain after installing the temporary fix as well.

ktbarrett, 2022-01-19:
@crawfxrd Is there an archive of firmware builds somewhere? I kinda expected Github "releases" for each FW release with all the build artifacts, but I didn't find anything like that in this org. You could do the whole release process via Github Actions to make it very clear how the FW is built and that the assets are legitimate, while automating nightly/weekly releases.

brbar, 2022-01-20:
I would add my two cents here.
On all versions of firmware I noticed an issue when I open the lid on lemp10 - it got frozen but I still able to open first tty Ctrl+Alt+F1 and see that vbetool process got crazy utilizing 100% CPU on my MX Linux 21. The solution is to kill -s KILL vbetool and if X server did not crash you can proceed work.

lukweza, 2022-01-30:
> @Ceridan If the fix works, of course it will be available through the official update channel in the near future.
>
>
> @davidskeck Extract the contents of the ZIP file onto a FAT32-formatted USB flash [...]

I extracted the contents of the lemp10-76ec.zip file posted earlier ( 2 directories: a directory called EFI and one called system76-2212f0e731..... ) to a fat32 formatted usb stick. 
The bios complains that it can not find anything to boot off.
Does the bios have to be reconfigured to allow booting from the contents of the zip ?

This bios update was really, really terrible. 
Had I received my laptop after this update - I would have sent it right back for a refund. 
Thankfully, I remember how excellent the laptop was running before the update .. - I would really like to get back to that state somehow.

jacobgkau, 2022-01-31:
We have reverted the lemp10 firmware update this morning. Anyone who manually flashed `2021-07-20_93c2809` from a USB will no longer receive update notifications; otherwise, if you are on `2022-01-06_c73e482`, you will receive a notification to downgrade back to `2021-07-20_93c2809`.

It may take ~24 hours for the firmware notification cache on your system to expire. You can force this to happen immediately instead by running this command:

```
sudo rm -rf /var/cache/system76-firmware-daemon/
```

...then closing and re-opening the Settings app (and going to the Firmware page.)

I apologize to any lemp10 owners affected by this issue. We would have reverted the update sooner had we known it would take this long to fix. Our QA processes have been adjusted to cover specific SSD models in each individual slot going forward for all firmware updates on all laptop models; this is more in-depth than we've previously seen the need for, but we are committed to maintaining stability while also providing the longest-term firmware support possible for our devices. Finally, we are open to providing an easier-to-use public archive of older firmware versions and/or ways to roll back as requested in https://github.com/system76/firmware-open/issues/99 and https://github.com/system76/firmware-open/issues/289.

AristidesMelendez, 2022-02-01:
I did the downgrade using the firmware update tool in the settings and it works. The suspension and wake up works again :+1:  :smiley:  
Thanks.

dkarter, 2022-02-04:
Hello! 

I have also experienced the issue ever since updating PopOS to 21.10 with the firmware update that came with it. 

Doing a full re-install of PopOS and creating a new user account helped fix the issue (or maybe it was a red herring? idk).

That is until today when I got the notification to "upgrade" the firmware (now realizing was technically a downgrade to `2021-07-20_93c2809`).

Ever since the downgrade I am now getting the black screen again, with no outputs or logs, green light is on! :(
The only recourse is to hard reset the device every time.

Is there anything I can do to fix this without having to re-install the OS again and reconfigure my system for the 3rd time with a new user account? Maybe "rolling back the rollback" to the newer version which seems buggy for others will solve it for me?

Has anyone else either started re-experiencing the black screen on resume from sleep since the downgrade?

Another issue I noticed which could potentially be related, is that the computer doesn't always "fully sleep". Sometimes I notice the green light on the side is solid instead of blinking when the lid is closed. If I leave my Lemur Pro in this state the battery will drain completely within a few hours. I am not using any "keep awake" apps or anything that should prevent the computer from sleeping - just have Firefox open.

[image]

dkarter, 2022-02-04:
Hello! 

I have also experienced the issue ever since updating PopOS to 21.10 with the firmware update that came with it. 

Doing a full re-install of PopOS and creating a new user account helped fix the issue (or maybe it was a red herring? idk).
That is until today when I got the notification to "upgrade" the firmware (now realizing was technically a downgrade to `2021-07-20_93c2809`).

Ever since the downgrade I am now getting the black screen again, with no outputs or logs, green light is on! :(
The only recourse is to hard reset the device every time.

Is there anything I can do to fix this without having to re-install the OS again and reconfigure my system for the 3rd time with a new user account? Maybe "rolling back the rollback" to the newer version which seems buggy for others will solve it for me?

Has anyone else either started re-experiencing the black screen on resume from sleep since the downgrade?

Another issue I noticed which could potentially be related, is that the computer doesn't always "fully sleep". Sometimes I notice the green light on the side is solid instead of blinking when the lid is closed. If I leave my Lemur Pro in this state the battery will drain completely within a few hours. I am not using any "keep awake" apps or anything that should prevent the computer from sleeping - just have Firefox open.

[image]
[image]

In the video I try to open the lid quickly after closing it and it seems as if it's working but when I close the lid and wait a few seconds it keeps the power led solid green and when I open the lid I get the unresponsive black screen
https://user-images.githubusercontent.com/551858/152484688-7a932126-e2a4-4860-87b4-12ed911e677d.mp4

redoxeon, 2022-02-11:
Not sure if it's exactly related, but I plugged in a Razer Core X eGPU, and it caused the sleep issue to trigger. I talked to support and the RMA technicians were able to fix it, but that was before I realized it was the eGPU, and when I got it back, I verified that it was fixed, then plugged in the eGPU and broke it again (lol).

Since then, I've reverted the firmware following the directions above, and the issue persists, even through a clean reinstall of Pop!_OS (but not when testing from a live USB, curiously). I'm on firmware version `2021-07-20_93c2809` and Pop!_OS 21.10.

Behavior is just closing the lid or selecting "suspend" in the menu. The screen goes dark, but the indicator light stays solid green, and the battery drains at non-sleep rates.

jacobgkau, 2022-02-11:
@redoxeon The issue suspending with Thunderbolt or USB-C devices on Tiger Lake-U systems such as lemp10 is being tracked separately: https://github.com/system76/firmware-open/issues/199

redoxeon, 2022-02-11:
o i c, thanks, and sorry for getting on the wrong issue :)

crawfxrd, 2022-05-05:
Should be fixed by configuring FSP to disable Advanced Error Reporting for the CPU PCIe RP.

crawfxrd, 2022-05-16:
Fixed with updated coreboot in fd9d73bdb9c8.

cement-head, 2022-05-17:
How does one update the System76 Firmware?  Does this happen automagically?

crawfxrd, 2023-01-17:
2022-11-21_b337ac6 has been released for all TGL-U models.

closed 2023-01-17 by crawfxrd

ktbarrett, 2023-01-19:
I have confirmed no issues with the new firmware and waking from sleep. This took a while to get out... is there any indication why?

leviport, 2023-01-19:
@ktbarrett Because we test firmware updates heavily before releasing them, and they don't always pass testing. I'm sure most users would prefer that we catch the regressions rather than them, but the users that don't mind taking the risk are free to build and flash their own versions from https://github.com/system76/firmware-open