system76/firmware-open #284: ACPI issues with Windows USB / fwts

ids1024, 2022-01-18:
- Model: gaze15
- BIOS version: 2021-07-20_93c2809
- EC version: 2021-07-20_93c2809
- OS: Windows 10 Installer USB / Pop!_OS 21.10

Trying to boot a recently created Windows 10 installer USB, I see a blue screen with stop code `ACPI_BIOS_ERROR`. (It looks like there should be a registry edit to get a more specific error code, but I can't seem to figure it out for an installer USB rather than installed OS.)

`fwts` also reports issues related to ACPI tables:

```
Test Failure Summary
================================================================================

Critical failures: NONE

High failures: 5
 acpi_pwrb: PWR_Button field in FACP should not be zero with ACPI PNP0C0C device.
 uniqueid: \_SB_.PCI0.I2C0.H02C._HID/_UID conflict with \_SB_.PCI0.I2C0.H015._HID/_UID
 uniqueid: \_SB_.PCI0.LPCB.TPM_._CID/_UID conflict with \_SB_.PCI0.LPCB.TPM_._HID/_UID
 uniqueid: \_SB_.PCI0.I2C0.H015._CID/_UID conflict with \_SB_.PCI0.I2C0.H015._HID/_UID
 uniqueid: \_SB_.PCI0.I2C0.H02C._CID/_UID conflict with \_SB_.PCI0.I2C0.H015._HID/_UID

Medium failures: 1
 autobrightness: 3059 brightness levels did not match the brightness level just set for backlight intel_backlight.

Low failures: NONE

Other failures: NONE
```

I don't know what the brightness issue is about, but the "high" failures seem like they should be addressed, and shouldn't be too hard. I wonder if there could be a way to run `fwts` on CI to make sure everything is passing...

crawfxrd, 2022-01-18:
https://doc.coreboot.org/acpi/uid.html

> Windows 10 verifies all devices with the same _HID or _CID and makes sure that no _UID is duplicated. If it is it’ll BSOD.

crawfxrd, 2022-05-16:
Blue screen fixed with updated coreboot in fd9d73bdb9c8.

crawfxrd, 2023-03-01:
Fix released in 2022-11-21_b337ac6.

closed 2023-03-01 by crawfxrd