system76/firmware-open #287: NVIDIA GPU not showing up on galp5-3050

jacobgkau, 2022-01-24:
- Model: galp5-3050
- BIOS version: `2021-07-20_93c2809`
- EC version: `2021-07-20_93c2809`
- OS: Windows 10

Support team (and a customer) are reporting that the galp5 3050 is not able to install the NVIDIA driver in Windows 10 because the NVIDIA GPU is not detected in Windows.

### Steps to reproduce

1. Install Windows 10
2. Attempt to install the NVIDIA driver

### Expected behavior

NVIDIA driver should install.

### Actual behavior

NVIDIA driver does not install because NVIDIA GPU is not detected.

### Additional info

@ahoneybun confirmed on the lab unit.

ahoneybun, 2022-01-24:
I did the following after installing Windows 10
- Installed Intel Chipset, SerialIO and Thunderbolt drivers then rebooted.
- Tried the latest 511.23 for Game Ready and the 511.09 for Studio Driver.

ahoneybun, 2022-01-24:
Same behavior on firmware version 2022-01-18_239e2cb.

echo-84, 2022-02-16:
The nvidia driver does not including the gpu's full device id is why the install isn't working/detecting. I think it was the DEV code that was missing but it could have been the VEN code. Been a few weeks since I messed with it.

When I messed with it, I did somehow manage to manually install the driver tho had issues reproducing it.
- download nvidia driver
- extract the nvidia driver
- open device manager
- select GPU
- update driver
- browser computer for driver
- ??? - Not sure what worked after this step because windows refused to show the list of gpu drivers available.

Note: I did this in a VM with pci passthrough on a oryp8.

ahoneybun, 2023-08-21:
Issue is still there on firmware version `2022-11-21_b337ac6`.

ahoneybun, 2023-08-21:
Not fixed in this PR either:

https://github.com/system76/firmware/pull/538

pidydx, 2024-09-21:
@ahoneybun @crawfxrd I am having this issue on firmware version 2023-12-12_6f1e653

After installing Windows, the System76 drivers, and the Intel DSA I still have 4 devices listed under "Other devices"

- 3D Video Controller (Device ID shows its the 3050)
- Base System Device
- PCI Data Acquisition and Signal Processing Controller 
- Unknown device 

Attempting to install the NVIDIA Graphics Driver 561.09 reports that it could not find compatible hardware.