system76/firmware-open #288: [Feature Request] Coreboot and PC Config tools

Localacct21, 2022-01-25:
I've noticed star-labs has a [coreboot configuration](https://support.starlabs.systems/kb/guides/coreboot-configurator) tool do we have such a tool or is one being made? Also TuxedoComputers has a [Control Panel](https://github.com/tuxedocomputers/tuxedo-control-center) for their Clevo Based Laptops.

are there plans to impliment the solutions they came up with or something totally S76 made?

bmantic, 2022-01-30:
Seconded. Some of us don’t have the luxury to solely use Linux, and while I don’t mind paying a premium to support what you guys do, I shouldn’t have to deal with a subpar windows experience (I.e. no fan control, acpi, etc.) because of it.

mrtrivia, 2022-01-30:
Thirded.I just bought a Gazelle because it's so expandable, and the hardware is great (and unlike most companies, Sys76 tells you everything about their computers on their website).But I've been planning all along to mostly run Windows on it.I don't expect Sys76 to support my installing and using Windows--which is fine.But I don't think Windows-compatible firmware (including maybe the cool keyboard lighting?) is too much to ask. (And apparently the Gazelle 16 can't be downgraded to the proprietary firmware.)Mitch

Localacct21, 2022-01-31:
No official response on this?

leviport, 2022-01-31:
>No official response on this?

On your feature request? Sure, I think it sounds like a cool idea. I'm not sure why people are jumping onto this feature request to ask about Windows support though.

Localacct21, 2022-01-31:
Because we need windows support as well. Linux isn't the main operating system at work. I have to use windows as most of the programs I'm told to use are windows only.  I can't even use the programs my IT department suggested because TPM support is broken. 

I watched a video by your founder and Jeremy Soller. It said they were not Interested in making coreboot just have the standard features the proprietary Bios has they said there would be more features. Well coreboot does work, but where's the TPM, ways to adjust memory timings, exct  Weres support for windows? These features are basic BIOS features that are not in coreboot.

leviport, 2022-01-31:
I understand, but is this issue a feature request for better ways to configure Coreboot, or is it an issue about having better Windows support? We're crossing the beams here. Confusing github issues are a lot less useful than focused github issues that address one single item.

If you want me to address Windows support, then I can tell you that we are working on it.

Localacct21, 2022-01-31:
> I understand, but is this issue a feature request for better ways to configure Coreboot, or is it an issue about having better Windows support? We're crossing the beams here. Confusing github issues [...]

Sorry for being off topic yes this issue is for better ways to configure Coreboot. The reason WHY is windows. But not the cause

leviport, 2022-01-31:
No worries. The reason I make the distinction is that Windows support can and probably will be improved before designing an application for customizing Coreboot configurations.

Localacct21, 2022-02-07:
> No worries. The reason I make the distinction is that Windows support can and probably will be improved before designing an application for customizing Coreboot configurations.

Do you feel that coreboot has lived up to what Jeremy Soller or Carl Richell stated in this talk? It's what made me want to get a S76

https://youtu.be/YbKSkPVz89o

leviport, 2022-02-07:
@Localacct21 Short answer: yes, but there is always room for further improvement. Let's please not get off-topic again though. If you're looking for discussion, https://chat.pop-os.org/pop-os/ is a much better place for it.

MilesBHuff, 2022-06-15:
As a Linux user, I would love to have the ability to configure the BIOS without recompiling.
I *especially* want to be able to undervolt the CPU.

ddetton, 2023-08-10:
I would like to be able to adjust the fan curve without having to reflash the firmware.  Tuxedo Computers has a software tool called Tuxedo Control Center (TCC) that allows you to create profiles so that you can have multiple fan curves that can be enabled depending on what you're doing.  My understanding is that TCC only works on Tuxedo hardware.  Having a System76 version of this seems like something that many System76 users would appreciate.

Etrinque, 2023-11-07:
I agree with ddetton here, being able to control curves and not have deal with min/max would be ideal. I would much rather be able to have my fans on a slow drone on my adder3 to keep baseline down, however jumping hoops isnt worth it. so I as well second some interface for custom curves.