system76/firmware-open #289: Provide and document firmware rollback solution

curiousercreative, 2022-01-31:
Given regressions will be introduced in firmware updates, users should have the ability to rollback a firmware update and/or self-serve flashing an arbitrary firmware release for their machine. However it is implemented, flashable firmware organized by supported device and sorted by release date should be available. Yes the source to build the firmware is available here, but I've personally failed to build firmware releases from 6+ months ago on the current Pop!_OS release (likely due to dependencies).

Relates to:
- https://github.com/system76/firmware-open/issues/99
- https://github.com/system76/firmware-open/issues/282#issuecomment-1016626105

jackpot51, 2022-01-31:
We intend to use LVFS at some point which will introduce this capability

jacobgkau, 2022-01-31:
This seems like a duplicate of https://github.com/system76/firmware-open/issues/99. That issue is requesting a rollback feature and this issue is requesting a rollback feature.

curiousercreative, 2022-01-31:
@jacobgkau it mostly duplicates, I agree.

jacobgkau, 2022-01-31:
Thanks. I'm going to close this as a duplicate of https://github.com/system76/firmware-open/issues/99 to keep things organized, but we are interested in making this easier/more accessible, both from a documentation and tooling perspective.

closed 2022-01-31 by jacobgkau