system76/firmware-open #301: lemp10 shuts off from suspend when plugging in a USB hub on the right

orangecms, 2022-03-21:
- Model: lemp10
- BIOS version:  2021-07-20_93c2809
- EC version: 
- OS: Pop!_OS 21.10

### Steps to reproduce

I have this USB hub with an ethernet port: https://i-tec.pro/de/produkt/u3metalg3hub-3/

When the laptop is sleeping and I plug that hub into the USB port on the right, it shuts off.

I first thought I had accidentally touched the power button, but veritifed that the hub us causing the issue after investigating and can reproduce it.

### Expected behavior

The laptop should remain in suspend.

### Actual behavior

:boom: it shuts off :disappointed: 

### Additional info

[image]

jacobgkau, 2022-03-21:
I don't have one of those hubs to test with (and this doesn't happen with e.g. a USB flash drive), but it could be helpful to know if this happens on the current development version of the firmware, since it's been 6+ months since the last release. You can build and flash development firmware with these commands:

```
git clone https://github.com/system76/firmware-open
cd firmware-open
./scripts/update.sh
./scripts/deps.sh
./scripts/build.sh lemp10
./scripts.flash.sh lemp10
```

(The last command will power off the machine.) When you're finished testing, you can get back onto regular firmware by opening Settings -> Firmware and clicking the Update button.

Just to clarify, the green LED on the left side of the laptop is blinking while it's in suspend, and it stops blinking/shuts off when you plug this hub in? Does the laptop successfully suspend (including blinking LED) and resume if you plug the hub in before suspending?

orangecms, 2022-03-21:
Yea, the LED goes off from its usual blinking when in suspend and I plug the hub in.

I've just ordered a different but similar hub for testing, will check tomorrow. 

I will also test the scenario you suggest, trying to suspend while the hub is already plugged in. There are some other issues here like #199 which may or may not be related. I hope we'll catch this. :-)

When all that is done, I'll see about the latest firmware version.

orangecms, 2022-03-21:
I did a few experiments now.

- suspend works when the hub is already plugged in
- the sudden shutoff did not occur while a charger was connected

orangecms, 2022-03-23:
So this other hub does _not_ cause the shutoff, luckily.

That one:
https://www.amazon.com/AmazonBasics-3-Port-Adapter-Gigabit-Ethernet/dp/B07V5K6LSF?th=1&psc=1

leviport, 2024-03-26:
The charger doesn't have high enough voltage to produce a visible spark. Are you sure you're not seeing electrostatic discharge?

leviport, 2024-03-26:
If it's static, you'll mostly want to make sure you, the machine, and the charger are better grounded. The default lemp9-12 charger doesn't have a ground pin on the AC cable, so the charger won't work as a ground like most others usually do.