system76/firmware-open #305: Authentication required to clone on fresh system

jacobgkau, 2022-04-08:
- Model:  lemp10
- BIOS version:  2021-07-20_93c2809
- EC version:  2021-07-20_93c2809
- OS:  Pop!_OS 22.04 Pre-release

### Steps to reproduce

```
git clone https://github.com/system76/firmware-open
cd firmware-open
./scripts/update.sh
```

### Expected behavior

All required submodules are updated.

### Actual behavior

The output stops at orbtk being hosted on Redox's infrastructure.

```
system76@pop-os:~/firmware-open$ ./scripts/update.sh 
Already up to date.
Cloning into '/home/system76/firmware-open/libs/coreboot-table/orbtk'...
Username for 'https://gitlab.redox-os.org': 
```

### Additional info

It's been a month or two since I've had to build firmware, but I used to be able to clone this on a fresh machine without setting up any GitHub or GitLab authentication.

crawfxrd, 2022-04-09:
@jackpot51 I'm guessing it was deleted from GitLab?

Opened [redox-os/coreboot-table#2](https://gitlab.redox-os.org/redox-os/coreboot-table/-/merge_requests/2) to update the submodule path.

jacobgkau, 2022-04-12:
This is getting farther now after https://gitlab.redox-os.org/redox-os/coreboot-table/-/merge_requests/2, but I'm now being prompted for other items later:

```
Cloning into '/home/jacob/firmware-open/tools/unME11'...
Username for 'https://github.com': 
Password for 'https://github.com': 
remote: Repository not found.
fatal: Authentication failed for 'https://github.com/ptresearch/unME11.git/'
fatal: clone of 'https://github.com/ptresearch/unME11.git' into submodule path '/home/jacob/firmware-open/tools/unME11' failed
Failed to clone 'tools/unME11'. Retry scheduled
Cloning into '/home/jacob/firmware-open/tools/unME12'...
Username for 'https://github.com': 
Password for 'https://github.com': 
remote: Repository not found.
fatal: Authentication failed for 'https://github.com/ptresearch/unME12.git/'
fatal: clone of 'https://github.com/ptresearch/unME12.git' into submodule path '/home/jacob/firmware-open/tools/unME12' failed
Failed to clone 'tools/unME12'. Retry scheduled
```

crawfxrd, 2022-04-12:
Looks like the entire Positive Technologies org (https://github.com/PositiveTechnologies) was deleted.

But I don't think we ever used `unME11`/`unME12`, so the tools can be removed.

kldload, 2022-04-16:
You guys didn't even test before you closed the issue. You still can't install deps from a fresh install of popos live. It fails at the redox part that was mentioned in the original issue...

coreboot-table/orbtk

Cloning into '/home/system76/firmware-open/libs/coreboot-table/orbtk'...
Username for 'https://gitlab.redox-os.org':

jacobgkau, 2022-04-22:
I am also seeing `update.sh` stop asking for authentication on `orbtk`.

jacobgkau, 2022-04-22:
https://github.com/system76/firmware-open/pull/310 addresses the `orbtk` issue, this will be closed again once that's merged.

closed 2022-04-26 by jackpot51