system76/firmware-open #307: lemp10: 8GB DIMM not running in dual-channel with 8GB soldered RAM

jacobgkau, 2022-04-12:
- Model:  lemp10
- BIOS version:  Tried current master firmware (`2022-03-10_05577ba`) and current release firmware (`2021-07-20_93c2809`).
- EC version:  Same as above.
- OS:  Pop!_OS 22.04 Pre-release & Pop!_OS 21.10

Like lemp9 was before it, lemp10 is being advertised as having dual-channel memory when ordering with 16GB of RAM (8GB soldered + 8GB DIMM). When running with RAM of the correct speed, lemp9 does show two channels in use in `sudo dmidecode -t 17` output, but lemp10 only shows 1 channel in use (channel 0) even when using an 8GB 3200 MHz RAM stick.

### Steps to reproduce

1. Populate lemp10 with an 8GB 3200 MHz RAM stick.
2. Boot and run `sudo dmidecode -t 17`.

### Expected behavior

RAM should run in dual-channel mode when using a DIMM that matches the speed and size of the soldered RAM.

### Actual behavior

RAM runs in single-channel mode even when matching the soldered RAM's size and speed.

### Additional info

The customer was shipped a `NMSO480E82-3200E` (a web search shows it's Goldkey brand), and the Neo Forza sticks we have in the lab for testing report the same model number.

crawfxrd, 2022-04-18:
Do other TGL models report single channel configuration?

jacobgkau, 2022-05-02:
> Do other TGL models report single channel configuration?
> 
> All TGL models I have report single channel config.

You're correct, darp7 with two identical RAM sticks (GKE800SO102408-2666A) reports both sticks in Channel 0.

crawfxrd, 2022-06-13:
TGL changed memory init. ADL presumably has this issue as well.

May be changing the index in romstage `spd_info` or overriding `DIMMS_PER_CHANNEL`, but I'm not sure what exactly they do.

crawfxrd, 2022-07-07:
Most likely it's only reporting it wrong and would require overriding SMBIOS data on every board.

Would need an FSP debug build to see what meminit actually does.

witchof0x20, 2022-11-23:
Just chiming in to report I have this same issue on my gaze17 (ADL).

witchof0x20, 2022-11-25:
gaze17 (ADL)
```
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.
# SMBIOS implementations newer than version 3.2.0 are not
# fully supported by this version of dmidecode.

Handle 0x000A, DMI type 17, 40 bytes
Memory Device
	Array Handle: 0x0009
	Error Information Handle: Not Provided
	Total Width: 64 bits
	Data Width: 64 bits
	Size: 32 GB
	Form Factor: SODIMM
	Set: None
	Locator: Channel-0-DIMM-0
	Bank Locator: BANK 0
	Type: DDR4
	Type Detail: Unknown Synchronous
	Speed: 3200 MT/s
	Manufacturer: GSkill
	Serial Number: 10250255
	Asset Tag: Channel-0-DIMM-0-AssetTag
	Part Number: F4-3200C22-32GRS
	Rank: 2
	Configured Memory Speed: 3200 MT/s
	Minimum Voltage: 1.2 V
	Maximum Voltage: 1.2 V
	Configured Voltage: 1.2 V

Handle 0x000B, DMI type 17, 40 bytes
Memory Device
	Array Handle: 0x0009
	Error Information Handle: Not Provided
	Total Width: 64 bits
	Data Width: 64 bits
	Size: 32 GB
	Form Factor: SODIMM
	Set: None
	Locator: Channel-0-DIMM-0
	Bank Locator: BANK 0
	Type: DDR4
	Type Detail: Unknown Synchronous
	Speed: 3200 MT/s
	Manufacturer: GSkill
	Serial Number: 10250256
	Asset Tag: Channel-0-DIMM-0-AssetTag
	Part Number: F4-3200C22-32GRS
	Rank: 2
	Configured Memory Speed: 3200 MT/s
	Minimum Voltage: 1.2 V
	Maximum Voltage: 1.2 V
	Configured Voltage: 1.2 V
	```

majal, 2024-08-12:
Reporting the same for **gaze16**:

`sudo dmidecode | grep Locator`

outputs:

```
Locator: Channel-0-DIMM-0
Bank Locator: BANK 0
Locator: Channel-0-DIMM-0
Bank Locator: BANK 0
```