system76/firmware-open #340: New releases for all models

crawfxrd, 2022-07-30:
After galp6 is done, make new releases for all the boards.

- [ ] addw2
- [ ] bonw14
- [ ] darp5 (new)
- [ ] darp6
- [ ] darp7
- [ ] darp8
- [ ] galp3-c (new)
- [ ] galp4 (with open EC)
- [ ] galp5
- [ ] gaze15
- [ ] gaze16-3050
- [ ] gaze16-3060
- [ ] gaze16-3060-b
- [ ] gaze17-3050
- [ ] gaze17-3060-b
- [ ] lemp9
- [ ] lemp10
- [ ] lemp11
- [ ] oryp5 (new)
- [ ] oryp6
- [ ] oryp7
- [ ] oryp8
- [ ] oryp9

## Resolves

- #160
- #210
- #233
- #246
- #266
- #278
- #279
- #282
- #284
- #309

And probably others. It's been a while.

curiousercreative, 2022-10-18:
@crawfxrd perhaps relevant here, I just built and flashed the latest on my galp5 (for [developing this](https://github.com/pop-os/system76-acpi-dkms/pull/18)) and now I have a new issue that I suspect is a firmware regression. galp5 will suspend normally (blinking green LED but after about four minutes, the green LED goes solid and remains so.

curiousercreative, 2022-10-19:
@crawfxrd I apologize, I meant to return here and delete my comment as I commented [here](https://github.com/system76/firmware-open/issues/342#issuecomment-1282656131) on an existing issue.

larsblumberg, 2022-12-02:
Sorry if I hijack this issue. I was wondering if lemp10 receives any firmware update beyond the last official release `2021-07-20_93c2809`?

If not, is there any value for lemp10 owners if they flash the most recent official firmware release, i.e. `2022-11-21`?

leviport, 2022-12-02:
2021-07-20_93c2809 is the most current for lemp10 at the moment. Since it is shown in the checklist in this issue, an update is planned. Once it passes testing, it will be released.

larsblumberg, 2022-12-05:
I appreciate you taking the time to respond. And clarifying that the model being listed means it gets an updated. Thank you for our continued support. You are doing an amazing job at System76.

crawfxrd, 2023-01-23:
ADL boards will need to be updated to coreboot 4.19.

crawfxrd, 2023-05-31:
Most boards have updates to 2023-05-25_5608a8d staged.

x0wllaar, 2023-06-13:
Is there any timeframe for general availability of 2023-05-25_5608a8d for lemp11? I'm not getting it in the Firmware Manager GUI :(

jacobgkau, 2023-08-03:
> Is there any timeframe for general availability of 2023-05-25_5608a8d for lemp11? I'm not getting it in the Firmware Manager GUI :(

2023-07-19_05cfb1a for lemp11 was released today.

rvolosatovs, 2023-11-08:
Is there any (approximate) timeline for when galp5 could be updated? Interested in secure boot support specifically

marcin-rzeznicki, 2023-12-09:
#437 was opened by me on the July 11. It took you 2 months to pick it up. Allegedly, it was fixed on the 5th of September, so 4 months have already passed and counting, or 6th months if you count from the time the bug was reported. I will ask again for the nth time. **When are you finally going to release the update** that fixes this $%@#$@, namely [2023-09-08_42bf7a6](https://github.com/system76/firmware/pull/557)

curiousercreative, 2023-12-10:
Understandably you're frustrated with how long firmware bugs take to be looked at, fixed, and eventually released. My galp5 has at least one firmware bug that must be as old as the model itself, I get it. It may already be painfully obvious, but there is far more firmware development, maintenance, and QA demand than supply and, from what I can tell, a lot regressions come from upstream. I'll bet the bugfix you want released was a regression as a result of upstream changes. In my experience, the best way to expedite the release of a fix is to assist with QA. If you're so confident this is resolved, flash your firmware and report back. Many firmware release candidates have more regressions identified and you can help identify those protectively. If you're adventurous enough to flash your firmware, I and others can point you in the right direction.

cassanof, 2023-12-10:
> #437 was opened by me on the July 11. It took you 2 months to pick it up. Allegedly, it was fixed on the 5th of September, so 4 months have already passed and counting, or 6th months if you count [...]

RELAX, man it's not that hard... i had the issue too and i just looked at the source code of the firmware. you just need to place whatever boot loader you have. see: https://github.com/system76/firmware-open/issues/153

cassanof, 2023-12-10:
look, took me 3 minutes to find it: https://github.com/system76/firmware-open/blob/master/docs/uefi.md

aservedio, 2023-12-10:
Personally I share the frustration. The system76 laptop i bought was the most amount of money I ever put in a laptop.

This issue, the fixes those firmware updates contains and its delays into getting it done, is always painful reminder of my mistake with that purchase.

Here's a laptop, thank you for your money, and make sure to not keep it too long coz we won't care about it way faster then its life expectency.

cassanof, 2023-12-10:
> Personally I share the frustration. The system76 laptop i bought was the most amount of money I ever put in a laptop.
>
> This issue, the fixes those firmware updates contains and its delays into [...]

I can't disagree more. I've been a customer for almost 4 years. This is a tinkerer's laptop, if you have issues, you can fix them yourself; if you don't like to hack around, then sure, i can see how this may be frustrating. Every part inside your laptop can be replaced if it's faulty, and system76 provides clear instructions on how to. Every single component in your laptop is customizable from a software and hardware point of view.

aservedio, 2023-12-10:
I don't know how to fully add secureboot/tpm to corebooy bios to install windows 11.

Over a few thinkering sessions I learned how to do it with regedit but then guess what happens if I try to sell the laptop and start explaining that.

I like thinkering, but I would assume at that price tag you'd be able to install windows 11 by now without regedit

devoidfury, 2023-12-10:
I have an oryp6 that used to be my daily driver. It has problems switching between different graphics modes, compute mode doesn't work at all, and no matter which mode I put it the nvidia drivers never seemed to fully work quite and never got it to work with ML tooling. And really using the card at all makes the machine very loud, the loudest computer I've ever used. After a few years of waiting for firmware updates to try and resolve graphics bugs, switching different distros and kernels, flashing it again, trying different driver versions, looking for workarounds... I went and got a replacement from a different vendor.

I'd be happy to pull it off the shelf and flash firmware on it to help with testing and moving this along if it helps these things work better; I like what you're doing with linux and really want to like system76, but honestly the experience with that laptop put me off it for now. I can't be the only one, maybe if you asked for help with QA in the blog or a newsletter you'd find some more volunteers.

curiousercreative, 2023-12-10:
@devoidfury can't speak for S76, but I'm sure an oryp6 has had the firmware flashed and appears to be working fine. It's harder to gain confidence that everything under the sun still functions as expected (suspend/resume, Thunderbolt, CPU and GPU PM, etc, etc). All to say, it's most helpful if many adventurous users are daily driving it across some variety of use cases.

marcin-rzeznicki, 2023-12-10:
> If you're adventurous enough to flash your firmware, I and others can point you in the right direction.

Thanks for the offer, that's very kind of you, but I have already done it - I think I can find my way around. I understand regressions, but even if there are unforeseen problems/bugs, I'd love to have a clear communication e.g. a GH issue or anything that people can track. Everything is better than an unticked checkbox lying dormant for months

marcin-rzeznicki, 2023-12-10:
> 
> RELAX, man it's not that hard... i had the issue too and i just looked at the source code of the firmware. you just need to place whatever boot loader you have into a specific path, i forget which one. see: #153

The bug is that NVRAM variables are reset randomly - this means that the boot loader path, boot order, timeout and other items are gone and you are booting from something random e.g. from an USB stick you happen to have inserted, or PopOS recovery partition. But usually not from what you want to boot

cassanof, 2023-12-10:
> The bug is that NVRAM variables are reset randomly - this means that the boot loader path, boot order, timeout and other items are gone and you are booting from something random e.g. from an USB [...]

Yes, I know. You don't *need* NVRAM variables; there is a hardcoded efi path you can use (`\EFI\BOOT\BOOTX64.EFI`)

jackpot51, 2023-12-10:
We'll look into the remaining updates not yet released. Most of our products have up to date firmware, a few updates were held due to regressions. Please try to keep discussion on topic, new issues can be created for unrelated items or discussion about old issues can take place om those issues themselves, even if they are already closed.

crawfxrd, 2024-05-06:
TODO:

- bonw14
- darp6
- gaze16-3050
- gaze16-3060
- gaze16-3060-b

closed 2024-05-06 by crawfxrd

yochananmarqos, 2024-05-06:
@crawfxrd Adding a TODO then closing the issue is a little confusing...