system76/firmware-open #342: lemp11: Sometimes fails to suspend on battery power

crawfxrd, 2022-08-02:
- Model: lemp11
- BIOS version: 2022-07-05_62f4da9
- EC version: 2022-07-05_62f4da9
- OS: Pop!_OS 22.04

The CPU sometime fails to go to C10 during S0ix while on battery power.

### Steps to reproduce

- `sudo systemctl suspend`

### Expected behavior

System suspends and power LED blinks

### Actual behavior

System suspends but CPU only goes to C3 at best.

### Additional info

Issue does not appear to be present if connected to AC power.

XV-02, 2022-08-17:
I have noticed what appears to be this issue presenting with AC attached. If there is guidance on diagnostic checks to confirm it is the same issue, that would be appreciated.

crawfxrd, 2022-08-17:
What is your hardware configuration?

Ensure it's not due to the RTD3 issues caused by certain drives.

XV-02, 2022-08-17:
It's a Lemp11, integrated ram only, running a 250GB 970 EVO Plus.

crawfxrd, 2022-08-17:
This is some bullshit. Unplug it and plug it back in. That makes it work for me.

codewiz, 2022-08-19:
> This is some bullshit. Unplug it and plug it back in. That makes it work for me.

Yup! Plugging and unpligging AC while the lid is closed makes it go to sleep (lemp11 running coreboot 2022-08-03_26f97c8).

thomas-zimmerman, 2022-09-06:
I think I was able to recreate this with a customer; 

We had turned on the fan with Fn+1 as we thought this would allow us to verify suspend state--ie, the fan would turn off when suspend is reached. The fun begins when you have the fan on:

Suspend system : LED will be solid green and fan running. Close lid and LED will start blinking showing suspend state
check output of `cat /sys/devices/system/cpu/cpuidle/low_power_idle_system_residency_us` before and after suspend--increases in this number show the system was reaching S0ix suspend states.

We are able to turn off the fan test state with Fn+1 while in suspend but not turn it back on.

Once the fan toggle has been made, the system seems to fail to reach full suspend state more* often.

loganbecket, 2022-09-14:
> * `sudo systemctl suspend`

I'm experiencing the same behavior.  Absolutely infuriating.  Plugging and then unplugging does get me into suspend mode, but that's hardly a solution because at that point I might as well leave it plugged in.  My entire motivation for suspending is to save battery while unplugged.

djlynux, 2022-09-17:
I can repro the same issue on my Lemp11 system as well. Logging this feedback as part of the tracking.

jackpot51, 2022-09-26:
@crawfxrd would https://github.com/system76/coreboot/pull/138 help here?

ceccopierangiolieugenio, 2022-09-28:
Same issue, 
Is it there a workaround to force the suspend state?

louffoster, 2022-10-08:
I have a new lemp11  with Pop_OS 22.04, base memory and the 250G drive. I  have the same issue. The only way I can get the system to actually suspend 100% of the time is to close the lid, plug into power, then remove the plug. Nothing else is consistent. Sometimes suspending from the terminal or super+esc, then closing the lid will do it, but not always.

leminhman0312, 2022-10-13:
> I have a new lemp11  with Pop_OS 22.04, base memory and the 250G drive. I  have the same issue. The only way I can get the system to actually suspend 100% of the time is to close the lid, plug into [...]

Yes,  I do have the same issue.  Without the LED light, I would not know if the laptop fully suspends.  It's supposed to be blinking yellow, not solid yellow.

curiousercreative, 2022-10-18:
Just a note that TGL-U also appears to be affected if running the lastest unreleased firmware. My galp5 is running firmware built from source today and the only difference is that the system suspends properly, but only for about four minutes and then LED goes solid. The tricks of lid closing or adding AC power seem to work here as well.

ygingras, 2022-10-19:
Same problem here with a lemp11 running a 12th Gen Intel i7-1255U CPU. I see the problem with kernel 5.19.0. Is this something that is supposed to be fixed with Linux 6.0? Ubuntu 22.10 is still on Kernel 5.19, but there are PPA for Linux 6.0 for it that might be a viable solution.

leviport, 2022-10-19:
>Is this something that is supposed to be fixed with Linux 6.0?

The kernel is not to blame. A firmware update to fix it is in the works. If it passes testing, you will be seeing it arrive soon.

As far as workarounds, it seems to suspend if you either plug in and unplug the charger, or use the Fn+1 hotkey to toggle the fans to full. The fans will have to be toggled back to "auto" upon resume, as you'll probably notice them running at full when you wake it up.

curiousercreative, 2022-10-19:
@crawfxrd responding to [this comment about suspend working fine with lid open](https://github.com/system76/firmware-open/issues/340#issuecomment-1284258151), for me it doesn't matter. No matte how I get to suspend, it comes out (but not into OS) after four minutes.

curiousercreative, 2022-10-22:
An update for galp5: I actually can't remain in suspend on batter power with any of these tricks. Closing the lid or pressing the power button only yields four more minutes of suspend before LED goes solid again. For now, I"m forced to power off or connect to power.

crawfxrd, 2022-10-22:
galp5 should be a regression in coreboot. I tested 2021-07-20_93c2809 with just the `SLP_S0#` change, and 2021-07-20_93c2809 with EC master, and both worked.

curiousercreative, 2022-10-23:
> galp5 should be a regression in coreboot

@crawfxrd I didn't know where in the firmware stack was causing this issue. So for ADL this is an EC bug, but for galp5 you think it's coreboot? Do you have an issue open for what I'm experiencing with galp5? Anything I can do to assist there?

crawfxrd, 2022-10-24:
I expect the issue (for all affected models) to be coreboot configs and/or FSP configs.

galp6 is now also apparently affected on `master`, while it was working at release.

You're the first person to report the problem with galp5, so there's no issue open for it.

codewiz, 2022-11-05:
I'd like to test the fix that was merged by @curiousercreative before the next release. Are there any nightly builds I could flash on a lemp11, or do I have to build from sources?

crawfxrd, 2022-11-23:
Fixes and workarounds for S0ix released in 2022-11-21_b337ac6.

closed 2022-11-23 by crawfxrd

codewiz, 2022-11-24:
> Fixes and workarounds for S0ix released in 2022-11-21_b337ac6.

I just flashed this version on a lemp11, and this bug is still present. In fact, it 's worse: before, I could get the laptop to enter suspend mode by plugging and unplugging a USB-C charger while the lid is closed. Now, the LED remains solid green no matter what I do.

```
bernie@giskard:~% sudo dmidecode | grep Version
        Version: 2022-11-21_b337ac6
        Version: lemp11
        Version: lemp11
        Version: Not Specified
        Version: 12th Gen Intel(R) Core(TM) i7-1255U
bernie@giskard:~% sudo system76_ectool info
board: system76/lemp11
version: 2022-11-21_b337ac6
```

crawfxrd, 2022-11-25:
What is your hardware configuration?

codewiz, 2022-11-25:
> What is your hardware configuration?

lemp11 with i7-1255U and 16GB ram, running kernel 6.0.9-300 (Fedora 37).

lspci:
```
00:00.0 Host bridge: Intel Corporation Device 4601 (rev 04)
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-UP3 GT2 [Iris Xe Graphics] (rev 0c)
00:06.0 PCI bridge: Intel Corporation 12th Gen Core Processor PCI Express x4 Controller #0 (rev 04)
00:07.0 PCI bridge: Intel Corporation Alder Lake-P Thunderbolt 4 PCI Express Root Port #0 (rev 04)
00:0a.0 Signal processing controller: Intel Corporation Platform Monitoring Technology (rev 01)
00:0d.0 USB controller: Intel Corporation Alder Lake-P Thunderbolt 4 USB Controller (rev 04)
00:0d.2 USB controller: Intel Corporation Alder Lake-P Thunderbolt 4 NHI #0 (rev 04)
00:14.0 USB controller: Intel Corporation Alder Lake PCH USB 3.2 xHCI Host Controller (rev 01)
00:14.2 RAM memory: Intel Corporation Alder Lake PCH Shared SRAM (rev 01)
00:14.3 Network controller: Intel Corporation Alder Lake-P PCH CNVi WiFi (rev 01)
00:15.0 Serial bus controller: Intel Corporation Alder Lake PCH Serial IO I2C Controller #0 (rev 01)
00:15.1 Serial bus controller: Intel Corporation Alder Lake PCH Serial IO I2C Controller #1 (rev 01)
00:16.0 Communication controller: Intel Corporation Alder Lake PCH HECI Controller (rev 01)
00:1c.0 PCI bridge: Intel Corporation Device 51bd (rev 01)
00:1f.0 ISA bridge: Intel Corporation Alder Lake PCH eSPI Controller (rev 01)
00:1f.3 Audio device: Intel Corporation Alder Lake PCH-P High Definition Audio Controller (rev 01)
00:1f.4 SMBus: Intel Corporation Alder Lake PCH-P SMBus Host Controller (rev 01)
00:1f.5 Serial bus controller: Intel Corporation Alder Lake-P PCH SPI Controller (rev 01)
01:00.0 Non-Volatile memory controller: Phison Electronics Corporation E12 NVMe Controller (rev 01)
2d:00.0 SD Host controller: O2 Micro, Inc. SD/MMC Card Reader Controller (rev 01)
```

lsusb (I also tried unplugging the yubikey before suspend):
```
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 5986:214c Acer, Inc BisonCam,NB Pro
Bus 003 Device 005: ID 1050:0200 Yubico.com Gnubby U2F
Bus 003 Device 004: ID 8087:0026 Intel Corp. AX201 Bluetooth
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

dmesg during a suspend/resume cycle:
```
*LID CLOSED*

[ 9020.439086] wlp0s20f3: deauthenticating from xx:xx:xx:xx:xx:xx by local choice (Reason: 3=DEAUTH_LEAVING)
[ 9020.930079] PM: suspend entry (s2idle)
[ 9020.936810] Filesystems sync: 0.006 seconds
[ 9021.263766] Freezing user space processes ... (elapsed 0.004 seconds) done.
[ 9021.267927] OOM killer disabled.
[ 9021.267930] Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
[ 9021.269915] printk: Suspending console(s) (use no_console_suspend to debug)
[ 9022.146280] PM: suspend devices took 0.876 seconds
[ 9022.186652] ACPI: EC: interrupt blocked

*LID OPENED*

[ 9026.186591] ACPI: EC: interrupt unblocked
[ 9026.256183] ACPI Error: No handler or method for GPE 6B, disabling event (20220331/evgpe-839)
[ 9026.279585] i915 0000:00:02.0: [drm] GuC firmware i915/adlp_guc_70.1.1.bin version 70.1
[ 9026.279595] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc_7.9.3.bin version 7.9
[ 9026.296372] i915 0000:00:02.0: [drm] HuC authenticated
[ 9026.297443] i915 0000:00:02.0: [drm] GuC submission enabled
[ 9026.297447] i915 0000:00:02.0: [drm] GuC SLPC enabled
[ 9026.298297] i915 0000:00:02.0: [drm] GuC RC: enabled
[ 9026.596360] PM: resume devices took 0.326 seconds
[ 9026.596382] OOM killer enabled.
[ 9026.596385] Restarting tasks ... done.
[ 9026.600246] random: crng reseeded on system resumption
[ 9026.604555] mei_hdcp 0000:00:16.0-XXXX: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[ 9026.605973] mei_pxp 0000:00:16.0-XXXX: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[ 9026.678831] video LNXVIDEO:00: Restoring backlight state
[ 9026.685986] PM: suspend exit
[ 9030.460234] wlp0s20f3: authenticate with xx:xx:xx:xx:xx:xx
[ 9030.465634] wlp0s20f3: bad VHT capabilities, disabling VHT
[ 9030.465637] wlp0s20f3: Invalid HE elem, Disable HE
[ 9030.465639] wlp0s20f3: 80 MHz not supported, disabling VHT
[ 9030.476846] wlp0s20f3: send auth to xx:xx:xx:xx:xx:xx (try 1/3)
[ 9030.530708] wlp0s20f3: authenticated
[ 9030.537673] wlp0s20f3: associate with xx:xx:xx:xx:xx:xx (try 1/3)
[ 9030.542866] wlp0s20f3: RX AssocResp from xx:xx:xx:xx:xx:xx (capab=0x431 status=0 aid=1)
[ 9030.552296] wlp0s20f3: associated
[ 9033.548478] wlp0s20f3: disassociated from xx:xx:xx:xx:xx:xx (Reason: 2=PREV_AUTH_NOT_VALID)
```

crawfxrd, 2022-11-25:
> 01:00.0 Non-Volatile memory controller: Phison Electronics Corporation E12 NVMe Controller (rev 01)

What is the exact model? Is this the drive that came with the unit? Do you have another drive you can test with?

codewiz, 2022-11-25:
> What is the exact model? Is this the drive that came with the unit?

It's a Sabrent Rocket 2280 2TB, with firmware RKT303.2. This device was previously housed in a Thinkpad X1, and suspend / resume worked fine.

> Do you have another drive you can test with?

I still have the original NVMe drive that came with the lemp11 and I could try swapping it back tomorrow.

cmm, 2022-11-25:
did you disable Intel ME at runtime using `nvmtool`?  I had to re-enable it for suspend to finally start working.

codewiz, 2022-11-25:
> did you disable Intel ME at runtime using `nvramtool`? I had to re-enable it for suspend to finally start working.

It was enabled. I disabled it with `./nvramtool -w me_state=Disable` and rebooted. No effect, it still won't suspend.

codewiz, 2022-11-27:
@crawfxrd  wrote:
> > Do you have another drive you can test with?
> 
> I still have the original NVMe drive that came with the lemp11 and I could try swapping it back tomorrow.

I restored the NVMe drive that came from System76 and booted into the original `Pop!_OS` image. The suspend bug is still present in this configuration.

crawfxrd, 2022-11-27:
If you disabled CSME you will need to reenable it for S0ix to work.

codewiz, 2022-11-29:
> If you disabled CSME you will need to reenable it for S0ix to work.

Oh, are these the Modern Standby states? With ThinkPads, S0i3 would drain the battery in 1-2 days, so I would disable it in the BIOS and use regular S3 instead. Is there an equivalent toggle for coreboot?

Anyway, I re-enabled the ME, rebooted, and now the laptop enters sleep when I close the lid and then plug/unplug the USB-C charger. This is with the Sabrent Rocket 2280.

I'm not sure why sleep had stopped working altogether after updating coreboot, even with the ME was enabled in NVRAM.

crawfxrd, 2022-11-29:
> Is there an equivalent toggle for coreboot?

No, S3 is not supported on ADL due to #344.

codewiz, 2022-12-19:
I'm not sure what exactly fixed it, but now my lemp11 goes to sleep a few seconds after closing the lid.
It's been working for a couple of weeks, but I thought I'd leave a comment for the record.

Perhaps, when I swapped back in the Sabrent Rocket drive I used the secondary NVMe slot. Could that have helped?

mirsev, 2023-04-27:
@crawfxrd, I have a very same issue with oryp10. Is there any schedule to release an updated firmware version?

leviport, 2023-04-27:
@mirsev Yes, once we're done with QA testing the new release. Releasing firmware that has regressions isn't much fun for anyone.