system76/firmware-open #344: adl: S3 suspend powers off on 13th cycle

crawfxrd, 2022-08-03:
- Model: ADL
- BIOS version:
- EC version:
- OS: Pop!_OS 22.04

We want to use S3 on ADL to allow disabling the CSME at runtime.

### Steps to reproduce

```
sudo fwts s3 --s3-multiple 20 --s3-min-delay 10 --s3-max-delay 15
```

### Expected behavior

- System completes fwts test

### Actual behavior

- System powers off on 13th cycle

### Additional info

Enable S3 with:

```diff
diff --git a/src/mainboard/system76/adl-p/Kconfig b/src/mainboard/system76/adl-p/Kconfig
index de6ba4bf1314..d81cb114ea24 100644
--- a/src/mainboard/system76/adl-p/Kconfig
+++ b/src/mainboard/system76/adl-p/Kconfig
@@ -12,6 +12,7 @@ config BOARD_SPECIFIC_OPTIONS
        select EC_SYSTEM76_EC_COLOR_KEYBOARD if BOARD_SYSTEM76_DARP8 || BOARD_SYSTEM76_ORYP9
        select EC_SYSTEM76_EC_DGPU if BOARD_SYSTEM76_ORYP9
        select HAVE_ACPI_TABLES
+       select HAVE_ACPI_RESUME
        select HAVE_CMOS_DEFAULT
        select HAVE_OPTION_TABLE
        select HAVE_SPD_IN_CBFS if BOARD_SYSTEM76_LEMP11
@@ -21,6 +22,7 @@ config BOARD_SPECIFIC_OPTIONS
        select MEMORY_MAPPED_TPM
        select NO_UART_ON_SUPERIO
        select SOC_INTEL_ALDERLAKE_PCH_P
+       select SOC_INTEL_ALDERLAKE_S3
        select SOC_INTEL_COMMON_BLOCK_HDA_VERB
        select SOC_INTEL_CRASHLOG
        select SOC_INTEL_ENABLE_USB4_PCIE_RESOURCES
diff --git a/src/mainboard/system76/adl-p/devicetree.cb b/src/mainboard/system76/adl-p/devicetree.cb
index 6285851b684c..7f2bc2a90693 100644
--- a/src/mainboard/system76/adl-p/devicetree.cb
+++ b/src/mainboard/system76/adl-p/devicetree.cb
@@ -11,8 +11,6 @@ chip soc/intel/alderlake
        # Enable Enhanced Intel SpeedStep
        register "eist_enable" = "1"
 
-       register "s0ix_enable" = "1"
-
        # Enable C6 DRAM
        register "enable_c6dram" = "1"
 
 ```

crawfxrd, 2022-08-11:
Logs for the power off: https://gist.github.com/crawfxrd/533bf51531a7554ba522e60aa72d9923

This occurs regularly:

```
[CRIT ]  CRITICAL ERROR: AP call expired. 19/19 CPUs accepted.
[ERROR]  MTRR programming failure
```

This happens on the resume that will powers off:

```
[EMERG]  ASSERTION ERROR: file 'src/commonlib/bsd/cbfs_mcache.c', line 106
[ERROR]  CBFS ERROR: CBFS mcache is not terminated!
[ERROR]  CBFS ERROR: error -1 when looking up 'cpu_microcode_blob.bin'
[WARN ]  microcode: failed because no ucode was found
```

curiousercreative, 2022-11-29:
I'll never cease to be surprised by the absurdity of the firmware bugs you have to deal with.

crawfxrd, 2023-02-27:
@jackpot51 has found that disabling the CBFS cache avoids the issue on RPL.

```diff
diff --git a/src/mainboard/system76/adl-p/Kconfig b/src/mainboard/system76/adl-p/Kconfig
index 80c42a80b260..58b5ebaeed59 100644
--- a/src/mainboard/system76/adl-p/Kconfig
+++ b/src/mainboard/system76/adl-p/Kconfig
@@ -13,8 +13,10 @@ config BOARD_SPECIFIC_OPTIONS
        select HAVE_OPTION_TABLE
        select INTEL_GMA_HAVE_VBT
        select INTEL_LPSS_UART_FOR_CONSOLE
+       select MAINBOARD_DISABLE_STAGE_CACHE
        select MAINBOARD_HAS_TPM2
        select MEMORY_MAPPED_TPM
+       select NO_CBFS_MCACHE
        select NO_UART_ON_SUPERIO
        select SOC_INTEL_ALDERLAKE_PCH_P
        select SOC_INTEL_COMMON_BLOCK_HDA_VERB
```

leedenison, 2023-04-17:
@crawfxrd - Please could you give a status update on the fix.  Did it work?  Is it scheduled for release?

jackpot51, 2023-04-17:
We have a fix that we will deploy along with enabling S3 suspend to affected models as new firmware passes QA testing.

czepiec, 2023-04-20:
I see S3 is back in [78bfb39 ](https://github.com/system76/coreboot/commit/78bfb3931a00f17281bff1ece0ab896e302fa9b3), I've been testing it for a week on my lemp11 (using dfd1b73) and it works perfectly.

Many thanks, I use an OS that does not support S0ix and until now I could not use sleep, which was very inconvenient.

crawfxrd, 2023-05-10:
Fixed in our fork of coreboot. Upstream was debating on what the correct behavior should even be (ref: [CB:74138](https://review.coreboot.org/c/coreboot/+/74138)).

closed 2023-05-10 by crawfxrd