system76/firmware-open #349: gaze17-3060-b WD Blue SN550 NVMe resumes in read only mode

linuxgnuru, 2022-08-12:
- Model: gaze17-3060-b
- BIOS version: 2022-06-07_090f9e0
- EC version: 2022-06-07_090f9e0
- OS: Pop!_OS 22.04

If laptop is suspended with the Western Digital Blue SN550 NVMe drive on resume the drive will be put into read only mode.

### Steps to reproduce

Install drive into either M.2 slot and install Pop 22.04 then suspend system and resume. mount and dmesg will show errors

### Expected behavior

Drive should be continue to be mounted in read write mode.

### Actual behavior

Drive mounted in read only mode giving errors with journalctl and other services not being able to write to drive until the system is rebooted.

### Additional info

Tested five times with 5.19 and 5.18 kernels as well as gaze17-3050 only the gaze17-3060-b had the issue.

crawfxrd, 2022-08-12:
This is fixed in master already, correct?

linuxgnuru, 2022-08-15:
while testing the 5.19 kernel we ran into this but only on the gaze17-3060-b. Same issue in 5.18 so not a regression

crawfxrd, 2022-08-15:
And I'm pretty sure it was fixed with https://github.com/system76/coreboot/pull/120 (system76/coreboot@4244f4c3d17b).

linuxgnuru, 2022-08-15:
sorry, I misunderstood, let me check

closed 2022-08-15 by linuxgnuru