system76/firmware-open #354: `darp8` Can get stuck in the in power off state

thomas-zimmerman, 2022-09-09:
- Model: `darp8
- BIOS version: 2022-07-13_8b8344a
- EC version: 2022-07-13_8b8344a
- OS: Pop!OS 22.04

1. if the laptop is discharged almost completely 
2. I attempt to boot
3. I enter my password to decrypt the disk
4. the battery dies during decryption 
5. I plug the laptop in or fully charge the battery
6. the laptop will not boot.
     * the battery light is green the power light is orange. 
     * the only way to boot after this is to reset the cmos battery.

### Expected behavior

Even with battery power loss, power state tracking should not lost.

### Additional info

This is rare but support has rare reports of systems that require CMOS reset to get back on.

thomas-zimmerman, 2022-10-12:
As an update we continue to see this rarely on `darp8` but have somewhat random reports on `galp6` as well.

sanchoPantser, 2022-10-24:
I've experienced some similar behavior, but on attempting to  **wake from suspend** with battery run down:

- darp8:
- coreboot version: 2022-07-13_8b8344a
- pop-os version 22.04 LTS
- Intel 12th Gen i5-1240Px16

On at least two occurrences, I've let the battery run down while OS was suspended and the following occurred:

1. I have two yellow LED's on the right
2. I plug in to AC adapter (barrel port, not USB-C power)
3. The machine will not wake with any sequence of key strokes, nor short or a long-press on the suspend/power button.
4. I allow the battery to charge fully and get one green LED and one yellow on the right side
5. Still no response to any input. 
6. The only solution is to do a CMOS reset.

After the first instance of this (in the first week out of factory) I was not able to reproduce this despite draining the battery at least 10 times in a row. With support from System76, we determined that it may have been a one-off instance. The next time this happened was 5 months later. I have not tried to reproduce it now, but will do so in the coming weeks. 

I don't see any correlation to anything that may have caused this to recur.

leviport, 2022-10-24:
If you see two yellow LEDs, the machine is powered off. It may have run out of battery while it was suspended and turned off.

sanchoPantser, 2022-10-24:
Fair enough. But there is no powering it back on from this state. So this puts me in the same condition as OP's Item 6. The failure is from POWER OFF condition rather than SUSPEND .

xonq, 2022-10-26:
getting this condition too. happens if I put the laptop to sleep, poor battery life while sleeping leads to it completely draining; then when I try to boot with 0% occasionally it will boot, then need the CMOS reset.

effectively can't use sleep mode on the laptop because 1) it will drain the battery significantly and 2) if it completely depletes then I have to take off the back cover

XV-02, 2022-11-11:
Testing with the lab unit confirms that this can be replicated. 
Loss of power during the decryption process itself results in the system acting as described. If power is lost before the decryption password is passed, or after the process has finished and handed off, the system will boot correctly.

crawfxrd, 2022-11-14:
Should see if I can prevent powering-on from battery power when capacity <5%, and blink the LED in a different sequence.

Can also enable the power switch WDT to allow reset by holding the power button for 10 seconds.

---

If left unplugged, does the power LED blink orange?

That is what I am seeing on galp6. The EC is in S5 instead of OFF:

```
72: Power switch press
power_state = 1
96: Power switch release
```

Indicating `EC_RSMRST` is still asserted after failing.

xonq, 2022-11-14:
Yes, if 

> Should see if I can prevent powering-on from battery power when capacity <5%, and blink the LED in a different sequence.
> 
> Can also enable the power switch WDT to allow reset by holding the power button for 10 seconds.
> 
> If left unplugged, does the power LED blink orange?
> 
> That is what I am seeing on galp6. The EC is in S5 instead of OFF:
> 
> ```
> 72: Power switch press
> power_state = 1
> 96: Power switch release
> ```
> 
> Indicating `EC_RSMRST` is still asserted after failing.
> 
> May be fixed by [system76/ec#276](https://github.com/system76/ec/pull/276)?

Yes, LED closest to screen blinks orange *after* power on failure triggers the CMOS issue

XV-02, 2022-11-17:
It looks like https://github.com/system76/ec/pull/276 is remedying this issue. 
Reliably, with currently released firmware, system shutdown due to lack of power in the middle of the LUKS decryption process would render the machine incapable of being booted, regardless of how charged the battery was, or if the machine was on AC or not. 
Attempting that mode of failure ~10 times with the change in https://github.com/system76/ec/pull/276 present did not result in an unbootable system.

crawfxrd, 2022-12-01:
Fix released in 2022-11-21_b337ac6.

closed 2022-12-01 by crawfxrd

xonq, 2022-12-28:
Happened again - had to reset the CMOS on my darp8 again to boot using system76-firmware 1.0.43-1 and system76-power 1.1.24-1. are these "fixed" versions?

sanchoPantser, 2022-12-28:
Where are you getting those s76 Firmware versions? I've updated to versions below and have not experienced a recurrence of this issue. As seen on my machine at Settings > Firmware:
[image]

xonq, 2022-12-28:
my bad, I assumed the arch package was the firmware, not a firmware installer, so I assume I did not actually update. I am updating now.

thomas-zimmerman, 2022-12-28:
@xonq If you're having any issues getting the new firmware, please reach out to support. We can provide a USB image to update firmware on request.