system76/firmware-open #356: oryp8: Not charging

crawfxrd, 2022-09-13:
- Model: oryp8
- BIOS version: 2022-09-07_74e75ff
- EC version: 2022-09-07_74e75ff
- OS: any

The system is not charging on the new release.

### Steps to reproduce

1. Discharge battery to <90%
2. Plug in battery to charge

### Expected behavior

The system immediately starts charging the battery.

### Actual behavior

The system reports "Not Charging" for ~10 seconds, then charges at a very slow rate.

### Additional info

I reverted the EC firmware to the previous release and it still has the problem.

Kill-A-Watt reports 0.1A and 8W while the system is off.

crawfxrd, 2022-09-13:
Kill-A-Watt is reporting 0.24A / 25W will in S3 or S5, which are the expected values.

crawfxrd, 2022-09-13:
On 2021-09-23_b9b0e89:

```
[oryp8 ~]$ cat /sys/class/power_supply/BAT0/status 
Charging
[oryp8 ~]$ cat /sys/class/power_supply/BAT0/current_now 
1490000
```

`upower -d` reports `pending-charge` and GNOME GUI shows "Not Charging" for a while, but eventually updates.

crawfxrd, 2022-09-13:
On 2022-09-07_74e75ff:

- `status` changing between "Not Charging" and "Charging several times
- `current_now` swings between 97000 and 625000 (so, not even 1A)

crawfxrd, 2022-09-14:
I flashed back to 2021-09-23_b9b0e89 and it still has the bad behavior.

crawfxrd, 2022-09-14:
Works: system76/ec@20d40f347741
Doesn't work: system76/ec@b78631e31606

So caused by: system76/ec#288

ahoneybun, 2022-10-06:
These are from a customer:

Steps Taken:

1.) upower -d says charge is at 16%.
2.) powered down
3.) unplugged power plug and plugged it back in
4.) left for 5 minutes
5.) powered back on
6.) upower -d says power is at 20%

closed 2022-10-06 by leviport

coyotte508, 2022-10-07:
Is there an ETA for the firmware upgrade to be available to the public? I don't want to manually mess with the firmware to downgrade it or upgrade it.

leviport, 2022-10-07:
Probably next week

JordanCClark, 2022-10-19:
Bumping for an ETA, as I don't know if it got past QC. ;)

JordanCClark, 2022-10-20:
Came through this evening, Thanks, eveyone!