system76/firmware-open #363: Lemp11: resume from hibernation always go to blackscreen if not using defaults (Pop OS + Pop OS's Gnome)

leminhman0312, 2022-10-12:
**Distribution:** 
NAME="Pop!_OS"
VERSION="22.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 22.04 LTS"
VERSION_ID="22.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=jammy
UBUNTU_CODENAME=jammy
LOGO=distributor-logo-pop-os

 
 
 
**Issue/Bug Description:**
Stock installation of Pop OS on Lemur Pro (lemp11).  Hibernation/resume works with default setup.  Hibernation seems to work with other desktop environment / window manager (XFCE, i3wm..).  Unfortunately, resume doesnt seem to work properly.  After authenticating with the luks encryption, the machine go straight to a black screen.  I figured this must be the screen where GDM is, because blindly typing the password and press Enter would seem to unlock the session.  After unlocking, I see the black screen again, so the actual desktop environment is also a  black screen.  Keyboard backlights work and volume control  also gives off sound to.  So, I am logged in, just not be able to see anything due to the black screen. 

**Steps to reproduce (if you know):**
1. Fresh installation of Pop OS 
2. Follow instructions here to enable hibernation: https://support.system76.com/articles/enable-hibernation/
3. Test to see if hibernation/resume works with stock installation. 
4. Download another desktop environment, say XFCE on Pop OS. 
5. Try to hibernate/resume again.  The black screen will show up. 
6.  Try other distro.  Resume from hibernation always fails. 

**Expected behavior:**
Regardless of desktop environments, hibernation/resume should work because it is a system wide settings, just like shutdown, reboot.  I should be able to resume from hibernation and see my files/window configuration are exactly where they are before hibernation.  I should be able to use my desktop environment properly (no black screen), just like before I attempted hibernation. 

**Other Notes:**
A quick look at the log file via journalctl shows the errors seem to be with i915:

```
Oct 08 13:03:49 pop-os systemd-journald[587]: Missed 118 kernel messages
Oct 08 13:03:49 pop-os kernel: i915 0000:00:02.0: [drm] *ERROR* Fault errors on pipe A: 0x00000080

```

There is a lot of these errors during the resume process.  

Hibernation is done via the systemctl command: `sudo systemctl hibernate`.
This is similar to reboot/shutdown and those command work: `sudo systemctl reboot`, `sudo systemctl shutdown`

x0wllaar, 2022-10-18:
This seems to be a problem with the i915 driver https://gitlab.freedesktop.org/drm/intel/-/issues/6506

jacobgkau, 2022-10-18:
This is a duplicate of the other issue you opened: https://github.com/pop-os/pop/issues/2640

adeptg, 2022-11-06:
I updated to 6.0.3 (from PopOS) - and it didn't help me with this issue on KDE Neon

x0wllaar, 2022-11-06:
Can you test it on 6.0.7 from https://kernel.ubuntu.com/~kernel-ppa/mainline/? I'm not sure if the backport made it to Pop's kernel

adeptg, 2022-11-06:
No, unfortunately, 6.0.7 didn't help me

adeptg, 2022-11-15:
I've also checked with 5.15 (default 22.04 kernel) on KDE Neon, and it also has the same issue