system76/firmware-open #367: `galp6` can get stuck in EC state with system off but power LED light as if it's on.

thomas-zimmerman, 2022-10-21:
- Model: `galp6`
- BIOS version: 2022-08-03_26f97c8
- EC version: 2022-08-03_26f97c8
- OS: Ubuntu 22.04 LTS

Can't get the system to power on. May have been in suspend or low battery. The power LED is light green but the system does not seem to be on. Pressing the power button does not change the system. LED goes out if unplugged and power button still is not responsive. 

If system can be opened, pulling the battery and the CMOS battery is able to reset the system so it can power back on.

### Steps to reproduce

No known method to get the system in this state yet. 

### Expected behavior

Able to power the system on or wake the system from suspend.

ohitsdylan, 2022-10-31:
I had this issue occur on my `galp6` as well.

I had the device in suspend for a few days until the battery was low enough for the system to power off. I pressed the power button and the system booted, all the way past unencrypting my SDD (I was able to enter my encryption key). The screen went black and became unresponsive. I figured the device had attempted to boot but ran out of power. I plugged in the battery charger.

Pressing the power button did not boot the system. I had to remove the bottom cover and reseat the battery connection. I did not remove the CMOS battery.

It has been a few days since this happened and I've been keeping an eye on the device when I put it into suspend mode. So far, no issues.

crawfxrd, 2022-11-23:
Needs testing on 2022-11-21_b337ac.

brs17, 2022-12-13:
Issue should be resolved with latest firmware that was released last week: `2022-11-21_b337ac6`.

closed 2022-12-13 by brs17