system76/firmware-open #368: galp5: fails to remain suspended for more than four minutes on battery power

curiousercreative, 2022-10-24:
- Model: galp5 (1650 Ti dGPU)
- BIOS version: 2022-10-23_2021-07-20_93c2809-99-gc82d2cd (master branch)
- EC version: master (reproducible on 2021-07-20)
- OS: Pop!_OS 22.04

Laptop successfully enters suspend (LED blinking green) but exits after about four minutes. Logs show that OS is not active during any of this time, something similar to [this suspend bug](https://github.com/system76/firmware-open/issues/342).

### Steps to reproduce

1. Disconnect AC power
1. `sudo systemctl suspend`
1. Wait five minutes

### Expected behavior
System should remain suspended (LED blinking green)

### Actual behavior
System exits suspend after four minutes (LED solid green)

### Additional info
- Issue does not appear to be present if connected to AC power
- [Original comments here](https://github.com/system76/firmware-open/issues/342#issuecomment-1284559347)

crawfxrd, 2022-10-25:
On d8d400d4ed34 with EC master, I see `SLP_S0#` asserted most of the time (blinking power LED), but it sometimes (at least twice) deasserts (solid power LED) briefly and then asserts it again.

crawfxrd, 2022-10-25:
What SSD are you using?

May somehow be caused by the SSD1 RTD3 config. Can you try with RTD3 config removed:

```diff
diff --git a/src/mainboard/system76/tgl-u/variants/galp5/overridetree.cb b/src/mainboard/system76/tgl-u/variants/galp5/overridetree.cb
index ce3ecc3a664e..445c41dbe172 100644
--- a/src/mainboard/system76/tgl-u/variants/galp5/overridetree.cb
+++ b/src/mainboard/system76/tgl-u/variants/galp5/overridetree.cb
@@ -21,12 +21,6 @@ chip soc/intel/tigerlake
                        # PCIe PEG0 x4, Clock 0 (SSD1)
                        register "PcieClkSrcUsage[0]" = "0x40"
                        register "PcieClkSrcClkReq[0]" = "0"
-                       chip soc/intel/common/block/pcie/rtd3
-                               register "enable_gpio" = "ACPI_GPIO_OUTPUT_ACTIVE_HIGH(GPP_D14)" # SSD1_PWR_DN#
-                               register "reset_gpio" = "ACPI_GPIO_OUTPUT_ACTIVE_LOW(GPP_H0)" # GPP_H0_RTD3
-                               register "srcclk_pin" = "0" # SSD1_CLKREQ#
-                               device generic 0 on end
-                       end
                end
                device ref north_xhci on # J_TYPEC2
                        register "UsbTcPortEn" = "1"
```

curiousercreative, 2022-10-25:
> What SSD are you using?

@crawfxrd [Samsung 980 1TB (144d:a809)](https://www.samsung.com/us/computing/memory-storage/solid-state-drives/980-pcie-3-0-nvme-gaming-ssd-1tb-mz-v8v1t0b-am/)

curiousercreative, 2022-10-25:
> Can you try with RTD3 config removed

@crawfxrd no difference with that patch applied, firmware flashed and system power cycled.

curiousercreative, 2022-11-11:
@crawfxrd stumbled upon an interesting tidbit: when I have my USB-C headset (Pixel wired earbuds) connected to the TB4 port (left side, USB-PD on my galp5), the machine remains suspended. If I disconnect the headset, it'll take about four minutes to go from blinking green to solid green. If I reconnect the headset, it'll immediately go back to blinking green.

curiousercreative, 2022-11-12:
Similar findings, passive USB-C to USB-A adapters do the trick as well. I tested both a 2.0 and 3.0 adapter with nothing on the other end. So I'd guess that anything connected to that port will keep suspend working.

crawfxrd, 2022-11-18:
Testing on the latest commit (5f1ef1731e67), I do not see the issue on either my dev unit or the newer batch unit that customers should have.

curiousercreative, 2022-11-20:
Nice, I'll test on mine

curiousercreative, 2022-11-21:
@crawfxrd after rebasing https://github.com/system76/firmware-open/pull/365, my galp5 remains in suspend on battery power, so let's close this for now. Thanks very much for your help.

curiousercreative, 2022-11-21:
I spoke too soon. While suspend with lid open works fine, lid closed continues to exit suspend after the same four minutes as previously. Can you test your unit for lid close please @crawfxrd ?

crawfxrd, 2022-11-22:
I retested both on b337ac66fb3b and don't see the issue. They remained in S0ix for 10 minutes before I woke them.