system76/firmware-open #373: `dapr8` Sometimes does not reach "S0" during suspend

thomas-zimmerman, 2022-11-04:
- Model: `darp8`
- BIOS version: 2022-07-13_8b8344a
- EC version:  2022-07-13_8b8344a
- OS: Pop!OS 22.04

During `fwts` suspend tests, the system fails to consistently reach "S0" sleep states. A failure looks like this in the `fwts` log:
```
s2idle cycle 96 of 100 
Using logind as the default power method.
Requesting Suspend action 
s2idle duration = 6.
pm-action returned 0 after 6 seconds.
FAILED [HIGH] S2idleNotDeepest: Test 1, Expected /sys/kernel/debug/pmc_core
/slp_s0_residency_usec to increase from 236839942, got 236839942.
Suspend/Resume Timings:
  Suspend: 0.849 seconds.
  Could not determine time to resume.
```

### Steps to reproduce

Suspend the system. 

### Expected behavior

Suspend reaches full "S0" sleep states and reliably resumes from suspend.

### Additional info

Full suspend log from customer testing on i5 version of `darp8` with Samsung 980 Pro NVMe 1TB OS drive.
[suspend-tests.log](https://github.com/system76/firmware-open/files/9942350/suspend-tests.log)

madmanxxx, 2022-11-05:
Side note: When I suspend my darp8 with Fn + F12 it *looks* like it is suspended, but the power LED is solid green. It starts blinking right after I close the lid.
I *think* the battery drains faster without closed lid and the back seems to be warm, but I haven't done any thorough tests...

Hardware: stock darp8 with i5
Software: Arch Linux updated today

adeptg, 2022-11-05:
@madmanxxx   I experience the behavior described here: https://github.com/system76/firmware-open/issues/342#issuecomment-1270732674
The second action (in your case - lid close) is needed for the firmware to go to the suspend state

crawfxrd, 2022-11-23:
Needs testing on 2022-11-21_b337ac.

But also, I don't think I've seen fwts work correctly since CML. It either ends early (like stops after 9 cycles of 20) or says `slp_s0_residency_usec` didn't change even when `SLP_S0#` asserted.

crawfxrd, 2022-12-01:
Fixes and workarounds for S0ix released in 2022-11-21_b337ac6.

closed 2022-12-01 by crawfxrd