system76/firmware-open #381: darp7 overdraw on battery while stressing

jacobgkau, 2022-11-22:
- Model:  darp7
- BIOS version:  `2021-07-20_93c2809` and `2022-11-21_b337ac6`
- EC version:  Same as BIOS
- OS:  Pop!_OS 22.04
- Kernel:  6.0.6-76060006

### Steps to reproduce

When running `stress -c 4 -i 4 -m 4 -d 4` while on battery power, the system sometimes powers off and cannot be powered on again for ~10 seconds (overdraw symptoms.) It may happen more consistently when the battery is low (<40% or as low as ~15% for testing).

### Expected behavior

 System does not power off.

### Actual behavior

 System powers off.

### Additional info

 Not specific to any RAM or SSD configuration.

jacobgkau, 2022-12-01:
> This happens on my oryx8 too. However it can also happen when the battery is fully charged. It mainly seems to happen when the system has recently come out of a sleep state.

Since the power limits are set separately for different models, I'm not sure if it would be a good thing to conflate one issue with reports of multiple models.

What specific oryp8 do you have (screen size and GPU)?

rnd-ash, 2022-12-04:
> > This happens on my oryx8 too. However it can also happen when the battery is fully charged. It mainly seems to happen when the system has recently come out of a sleep state.
> 
> Since the power limits are set separately for different models, I'm not sure if it would be a good thing to conflate one issue page with reports of multiple models.
> 
> What specific oryp8 do you have (screen size and GPU)?

Oryx 8 - 15.6" with rtx 3070.

I did some more testing today, and have found the issue only seems to occur after the laptop was asleep and unplugged. As soon as it wakes up and fire anything up intensive (even just an IDE), It dies.

MilesBHuff, 2023-06-27:
This seems to happen to me on Oryx Pro 7 while plugged-in.