system76/firmware-open #382: bug in calling system76-power charge-threshold

jontis, 2022-11-24:
Lemur lemp9
Bios 2021-07-20_93c2809
PopOs 22.04
Kernel 6.0.6-76060006-generic

Copied exactly from docs and worked before:
https://support.system76.com/articles/laptop-battery-thresholds/

$ system76-power charge-thresholds 40 80
thread 'main' panicked at 'Mismatch between definition and access of `thresholds`. Could not downcast to TypeId { t: 4014619811965266330 }, need to downcast to TypeId { t: 16733916091185954916 }
', /build/system76-power-95IAuY/system76-power-1.1.24~1667870429~22.04~e0a6fb8/vendor/clap/src/parser/error.rs:30:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

All system components are up to date as of 221124

ahoneybun, 2022-11-25:
This has been reported in this bug report:

https://github.com/pop-os/system76-power/issues/368

leviport, 2022-11-28:
Closing as a duplicate of https://github.com/pop-os/system76-power/issues/368

This was fixed in https://github.com/pop-os/system76-power/pull/369. We don't generally push updates before weekends (especially if they're long holiday weekends), but the fix should be released today.

closed 2022-11-28 by leviport