system76/firmware-open #384: darp8: no s0ix residency observed + laptop doen't cool down and power LED stays solid green in s2idle.

rkintzi, 2022-12-05:
- Model: darp8
- BIOS version: 2022-11-21_b337ac6
- EC version: as above
- OS: Arch Linux
- Kernel: 6.0.10-arch2-1

After firmware upgrade, laptop doesn't suspend properly after `systemctl suspend` or `echo s2idle > /sys/power/mem_sleep && echo mem > /sys/power/state`. The power led (the one to the right of the power button) stays solid green and the laptop doesn't cool down. I expect a huge battery drain (yet to test). 

With previous firmware version (2022-07-13_8b8344a) power led was blinking green and laptop cooled down when suspended. 

Note, that even with the previous firmware version the laptop never really entered the s0ix state. As a result, the battery level used to drop 20% overnight. I don’t expect anything better on the new firmware (contrary, I expect an even larger drop).

I link two files with results of `s0ix-selftest-tool.sh` run using recent and previous firmware:
 
* [2022-07-13_8b8344a-s0ix-output.log](https://0x0.st/ok5B.log)
* [2022-11-21_b337ac6-s0ix-output.log](https://0x0.st/ok5M.log)
* [s0ix-selftest-tool.sh](https://github.com/intel/S0ixSelftestTool)

### Steps to reproduce

1. Upgrade firmware
2. Run `echo s2idle > /sys/power/mem_sleep && echo mem > /sys/power/state`.
3. Observe LED and temperature.
4. Run `./s0ix-selftest-tool.sh` as root and check results.

### Expected behavior

1. Power LED blinks green when laptop is suspended
2. Laptop cools down when suspended
3. Laptop enters s0ix state 
4. Battery is preserved in suspend

### Actual behavior

1. Power LED stays solid green.
2. Laptop doesn't cool down
3. Battery drop in suspend in about 20% overnight.

crawfxrd, 2022-12-05:
What SSDs are you using?

jacobgkau, 2022-12-05:
> BTW: what is most convenient way of downgrading firmware ?

The previous release was version `2022-07-13_8b8344a`. The part at the end is a commit hash, so you can check out that commit and then build/flash firmware on the machine if you'd like:

```
git clone https://github.com/system76/firmware-open
git checkout 8b8344a
./scripts/update.sh
git submodule update --init --recursive
./scripts/deps.sh
./scripts/build.sh darp8
./scripts/flash.sh darp8
```

If you flash this way, be aware that the last command will power off the system.

For something slightly more "convenient" (since I don't know how all of the build dependencies map on Arch), here is an image that you can flash to a USB drive using Popsicle: https://www.dropbox.com/s/irs2l84fix83ha7/oryp8-8b8344a.img?dl=1 Flashing the image will erase the USB drive. After flashing, boot from the USB drive, and you should see an updater similar to the one you saw when performing the over-the-air update.

rkintzi, 2022-12-06:
I tried to build the firmware on Arch, but this failed with g++ reporting warnings and treating them as errors. I guess Arch's g++ is newer or differently configured. I link to the output below for your reference.

Then I tried to build inside the Pop!_OS VM and this worked. At least `./scripts/build.sh darp8` didn’t report any errors. But I got a single file `build/darp8/UEFIPAYLOAD.fd` as a result which was a bit surprising. Anyway, I wasn’t sure what should  move to the host machine in order to execute the next step (ie. installation) there, so I gave up and installed PopOS on a spare partition.

I built again and got the same results (a single file in `build/darp8` directory). I ran the `flash.sh` script which ended with an error which isn't very informative. I don’t really know what file is missing:

```
$  ./scripts/flash.sh darp8
    Updating crates.io index
  Downloaded redox_intelflash v0.1.3
  Downloaded libc v0.2.138
  Downloaded redox_uefi v0.1.9
  Downloaded coreboot-fs v0.1.1
  Downloaded 4 crates (636.0 KB) in 1.57s
   Compiling libc v0.2.138
   Compiling plain v0.2.3
   Compiling redox_uefi v0.1.9
   Compiling bitflags v1.3.2
   Compiling coreboot-fs v0.1.1
   Compiling redox_intelflash v0.1.3
   Compiling intel-spi v0.1.4 (/home/radek/firmware-open/libs/intel-spi)
    Finished release [optimized] target(s) in 1m 48s
SPI HSFSTS_CTL: FDONE | WRSDIS | PRR34_LOCKDN | FDOPSS | FDV | FLOCKDN | FCYCLE_1 | FCYCLE_2 | FDBC_0 | FDBC_2
SPI FILE: Loadingthread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 2, kind: NotFound, message: "No such file or directory" }', src/main.rs:68:30
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

[Output of ./script/build.sh darp8 on Pop!_OS (bare metal)](https://0x0.st/odz4.out)
[Output of ./script/build.sh darp8 on Arch](https://0x0.st/odis.log) - g++ version attached on top

I haven't try the second approach, as I want to tweak fan points, so I need to figure it out how to build firmware anyway. Ideally on Arch (or in Pos!_OS VM).

crawfxrd, 2022-12-06:
Commits before 42814e6e5c18 depends on the version of host compiler used to build edk2. You may also have issues building older versions of the coreboot toolchain with newer versions of GCC.

Pop!_OS is failing to build due to:

```
python: can't open file '/home/radek/firmware-open/coreboot/3rdparty/fsp/Tools/SplitFspBin.py': [Errno 2] No such file or directory
building with libflashrom support
make: *** [src/drivers/intel/fsp2_0/Makefile.inc:90: build/Fsp_M.fd] Error 2
make: *** Waiting for unfinished jobs....
```

rkintzi, 2023-02-11:
Any progress on this issue? This is really annoying that I have to hibernate instead suspending. 

I did more tests only to confirm that the issue isn't really dependent on ssd disks (as output from s0ix-selftest-tool.sh suggests). No matter which disk I use, system refuse to suspend to idle not mentioning about s0ix.

I won't able to downgrade the firmware. The image provided by jacobgkau didn't work with simple message about the error - not details.

crawfxrd, 2023-02-11:
If you would like to build and test, the coreboot 4.19 rebase was recently merged. It contains fixes that allow up to drop the PMC hack for S0ix, and so may fix the issue you have.

But new releases for ADL won't be made until after the RPL work is done.

rkintzi, 2023-02-11:
Also, let me add one more to the list: the highest package c-state I can observe is c2. Even when idling on console.

DrymarchonShaun, 2023-10-02:
For what its worth, I'm getting [20231002-14-30-s0ix-output.log](https://github.com/system76/firmware-open/files/12786662/20231002-14-30-s0ix-output.log) on the most recent (stable) firmware version, and kernel version 6.5.5

the presumed important bits probably being 
```

Below are the deeper S0ix substate required IPs did not show YES:

[31m                     SPB_PG_STS    Required             
       MPHY_Core_DL_REQ_STS_16    Required              [0m 

Your system south port controller did not meet S0ix requirement: [31mSPB
SPC[0m

....

Checking PCI Devices tree diagram:
-[0000:00]-+-00.0  Intel Corporation Device 4621
           +-02.0  Intel Corporation Alder Lake-P Integrated Graphics Controller
           +-06.0-[01]----00.0  Sandisk Corp WD Blue SN550 NVMe SSD
           +-07.0-[02-2c]--
           +-0a.0  Intel Corporation Platform Monitoring Technology
           +-0d.0  Intel Corporation Alder Lake-P Thunderbolt 4 USB Controller
           +-0d.2  Intel Corporation Alder Lake-P Thunderbolt 4 NHI #0
           +-14.0  Intel Corporation Alder Lake PCH USB 3.2 xHCI Host Controller
           +-14.2  Intel Corporation Alder Lake PCH Shared SRAM
           +-14.3  Intel Corporation Alder Lake-P PCH CNVi WiFi
           +-15.0  Intel Corporation Alder Lake PCH Serial IO I2C Controller #0
           +-15.1  Intel Corporation Alder Lake PCH Serial IO I2C Controller #1
           +-16.0  Intel Corporation Alder Lake PCH HECI Controller
           +-1c.0-[2d]----00.0  O2 Micro, Inc. SD/MMC Card Reader Controller
           +-1c.7-[2e]----00.0  Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
           +-1d.0-[2f]----00.0  Sandisk Corp WD Blue SN570 NVMe SSD 1TB
           +-1f.0  Intel Corporation Alder Lake PCH eSPI Controller
           +-1f.3  Intel Corporation Alder Lake PCH-P High Definition Audio Controller
           +-1f.4  Intel Corporation Alder Lake PCH-P SMBus Host Controller
           \-1f.5  Intel Corporation Alder Lake-P PCH SPI Controller

[31mThe pcieroot port 0000:00:06.0 ASPM setting is Enabled, its D state and Link PM are not expected,
please investigate or report a bug.[0m
```

crawfxrd, 2024-07-20:
Requires retesting on 2024-01-10_6c402c3.